esphome/issues #989: All nodes started to disconnct

joselito11, 2020-01-14:
Esphome gone weird. All nodes started to disconnecting, deleted all and added one node, problem still persist:
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.171:6053 (192.168.1.171)
INFO Successfully connected to 192.168.1.171
[08:01:06][D][ccs811:077]: Got co2=0 ppm, tvoc=0 ppb, baseline=0x9BBF
[08:01:06][D][sensor:092]: 'CCS811 eCO2 Value': Sending state 0.00000 ppm with 0 decimals of accuracy
[08:01:06][D][sensor:092]: 'CCS811 Total Volatile Organic Compound': Sending state 0.00000 ppb with 0 decimals of accuracy
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.171:6053 (192.168.1.171)
INFO Successfully connected to 192.168.1.171
[08:01:26][D][ccs811:077]: Got co2=0 ppm, tvoc=0 ppb, baseline=0x9BBF
[08:01:26][D][sensor:092]: 'CCS811 eCO2 Value': Sending state 0.00000 ppm with 0 decimals of accuracy
[08:01:26][D][sensor:092]: 'CCS811 Total Volatile Organic Compound': Sending state 0.00000 ppb with 0 decimals of accuracy
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.171:6053 (192.168.1.171)
INFO Successfully connected to 192.168.1.171
[08:01:46][D][ccs811:077]: Got co2=0 ppm, tvoc=0 ppb, baseline=0x9BBF
[08:01:46][D][sensor:092]: 'CCS811 eCO2 Value': Sending state 0.00000 ppm with 0 decimals of accuracy
[08:01:46][D][sensor:092]: 'CCS811 Total Volatile Organic Compound': Sending state 0.00000 ppb with 0 decimals of accuracy
[08:01:47][I][ota:046]: Boot seems successful, resetting boot loop counter.
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.171:6053 (192.168.1.171)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.1.171:6053 (192.168.1.171)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.1.171:6053 (192.168.1.171)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 2 seconds
INFO Connecting to 192.168.1.171:6053 (192.168.1.171)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 3 seconds
INFO Connecting to 192.168.1.171:6053 (192.168.1.171)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 5 seconds
INFO Connecting to 192.168.1.171:6053 (192.168.1.171)
INFO Successfully connected to 192.168.1.171
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.171:6053 (192.168.1.171)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.1.171:6053 (192.168.1.171)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.1.171:6053 (192.168.1.171)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 2 seconds
INFO Connecting to 192.168.1.171:6053 (192.168.1.171)
INFO Successfully connected to 192.168.1.171
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.171:6053 (192.168.1.171)
INFO Successfully connected to 192.168.1.171
[08:03:32][D][ccs811:077]: Got co2=0 ppm, tvoc=0 ppb, baseline=0x9BBF
[08:03:32][D][sensor:092]: 'CCS811 eCO2 Value': Sending state 0.00000 ppm with 0 decimals of accuracy
[08:03:32][D][sensor:092]: 'CCS811 Total Volatile Organic Compound': Sending state 0.00000 ppb with 0 decimals of accuracy
[08:03:52][D][ccs811:077]: Got co2=0 ppm, tvoc=0 ppb, baseline=0x9BBF
[08:03:52][D][sensor:092]: 'CCS811 eCO2 Value': Sending state 0.00000 ppm with 0 decimals of accuracy
[08:03:52][D][sensor:092]: 'CCS811 Total Volatile Organic Compound': Sending state 0.00000 ppb with 0 decimals of accuracy
[08:04:12][D][ccs811:077]: Got co2=0 ppm, tvoc=0 ppb, baseline=0x9BBF
[08:04:12][D][sensor:092]: 'CCS811 eCO2 Value': Sending state 0.00000 ppm with 0 decimals of accuracy
[08:04:12][D][sensor:092]: 'CCS811 Total Volatile Organic Compound': Sending state 0.00000 ppb with 0 decimals of accuracy
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.171:6053 (192.168.1.171)
INFO Successfully connected to 192.168.1.171
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.171:6053 (192.168.1.171)
INFO Successfully connected to 192.168.1.171

Please help! Nodes worked for 2 weeks

closed 2020-01-14 by close-issue-app[bot]

joselito11, 2020-01-14:
Esphome as Home assistant addon. Host OS is Ubuntu server.

Problem: Had 3 nodes, after two week, they started to disconnecting. Tried to reflash them, downgraded esphome to 1.14.1., reintall. Tried auto dhcp, then manual. But think is something with HA connection. Tried also with mqtt, this is also disconnecting, after 1minute.

template:
```
esphome:
  name: co2
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "Wifi"
  password: "pass"
  manual_ip:
    static_ip: 192.168.1.171
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.13
    dns2: 192.168.1.1

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Co2 Fallback Hotspot"
    password: "pass"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

i2c:
  sda: D1
  scl: D2

sensor:
  - platform: ccs811
    eco2:
      name: "CCS811 eCO2 Value"
    tvoc:
      name: "CCS811 Total Volatile Organic Compound"
    address: 0x5B
    baseline: 0x9BBF
    update_interval: 20s
```

glmnet, 2020-01-14:
you'll need to work harder to fix your issue:

A few things to try:

* fill template properly
* ping nodes
* connect to their log via USB
* does the node switch to ap? captive portal show up?

joselito11, 2020-01-14:
Router restart fixed it

glmnet, 2020-01-14:
of course, also leaving the house and coming back later sometimes works