esphome/issues #990: Disabling Access point resets ESP - connection problem

fremut, 2020-01-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

PIP on MacOS

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 Wemos D1 mini Lite

**Affected component:**

https://esphome.io/components/wifi.html

**Description of problem:**
ESPhome restarts ESP after disabling access point (with captive portal). That is problem during short-term periods of home wifi network disconnections (router off, etc.)

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: plugswitch01_test
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "XXXXX"
  password: "XXXXX"
  reboot_timeout: 0s

  ap:
    ssid: "XXXXX"
    password: "XXXXXXX"

captive_portal:
# Enable logging
logger:
# Enable Home Assistant API
api:
  password: "esp8266"
  reboot_timeout: 0s

ota:
  password: "esp8266"

sensor:
  - platform: uptime
    name: "Uptime"
    update_interval: 1s
    id: uptime_sensor
    unit_of_measurement: "sec"

```

**Logs (if applicable):**

```
[12:20:06][C][wifi:298]:   DNS1: 213.46.172.36
[12:20:06][C][wifi:299]:   DNS2: 213.46.172.37
[12:20:06][D][wifi:432]: Disabling AP...
[12:20:06][D][sensor:092]: 'Uptime': Sending state 564.92297 sec with 0 decimals of accuracy
[12:20:06]
[12:20:06]Exception (28):
WARNING Exception type: unknown
[12:20:06]epc1=0x4021d1f4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000028 depc=0x00000000
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d desktop/HomeAss/config/plugswitch01_test -t idedata
Usage: platformio run [OPTIONS]
Try "platformio run -h" for help.

Error: Invalid value for "-d" / "--project-dir": Path "desktop/HomeAss/config/plugswitch01_test" does not exist.
[12:20:06]
[12:20:06]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[12:20:06]
[12:20:06]ctx: cont
[12:20:06]sp: 3ffffdb0 end: 3fffffc0 offset: 01a0
[12:20:06]3fffff50:  3fffdad0 00000002 3fff02a0 4020a09f  
[12:20:06]3fffff60:  00000001 000003a3 00000000 40218f67  
[12:20:06]3fffff70:  3fff1908 402266fe 4020f3cc 4021d62c  
[12:20:06]3fffff80:  3fff1908 402266fe 00000000 3fff05a0  
[12:20:06]3fffff90:  3fffdad0 00000000 3fff0570 4020c61c  
[12:20:06]3fffffa0:  3fffdad0 00000000 3fff0570 4021ab64  
[12:20:06]3fffffb0:  feefeffe feefeffe 3ffe8538 401007e5  
[12:20:06]<<<stack<<<
[12:20:06]
[12:20:06] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[12:20:06]
[12:20:06]load 0x4010f000, len 1384, room 16 
[12:20:06]tail 8
[12:20:06]chksum 0x2d
[12:20:06]csum 0x2d
[12:20:06]v8b899c12
[12:20:06]~ld
[12:20:06][I][logger:156]: Log initialized
[12:20:06][C][ota:364]: There have been 0 suspected unsuccessful boot attempts.
[12:20:06][I][app:028]: Running through setup()...
[12:20:06][C][wifi:033]: Setting up WiFi...
[12:20:06][D][wifi:304]: Starting scan...
[12:20:06][D][sensor:092]: 'Uptime': Sending state 0.07100 sec with 0 decimals of accuracy
[12:20:07][D][sensor:092]: 'Uptime': Sending state 0.75400 sec with 0 decimals of accuracy
[12:20:08][D][sensor:092]: 'Uptime': Sending state 1.75400 sec with 0 decimals of accuracy
[12:20:09][D][sensor:092]: 'Up

```

**Additional information and things you've tried:**

fremut, 2020-01-14:
[image]
Detailed logging VERY_VERBOSE

brandond, 2020-01-14:
For some reason the platformio command to retrieve idedata (debugging symbols) is failing; it says your project directory doesn't exist. Have you deleted any files (desktop/HomeAss/config/plugswitch01_test) in particular) since compiling the esphome firmware? Without a decoded stack trace it's hard to figure out what's going on.

fremut, 2020-01-15:
After delete whole folder there is another log:
[01:00:10][D][wifi:432]: Disabling AP...
[01:00:10][V][wifi_esp8266:043]: Disabling AP.
[01:00:10]bcn 0
[01:00:10]del if1
[01:00:10]pm open,type:0 0
[01:00:10]mode : sta(68:c6:3a:e9:ed:b4)
[01:00:10][V][wifi_esp8266:376]: Event: Changed Mode old=AP+STA new=STA
[01:00:10]wifi evt: 8
[01:00:10][VV][scheduler:131]: Running interval 'update' with interval=1000 last_execution=148775 (now=149783)
[01:00:10][V][sensor:013]: 'Uptime': Received new state 149.787994
[01:00:10][D][sensor:092]: 'Uptime': Sending state 149.78799 sec with 0 decimals of accuracy
[01:00:19]
[01:00:19] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[01:00:19]
[01:00:19]wdt reset
[01:00:19]load 0x4010f000, len 1384, room 16 
[01:00:19]tail 8
[01:00:19]chksum 0x2d
[01:00:19]csum 0x2d
[01:00:19]v8b899c12
[01:00:19]~ld
[01:00:19]
[01:00:19]SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
[01:00:19][I][logger:156]: Log initialized
[01:00:19][I][app:028]: Running through setup()...
[01:00:19][V][app:029]: Sorting components by setup priority...
[01:00:19][VV][scheduler:056]: set_interval(name='update', interval=1000, offset=338)
[01:00:19][VV][preferences:038]: LOAD 0: valid=NO, 0=0xFFFFFFFF 1=0xFFFFFFFF (Type=80396918, CRC=0x342E8155)
[01:00:19][C][wifi:033]: Setting up WiFi...
[01:00:19][V][wifi_esp8266:043]: Disabling AP.
[01:00:19]bcn 0
[01:00:19]del if1
[01:00:19]usl
[01:00:19]mode : null
[01:00:19][V][wifi_esp8266:033]: Enabling STA.
[01:00:19]mode :[V][wifi_esp8266:376]: Event: Changed Mode old=AP new=OFF
[01:00:19]wifi evt: 8
[01:00:19] sta(68:c6:3a:e9:ed:b4)
[01:00:19]add if0
[01:00:19][V][wifi_esp8266:376]: Event: Changed Mode old=OFF new=STA
[01:00:19]wifi evt: 8
[01:00:19]sleep disable
[01:00:19][D][wifi:304]: Starting scan...
[01:00:19][VV][scheduler:131]: Running interval 'update' with interval=1000 last_execution=4294966021 (now=120)
[01:00:19][V][sensor:013]: 'Uptime': Received new state 0.128000
[01:00:19][D][sensor:092]: 'Uptime': Sending state 0.12800 sec with 0 decimals of accuracy
[01:00:19][VV][scheduler:131]: Running interval 'update' with interval=1000 last_execution=4294967021 (now=725)
[01:00:19][V][sensor:013]: 'Uptime': Received new state 0.725000
[01:00:19][D][sensor:092]: 'Uptime': Sending state 0.72500 sec with 0 decimals of accuracy
[01:00:20][VV][scheduler:131]: Running interval 'update' with interval=1000 last_execution=725 (now=1725)
[01:00:20][V][sensor:013]: 'Uptime': Received new state 1.725000
[01:00:20][D][sensor:092]: 'Uptime': Sending state 1.72500 sec with 0 decimals of accuracy
[01:00:21][VV][scheduler:131]: Running interval 'update' with interval=1000 last_execution=1725 (now=2725)
[01:00:21][V][sens

fremut, 2020-01-15:
And next test with this result (contained some python errors) :

[01:06:22][C][wifi:298]:   DNS1: 213.46.172.36
[01:06:22][C][wifi:299]:   DNS2: 213.46.172.37
[01:06:22][D][wifi:432]: Disabling AP...
[01:06:22][V][wifi_esp8266:043]: Disabling AP.
[01:06:22]bcn 0
[01:06:22]del if1
[01:06:22]pm open,type:0 0
[01:06:22]mode : sta(68:c6:3a:e9:ed:b4)
[01:06:22]Fatal exception 9(LoadStoreAlignmentCause):
[01:06:22]epc1=0x40217e5f, ep[V][wifi_esp8266:376]: Event: Changed Mode old=AP+STA new=STA
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d desktop/HomeAss/config/plugswitch01_test -t idedata
Error: Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/platformio/commands/run/command.py", line 138, in cli
    is_test_running,
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/platformio/commands/run/command.py", line 165, in process_env
    print_processing_header(name, config, verbose)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/platformio/commands/run/command.py", line 196, in print_processing_header
    % (click.style(env, fg="cyan", bold=True), "; ".join(env_dump))
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/platformio/__main__.py", line 90, in <lambda>
    click.echo = lambda *args, **kwargs: _safe_echo(0, *args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/platformio/__main__.py", line 84, in _safe_echo
    click_echo_origin[origin](*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/utils.py", line 260, in echo
    file.write(message)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/_compat.py", line 71, in write
    return self.buffer.write(str(x))
TypeError: unicode argument expected, got 'str'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

[01:06:23]wifi evt: 8
[01:06:23]c2=0x00000000, epc3=0x00000000, excvaddr=0xa5a5a5c9, depc=0x00000000
[01:06:23]
[01:06:23]Exception (9):
WARNING Exception type: unknown
[01:06:23]epc1=0x40217e5f epc2=0x00000000 epc3=0x00000000 excvaddr=0xa5a5a5c9 depc=0x00000000
[01:06:23]
[01:06:23]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[01:06:23]
[01:06:23]ctx: cont
[01:06:23]sp: 3ffffdb0 end: 3fffffc0 offset: 01a0
[01:06:23]3fffff50:  3fffdad0 00000102 3ffef4cc 40205c35  
[01:06:23]3fffff60:  00000001 000000f1 00000000 402131eb  
[01:06:23]3fffff70:  3fff0b18 3fff0e40 40209388 40218080  
[01:06:23]3fffff80:  3fff0b18 3fff0e40 00000000 3ffef7c0  
[01:06:23]3fffff90:  3fffdad0 00000000 3ffef790 402080b0  
[01:06:23]3fffffa0:  3fffdad0 00000000 3ffef790 40215114  
[01:06:23]3fffffb0:  feefeffe feefeffe 3ffe8534 40100831  
[01:06:23]<<<stack<<<
[01:06:23]
[01:06:23] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[01:06:23]
[01:06:23]wdt reset
[01:06:23]load 0x4010f000, len 1384, room 16 
[01:06:23]tail 8
[01:06:23]chksum 0x2d
[01:06:23]csum 0x2d
[01:06:23]v8b899c12
[01:06:23]~ld
[01:06:23]
[01:06:23]SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
[01:06:23][I][logger:156]: Log initialized
[01:06:23][I][app:028]: Running through setup()...
[01:06:23][V][app:029]: Sorting components by setup priority...
[01:06:23][VV][scheduler:056]: set_interval(nam

fremut, 2020-01-15:
After update to python 3.8 and recompiling project I get this:

[01:21:07][C][wifi:299]:   DNS2: 213.46.172.37
[01:21:07][D][wifi:432]: Disabling AP...
[01:21:07][V][wifi_esp8266:043]: Disabling AP.
[01:21:07]bcn 0
[01:21:07]del if1
[01:21:07]pm open,type:0 0
[01:21:07]mode : sta(68:c6:3a:e9:ed:b4)
[01:21:07][V][wifi_esp8266:376]: Event: Changed Mode old=AP+STA new=STA
[01:21:07]wifi evt: 8
[01:21:24]Fatal exception 9(LoadStoreAlignmentCause):
[01:21:24]epc1=0x40208fde, epc2=0x00000000, epc3=0x00000000, excvaddr=0x02a4760d, depc=0x00000000
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d config/plugswitch01_test -t idedata
WARNING Decoded 0x40208fde: WiFiUDP::parsePacket()
[01:21:26]
[01:21:26]Exception (9):
WARNING Exception type: unknown
[01:21:26]epc1=0x40208fde epc2=0x00000000 epc3=0x00000000 excvaddr=0x02a4760d depc=0x00000000
WARNING Decoded 0x40208fde: WiFiUDP::parsePacket()
[01:21:26]
[01:21:26]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[01:21:26]
[01:21:26]ctx: cont
[01:21:26]sp: 3ffffd60 end: 3fffffc0 offset: 01a0
[01:21:26]3fffff00:  3fff2304 00000002 3ffef4cc 40212564  
WARNING Decoded 0x40212564: DNSServer::processNextRequest()
[01:21:26]3fffff10:  3fff0e3c 00000002 3ffef4cc 40201049  
WARNING Decoded 0x40201049: esphome::captive_portal::CaptivePortal::loop()
[01:21:26]3fffff20:  3fff0e2c 00022a44 401004ba 4020105c  
WARNING Decoded 0x401004ba: millis
WARNING Decoded 0x4020105c: non-virtual thunk to esphome::captive_portal::CaptivePortal::loop()
[01:21:26]3fffff30:  00000000 00000000 3ffef4cc 40217dd0  
WARNING Decoded 0x40217dd0: esphome::Component::call_loop()
[01:21:26]3fffff40:  3fff0e3c 00000002 3ffef4cc 40217e64  
WARNING Decoded 0x40217e64: esphome::Component::call()
[01:21:26]3fffff50:  3fffdad0 00000002 3ffef4cc 40205c35  
WARNING Decoded 0x40205c35: esphome::Application::loop()
[01:21:26]3fffff60:  00000001 000003da 00000000 402131eb  
WARNING Decoded 0x402131eb: run_scheduled_functions()
[01:21:26]3fffff70:  3fff0b18 3fff0e40 40209388 40218080  
WARNING Decoded 0x40209388: std::_Function_base::_Base_manager<std::_Bind<std::_Mem_fn<bool (esp8266::MDNSImplementation::MDNSResponder::*)()> (esp8266::MDNSImplementation::MDNSResponder*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation)
WARNING Decoded 0x40218080: std::_Function_handler<void (), std::_Bind<std::_Mem_fn<bool (esp8266::MDNSImplementation::MDNSResponder::*)()> (esp8266::MDNSImplementation::MDNSResponder*)> >::_M_invoke(std::_Any_data const&)
[01:21:26]3fffff80:  3fff0b18 3fff0e40 00000000 3ffef7c0  
[01:21:26]3fffff90:  3fffdad0 00000000 3ffef790 402080b0  
WARNING Decoded 0x402080b0: loop
[01:21:26]3fffffa0:  3fffdad0 00000000 3ffef790 40215114  
WARNING Decoded 0x40215114: loop_wrapper() at core_esp8266_main.cpp
[01:21:26]3fffffb0:  feefeffe feefeffe 3ffe8534 40100831  
WARNING Decoded 0x40100831: cont_wrapper
[01:21:26]<<<stack<<<
[01:21:26]
[01:21:26] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[01:21:26]
[01:21:26]load 0x4010f000, len 1384, room 16 
[01:21:26]tail 8
[01:21:26]chksum 0x2d
[01:21:26]csum 0x2d
[01:21:26]v8b899c12
[01:21:26]~ld
[01:21:26]
[01:21:26]SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
[01:21:26][I][logger:156]: Log initialized
[01:21:26][I][app:028]: Running through setup()...
[01:21:26][V][app:029]: Sorting components by setup priority...
[01:21:26][VV][scheduler:056]: set_interval(name='update', interval=1000, offset=374)
[01:21:26][VV][preferences:038]: LOAD 0: valid=NO, 0=0xFFFFFFFF 1=0xFFFFFFFF (Type=2208194400, CRC=0xB37A1C43)
[01:21:26][C][wifi:033]: Setting up WiFi...
[01:21:26][V][wifi_esp8266:043]: Disabling AP.
[01:21:26]bcn 0
[01:21:26]del if1
[01:21:26]usl
[01:21:26]mode : null
[01:21:26][V][wifi_esp8266:033]: Enabling STA.
[01:21:26]mode :[V][wifi_esp8266:376]: Event: Changed Mode old=AP new=OFF
[01:21:26]wifi evt: 8
[01:21:26] sta(68:c6:3a:e9:ed:b4)
[01:21:26]add if0
[01:21:26][V][wifi_esp8266:376]: Event: Changed Mode old=OFF new=STA
[01:21:26]wifi evt: 8
[01:21:26]sleep disable
[01:21:26][D][wifi:304]: Starting scan...
[01:21:26][VV][scheduler:131]: Running interval 'update' with interval=1000 last_execution=4294965985 (now=120)
[01:21:26][V][sensor:013]: 'Uptime': Received new state 0.127000
[01:21:26][D][sensor:092]: 'Uptime': Sending state 0.12700 sec with 0 decimals of accuracy

fremut, 2020-01-15:
I am afraid that every trying cause different stack :(

brandond, 2020-01-15:
Can you share a couple? Do they have anything in common?

fremut, 2020-01-15:
I tried to test it 10 times. Same test routine: Start esp, wait for wifi connection, Disable home wifi network, wait for starting captive portal, start home wifi network back again, wait for connecting ....  it reset esp during disabling AP.

fremut, 2020-01-15:
In logs are same( or common) errors what i shared before:

1. WiFiUDP::parsePacket()
2. DNSServer::processNextRequest()
3. esphome::captive_portal::CaptivePortal::loop()
4. esphome::Component::call_loop()
5. esphome::Component::call()
6. esphome::Application::loop()
7. run_scheduled_functions()
8. MDNSImplementation::MDNSResponder::*)()> 
9. MDNSImplementation::MDNSResponder::*)()> 
loop and than reset

fremut, 2020-01-15:
In some cases it starts with point nr. 4 but rest is same

fremut, 2020-01-15:
Like this:
'''
[23:32:03]Fatal exception 9(LoadStoreAlignmentCause):
[23:32:03]epc1=0x4021ee70, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00e0a629, depc=0x00000000
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d config/plugswitch01_test -t idedata
WARNING Decoded 0x4021ee70: esphome::Component::call()
[23:32:04]
[23:32:04]Exception (9):
WARNING Exception type: unknown
[23:32:04]epc1=0x4021ee70 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00e0a629 depc=0x00000000
WARNING Decoded 0x4021ee70: esphome::Component::call()
[23:32:04]
[23:32:04]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[23:32:04]
[23:32:04]ctx: cont
[23:32:04]sp: 3ffffdb0 end: 3fffffc0 offset: 01a0
[23:32:04]3fffff50:  3fffdad0 00000002 3fff05b4 4020b435  
WARNING Decoded 0x4020b435: esphome::Application::loop()
[23:32:04]3fffff60:  00000001 000001cb 00000000 4021a81f  
WARNING Decoded 0x4021a81f: run_scheduled_functions()
[23:32:04]3fffff70:  3fff1c70 3fff1f0c 40210b2c 4021f2d4  
WARNING Decoded 0x40210b2c: std::_Function_base::_Base_manager<std::_Bind<std::_Mem_fn<bool (esp8266::MDNSImplementation::MDNSResponder::*)()> (esp8266::MDNSImplementation::MDNSResponder*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation)
WARNING Decoded 0x4021f2d4: std::_Function_handler<void (), std::_Bind<std::_Mem_fn<bool (esp8266::MDNSImplementation::MDNSResponder::*)()> (esp8266::MDNSImplementation::MDNSResponder*)> >::_M_invoke(std::_Any_data const&)
[23:32:04]3fffff80:  3fff1fac 3fff1f0c 00000000 3fff08b8  
[23:32:04]3fffff90:  3fffdad0 00000000 3fff0888 4020dc14  
WARNING Decoded 0x4020dc14: loop
[23:32:04]3fffffa0:  3fffdad0 00000000 3fff0888 4021c7e0  
WARNING Decoded 0x4021c7e0: loop_wrapper() at core_esp8266_main.cpp
[23:32:04]3fffffb0:  feefeffe feefeffe 3ffe8538 401008cd  
WARNING Decoded 0x401008cd: cont_wrapper
[23:32:04]<<<stack<<<
[23:32:04]
[23:32:04] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[23:32:04]
[23:32:04]load 0x4010f000, len 1384, room 16 
[23:32:04]tail 8
[23:32:04]chksum 0x2d
[23:32:04]csum 0x2d
[23:32:04]v8b899c12
[23:32:04]~ld
[23:32:04]
[23:32:04]SDK:2.2.1(cfd48f3)/Core:2.5.2=20502000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
[23:32:04][I][logger:156]: Log initialized
[23:32:04][C][status_led:014]: Setting up Status LED...
[23:32:04][VV][preferences:038]: LOAD 32: valid=YES, 0=0x00000000 1=0x0DEFE4E3 (Type=233825507, CRC=0x0DEFE4E3)
[23:32:04][C][ota:364]: There have been
'''

fremut, 2020-01-17:
workaround:
YAML:
```yaml
ap:
    ssid: "XXXXXX"
    password: "XXXXXX"
    ap_timeout: 4294967295ms
```

Add some button or hold some button:
```yaml
id: button
    on_press:
      - lambda: |-
          wifi_wificomponent->set_ap_timeout(10000);
```

So AP never start until you want it. Wifi connection or damage of connectin stops ocur esp resets

closed 2020-12-05 by stale[bot]

taraant, 2021-06-12:
Problem still exists!!!
I have the same thing.
I want to enable AP together with STA, as the documentation says:
[https://esphome.io/components/wifi.html#access-point-mode](https://esphome.io/components/wifi.html#access-point-mode):
_Additionally, you can specify both a “normal” station mode and AP mode at the same time._

The search for parameters on how to do this did not lead to success. :(
Looking at the source code:
`src / esphome / components / wifi / wifi_component.cpp`
From line 114-115: 
```
if (this-> has_ap () &&! This-> ap_setup_) {
  if (now - this-> last_connected_> this-> ap_timeout_) {
```
I realized that the AP starts if the timeout is less than the time of the last activity (that is, setting it to 0 - this condition will be met).

By making the timeout equal to "0", I get the device to reboot as soon as it connects to the access point:
 
My config:
esphome:
  name: MYDEV
  platform: ESP32
  board: ttgo-t1

web_server:
  port: 80

wifi:
  ssid: "MYSSID"
  password: "MYPASS"
  reboot_timeout: 60s
  power_save_mode: none
  ap:
    ssid: "MYDEV - Fallback AP"
    password: "MYPASS"
    ap_timeout: 0s 
    
Log out:```

[21:52:58]Rebooting...
[21:52:58]ets Jul 29 2019 12:21:46
[21:52:58]
[21:52:58]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[21:52:58]configsip: 0, SPIWP:0xee
[21:52:58]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[21:52:58]mode:DIO, clock div:2
[21:52:58]load:0x3fff0018,len:4
[21:52:58]load:0x3fff001c,len:1216
[21:52:58]ho 0 tail 12 room 4
[21:52:58]load:0x40078000,len:9720
[21:52:58]ho 0 tail 12 room 4
[21:52:58]load:0x40080400,len:6364
[21:52:58]entry 0x400806b8
[21:52:58][I][logger:170]: Log initialized
[21:52:58][I][app:029]: Running through setup()...
[21:52:58][I][wifi:116]: Starting fallback AP!
[21:53:01][I][wifi:385]: - 'MYSSID' (MYMAC) ▂▄▆█
[21:53:01][I][wifi:194]: WiFi Connecting to 'MYSSID'...
[21:53:03][I][wifi:457]: WiFi Connected!
[21:53:03]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[21:53:03]Core 1 register dump:
[21:53:03]PC      : 0x400de0ef  PS      : 0x00060330  A0      : 0x800e4cb0  A1      : 0x3ffb1bc0  
WARNING Decoded 0x400de0ef: WiFiUDP::stop() at /home/myhome/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiUdp.cpp:129
[21:53:03]A2      : 0x00000000  A3      : 0x400859fc  A4      : 0x3ffc19d0  A5      : 0x3ffd01d4  
[21:53:03]A6      : 0x00000000  A7      : 0x3ffb0060  A8      : 0x8000beca  A9      : 0x3ffb1bb0  
[21:53:03]A10     : 0x3ffb80c0  A11     : 0x3ffd0194  A12     : 0x00000003  A13     : 0x00000020  
[21:53:03]A14     : 0x00000020  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
[21:53:03]EXCVADDR: 0x00000028  LBEG    : 0x4000142d  LEND    : 0x4000143a  LCOUNT  : 0xffffffe7  
[21:53:03]
[21:53:03]Backtrace: 0x400de0ef:0x3ffb1bc0 0x400e4cad:0x3ffb1bf0 0x400d8db7:0x3ffb1c10 0x400d8f11:0x3ffb1c50 0x4015f98d:0x3ffb1c70 0x4015fa2d:0x3ffb1c90 0x400dab55:0x3ffb1cb0 0x400dce96:0x3ffb1d00 0x400e753f:0x3ffb1fb0 0x4008a1a9:0x3ffb1fd0

```