esphome/issues #991: Flickering on Arilux E27, completely wrong colors

alexiri, 2020-01-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

Arilux E27 lightbulb, running esp01_1m

**Affected component:**

https://esphome.io/components/output/my9231.html
https://esphome.io/components/light/rgbw.html

**Description of problem:**
I'm trying to migrate my Arilux E27 lightbulb from ESPurna to ESPHome, but it doesn't work correctly in ESPHome. I can turn it on or off, but it starts flickering like crazy with all transitions. The colors also don't really work, the bulb flickers and stays mostly white.

Are my output channels all wrong? Have I misconfigured something? I basically just copied what I found [here](https://github.com/maartenpaauw/esphome-config/blob/master/arilux_e27_1.yaml).

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: arilux_e27_1
  platform: ESP8266
  board: esp01_1m

my9231:
  data_pin: GPIO13
  clock_pin: GPIO15

light:
  - platform: rgbw
    name: arilux_e27_1_light
    red: red
    green: green
    blue: blue
    white: white

output:
  - platform: my9231
    id: red
    channel: 3
  - platform: my9231
    id: green
    channel: 2
  - platform: my9231
    id: blue
    channel: 1
  - platform: my9231
    id: white
    channel: 0
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Lightbulb works perfectly on ESPurna 1.13.5.

anduchs, 2020-02-01:
I can second this bug for my Sonoff B1.
My config was from esphome.io: https://esphome.io/components/output/my9231.html#sonoff-b1-configuration-example

0-15% brightness seems to dim the warm white cannel correctly. After 15% I get a bunch of flickering. Color Temperature does not change anything. Colors are off; blue is purple for example, so not just a channel screw up.

anduchs, 2020-02-07:
Ok, I got it working by setting `bit_depth: 8` on the `my9231` component in config.
@alexiri maybe this solves you problem as well.

closed 2020-02-09 by OttoWinter