esphome/issues #993: esphome unable to download package from repository

mistrals, 2020-01-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio 32bit on raspberry pi3 B+
**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU V1
**Affected component:**

https://esphome.io/guides/getting_started_hassio.html

**Description of problem:**
when i try to compile the firmware that fail due to download library error. In effect checking connection that is ok. throught terminal ping platformio server is ok

**Problem-relevant YAML-configuration entries:**
```
esphome:
  name: temperatura
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "ducks-net"
  password: "*****"
  power_save_mode: none

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Temperatura Fallback Hotspot"
    password: "******"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "****"

ota:
  password: "****"

spi:
  miso_pin: D6
  clk_pin: D5

sensor:
  - platform: max6675
    name: "Living Room Temperature"
    cs_pin: D7
    update_interval: 120s
```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/temperatura.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/temperatura
Processing temperatura (board: nodemcuv2; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Looking for ESPAsyncTCP-esphome library in registry
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Looking for ESPAsyncWebServer-esphome library in registry
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
|-- <SPI> 1.0
Compiling /data/temperatura/.pioenvs/temperatura/src/esphome/components/api/api_connection.cpp.o
Compiling /data/temperatura/.pioenvs/temperatura/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/temperatura/.pioenvs/temperatura/src/esphome/components/api/api_server.cpp.o
Compiling /data/temperatura/.pioenvs/temperatura/src/esphome/components/api/list_entities.cpp.o
In file included from src/esphome/components/api/api_connection.h:7:0,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/api_server.h:19:25: fatal error: ESPAsyncTCP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncTCP.h
*
*********************************************************************

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
*** [/data/temperatura/.pioenvs/temperatura/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from src/esphome/components/api/api_server.cpp:1:0:
src/esphome/components/api/api_server.h:19:25: fatal error: ESPAsyncTCP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncTCP.h
*
*********************************************************************

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
*** [/data/temperatura/.pioenvs/temperatura/src/esphome/components/api/api_server.cpp.o] Error 1
In file included from src/esphome/components/api/list_entities.h:52:0,
                 from src/esphome/components/api/list_entities.cpp:1:
src/esphome/components/api/api_server.h:19:25: fatal error: ESPAsyncTCP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncTCP.h
*
*********************************************************************

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
*** [/data/temperatura/.pioenvs/temperatura/src/esphome/components/api/list_entities.cpp.o] Error 1
========================= [FAILED] Took 11.06 seconds =========================

```

**Additional information and things you've tried:**
➜  ~ ping platformio.org
PING platformio.org (78.46.220.20): 56 data bytes
64 bytes from 78.46.220.20: seq=0 ttl=44 time=65.270 ms
64 bytes from 78.46.220.20: seq=1 ttl=44 time=63.885 ms
64 bytes from 78.46.220.20: seq=2 ttl=44 time=67.233 ms
64 bytes from 78.46.220.20: seq=3 ttl=44 time=56.875 ms

fremut, 2020-01-17:
In logs is written that you dont have internet connection so your compiler unable to download libraries.

mistrals, 2020-01-18:
Really? do you have tried to read the whole post?

fremut, 2020-02-08:
For me sometimes work when i delete all folders (also hidden) with before compiling, libraries and bin files. (Folder has same name as your yaml)

OttoWinter, 2020-02-09:
That is an issue with your Hass.io setup.

There's an issue with Hass.io where containers get bad DNS configs. If you try to execute the ping command in the ESPHome container you will get an error (because of faulty /etc/resolv.conf).

Please check you do not have any add-ons that mess with DNS installed (pi-hole, etc) and see if disabling them helps. If so, the issue is the DNS server blocking the requests.

mistrals, 2020-02-09:
Was an isp issue. 
Solved

closed 2020-02-09 by mistrals