esphome/issues #994: ESPhome (device) reboots when Home Assistant is unavailable?

GoSpursGoNL, 2020-01-17:
Hi, I recently migrated a couple of Sonoff devices from Tasmota to ESPHome (1.14.3).
In general everything is working perfectly.

In the meantime I had some issues with my RPi (hardware related) causing Home Assistant not to run stable and even be down for a couple of hours or days.

I noticed that during that period, the Sonoff devices sometimes (likely) rebooted. I don't understand why that happens.

How did I find out?
- the Sonoff POW has a blue led that switches on when the device boots. I manually turned it off (via HA). All of a sudden, while HA was not available for days, the light was on again
- via the uptime sensor that I activated on the POW
- I configured the Sonoff S20 to turn the relay on when booting. During the days that HA was unavailable when I went to bed the relay was off. When I woke up during the night, all of a sudden it was on. HA couldn't have done that.

Is this a bug or a feature? I would like to prevent this from happening.

fremut, 2020-01-17:
Set reboot_timeout to API yaml configuration: The amount of time to wait before rebooting when no client connects to the API. This is needed because sometimes the low level ESP functions report that the ESP is connected to the network, when in fact it is not - only a full reboot fixes it. Can be disabled by setting this to 0s. Defaults to 15min. more: https://esphome.io/components/api.html

GoSpursGoNL, 2020-01-17:
Perfect!

closed 2020-01-17 by GoSpursGoNL

glmnet, 2020-01-17:
I've just added an entry on the faq regarding this https://esphome.io/guides/faq.html#my-node-keeps-reconnecting-randomly