esphome/issues #996: Uptime sensor counts on wifi connect

sti0, 2020-01-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 and WeMos D1 Mini Pro (with external antenna)

**Affected component:**

Uptime sensor
https://esphome.io/components/sensor/uptime.html

**Description of problem:**
I like to use an ESP at my car to auto open the garage door if I arrive at home. If the car is started and parked on the outside the garage door should not open. For this behavior I like to use the uptime sensor. If the uptime sensor is >= 240 seconds the garage door should open. If it's less nothing happens (= car started). On my test drives the uptime sensor seems to start (new) counting after wifi connect. I drive for 5 minutes (=300 sconds) and come back home. I receive a notification that the ESP is online (published by MQTT) but the uptime sensor is not ready yet. What I can see is that if the car arrives and den ESP connected to wifi the uptime sensor starts from 0. That's not what I'm expected.

**Problem-relevant YAML-configuration entries:**
```
esphome:
  name: car_opener
  platform: ESP8266
  board: d1_mini_pro

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pwd
  fast_connect: true
  reboot_timeout: 0s
  manual_ip:
    static_ip: 192.168.XXX.XXX
    gateway: 192.168.XXX.XXX
    subnet: 255.255.XXX.XXX
  ap:
    ssid: !secret wifi_fallback_ssid
    password: !secret wifi_fallback_pwd

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret esphomeota
  reboot_timeout: 0s
  
ota:
  password: !secret esphomeota

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_user
  password: !secret mqtt_pwd
  client_id: 'car'
  keepalive: 2s
  reboot_timeout: 0s
  discovery: false
  topic_prefix: 'home/garage_opener/car'

sensor:
  - platform: uptime
    name: "car_garage_opener_uptime"
    id: sensor_uptime
    update_interval: 1s
  - platform: wifi_signal
    name: "astra_garage_opener_wifi_strength"
    update_interval: 5s
    state_topic: 'home/garage_opener/car/sensor/wifi_signal/state'

binary_sensor:
  - platform: status
    name: "car_garage_opener"
  - platform: template
    name: uptime_grt240s
    id: uptime_grt240s
    state_topic: 'home/garage_opener/car/binary_sensor/uptime_grt240s/state'
    lambda: |-
      if (id(sensor_uptime).state >= 240) {
        return true;
      } else {
        return false;
      }

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```
=> Because I'm in the car I can't provide debug logs. If I connect the ESP at home it connects instantly to wifi and starts counting. But thats not the behavior as it is when I'm on the ride.

**Additional information and things you've tried:**
[image]

brandond, 2020-01-18:
For the log collection - can you try something basic like unplugging your AP, powering on the ESP at home, and then plugging your AP back in?

sti0, 2020-01-18:
I used my guest wifi to reproduce this issue. I flashed the new credentials with usb and got the log from ESPHomeFlasher. As you can see the uptime sensor counts up than it connects to wifi (I turned guest wifi on) and it starts from beginning. Here's a snippet of the log:

> [09:50:58][I][wifi:193]: WiFi Connecting to 'guestwifi'...
> [09:50:58][D][sensor:092]: 'car_garage_opener_uptime': Sending state 101.83700 s with 0 decimals of accuracy
> [09:50:59][D][sensor:092]: 'car_garage_opener_uptime': Sending state 102.83700 s with 0 decimals of accuracy
> [09:51:00][D][sensor:092]: 'car_garage_opener_uptime': Sending state 103.83700 s with 0 decimals of accuracy
> [09:51:01][W][wifi_esp8266:334]: Event: Disconnected ssid='guestwifi' reason='Probe Request Unsuccessful'
> [09:51:01][W][wifi:451]: Error while connecting to network.
> [09:51:01][D][sensor:092]: 'car_garage_opener_uptime': Sending state 104.83700 s with 0 decimals of accuracy
> [09:51:02][D][sensor:092]: 'car_garage_opener_uptime': Sending state 105.83700 s with 0 decimals of accuracy
> [09:51:03][D][sensor:092]: 'car_garage_opener_uptime': Sending state 106.83700 s with 0 decimals of accuracy
> [09:51:04][D][sensor:092]: 'car_garage_opener_uptime': Sending state 107.83700 s with 0 decimals of accuracy
> [09:51:05][D][sensor:092]: 'car_garage_opener_uptime': Sending state 108.83700 s with 0 decimals of accuracy
> [09:51:06][I][wifi:193]: WiFi Connecting to 'guestwifi'...
> [09:51:06][D][sensor:092]: 'car_garage_opener_uptime': Sending state 109.83700 s with 0 decimals of accuracy
> [09:51:07][D][sensor:092]: 'car_garage_opener_uptime': Sending state 110.83700 s with 0 decimals of accuracy
> [09:51:08][D][sensor:092]: 'car_garage_opener_uptime': Sending state 111.83700 s with 0 decimals of accuracy
> [09:51:10][D][sensor:092]: 'car_garage_opener_uptime': Sending state 113.12600 s with 0 decimals of accuracy
**> [09:51:10][D][sensor:092]: 'car_garage_opener_uptime': Sending state 113.83700 s with 0 decimals of accuracy**
**> [09:51:11][I][wifi:423]: WiFi Connected!**
> [09:51:11][C][wifi:283]:   SSID: 'guestwifi'
> [09:51:11][C][wifi:284]:   IP Address: 192.168.XXX.XXX
> [09:51:11][C][wifi:286]:   BSSID: XXXX
> [09:51:11][C][wifi:287]:   Hostname: 'car_opener'
> [09:51:11][C][wifi:291]:   Signal strength: -42 dB ▂▄▆█
> [09:51:11][C][wifi:295]:   Channel: 6
> [09:51:11][C][wifi:296]:   Subnet: 255.255.XXX.XXX
> [09:51:11][C][wifi:297]:   Gateway: 192.168.XXX.XXX
> [09:51:11][C][wifi:298]:   DNS1: 192.168.XXX.XXX
> [09:51:11][C][wifi:299]:   DNS2: (IP unset)
> [09:51:11][D][wifi:432]: Disabling AP...
> [09:51:18]
> [09:51:18] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
> [09:51:18]
> [09:51:18]wdt reset
> [09:51:18]load 0x4010f000, len 1384, room 16 
> [09:51:18]tail 8
> [09:51:18]chksum 0x2d
> [09:51:18]csum 0x2d
> [09:51:18]v8b899c12
> [09:51:18]~ld
> [09:51:18][I][logger:156]: Log initialized
> [09:51:18][C][ota:364]: There have been 4 suspected unsuccessful boot attempts.
> [09:51:18][I][app:028]: Running through setup()...
> [09:51:18][D][binary_sensor:034]: 'car_garage_opener': Sending initial state OFF
> [09:51:18][C][wifi:033]: Setting up WiFi...
> [09:51:18][I][wifi:193]: WiFi Connecting to 'guestwifi'...
**> [09:51:18][D][sensor:092]: 'car_garage_opener_uptime': Sending state 0.10100 s with 0 decimals of accuracy**
> [09:51:18][D][binary_sensor:036]: 'uptime_grt240s': Sending state OFF
> [09:51:19][D][sensor:092]: 'car_garage_opener_uptime': Sending state 0.90400 s with 0 decimals of accuracy
> [09:51:20][D][sensor:092]: 'car_garage_opener_uptime': Sending state 1.90400 s with 0 decimals of accuracy
> [09:51:21][D][sensor:092]: 'car_garage_opener_uptime': Sending state 2.90400 s with 0 decimals of accuracy
> [09:51:22][I][wifi:423]: WiFi Connected!
> [09:51:22][C][wifi:283]:   SSID: 'guestwifi'
> [09:51:22][C][wifi:284]:   IP Address: 192.168.XXX.XXX
> [09:51:22][C][wifi:286]:   BSSID: XXXX
> [09:51:22][C][wifi:287]:   Hostname: 'car_opener'
> [09:51:22][C][wifi:291]:   Signal strength: -40 dB ▂▄▆█
> [09:51:22][C][wifi:295]:   Channel: 6
> [09:51:22][C][wifi:296]:   Subnet: 255.255.XXX.XXX
> [09:51:22][C][wifi:297]:   Gateway: 192.168.XXX.XXX
> [09:51:22][C][wifi:298]:   DNS1: 192.168.XXX.XXX
> [09:51:22][C][wifi:299]:   DNS2: (IP unset)
> [09:51:22][D][wifi:432]: Disabling AP...
> [09:51:22][C][ota:029]: Over-The-Air Updates:
> [09:51:22][C][ota:030]:   Address: car_opener.local:8266
> [09:51:22][C][ota:032]:   Using Password.
> [09:51:22][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 6 restarts
> [09:51:22][C][api:022]: Setting up Home Assistant API server...
> [09:51:22][C][mqtt:025]: Setting up MQTT...
> [09:51:22][I][mqtt:162]: Connecting to MQTT...
> [09:51:22][D][sensor:092]: 'car_garage_opener_uptime': Sending state 3.98500 s with 0 decimals of accuracy
> [09:51:23][D][sensor:092]: 'car_garage_opener_uptime': Sending state 4.90400 s with 0 decimals of accuracy
> [09:51:24][D][sensor:092]: 'car_garage_opener_uptime': Sending state 5.90400 s with 0 decimals of accuracy
> [09:51:25][D][sensor:092]: 'car_garage_opener_uptime': Sending state 6.90400 s with 0 decimals of accuracy
> [09:51:26][D][sensor:092]: 'car_garage_opener_uptime': Sending state 7.90400 s with 0 decimals of accuracy
> [09:51:27][D][sensor:092]: 'car_garage_opener_uptime': Sending state 8.90400 s with 0 decimals of accuracy

brandond, 2020-01-18:
Ok so yeah, it crashes and resets with a watchdog timeout when trying to disable the fallback AP after connecting to your WiFi. That's why the uptime resets.

Duplicate of https://github.com/esphome/issues/issues/990

sti0, 2020-01-18:
The `captive_portal `integration seems to be the problem. If I remove it, the fallback AP shows up if no wifi is connected and after connecting to the WiFi the uptime sensor still counts. If captive_portal is activated it throws an exception after connecting to WiFi and than starts by 0.

> [10:22:13][I][wifi:193]: WiFi Connecting to 'guestwifi'...
> [10:22:13][D][sensor:092]: 'car_garage_opener_uptime': Sending state 106.86400 s with 0 decimals of accuracy
> [10:22:14][D][sensor:092]: 'car_garage_opener_uptime': Sending state 108.01900 s with 0 decimals of accuracy
> [10:22:15][D][sensor:092]: 'car_garage_opener_uptime': Sending state 108.86400 s with 0 decimals of accuracy
**> [10:22:16][D][sensor:092]: 'car_garage_opener_uptime': Sending state 109.86400 s with 0 decimals of accuracy
> [10:22:16][I][wifi:423]: WiFi Connected!**
> [10:22:16][C][wifi:283]:   SSID: 'guestwifi'
> [10:22:16][C][wifi:284]:   IP Address: 192.168.XXX.XXX
> [10:22:16][C][wifi:286]:   BSSID: XXXX
> [10:22:16][C][wifi:287]:   Hostname: 'car_opener'
> [10:22:16][C][wifi:291]:   Signal strength: -40 dB ▂▄▆█
> [10:22:16][C][wifi:295]:   Channel: 6
> [10:22:16][C][wifi:296]:   Subnet: 255.255.XXX.XXX
> [10:22:16][C][wifi:297]:   Gateway: 192.168.XXX.XXX
> [10:22:16][C][wifi:298]:   DNS1: 192.168.XXX.XXX
> [10:22:16][C][wifi:299]:   DNS2: (IP unset)
> [10:22:16][D][wifi:432]: Disabling AP...
> [10:22:16][C][ota:029]: Over-The-Air Updates:
> [10:22:16][C][ota:030]:   Address: car_opener.local:8266
> [10:22:16][C][ota:032]:   Using Password.
> [10:22:16][C][api:022]: Setting up Home Assistant API server...
> [10:22:16][C][mqtt:025]: Setting up MQTT...
> [10:22:16][I][mqtt:162]: Connecting to MQTT...
**> [10:22:17][D][sensor:092]: 'car_garage_opener_uptime': Sending state 110.86400 s with 0 decimals of accuracy**
> [10:22:18][D][sensor:092]: 'car_garage_opener_uptime': Sending state 111.86400 s with 0 decimals of accuracy
> [10:22:19][D][sensor:092]: 'car_garage_opener_uptime': Sending state 112.86400 s with 0 decimals of accuracy
> [10:22:20][D][sensor:092]: 'car_garage_opener_uptime': Sending state 113.86400 s with 0 decimals of accuracy
> [10:22:21][D][sensor:092]: 'car_garage_opener_uptime': Sending state 114.86400 s with 0 decimals of accuracy
> [10:22:22][D][sensor:092]: 'car_garage_opener_uptime': Sending state 115.86400 s with 0 decimals of accuracy
> [10:22:23][D][sensor:092]: 'car_garage_opener_uptime': Sending state 116.86400 s with 0 decimals of accuracy
> [10:22:24][D][sensor:092]: 'car_garage_opener_uptime': Sending state 117.86400 s with 0 decimals of accuracy
> [10:22:25][D][sensor:092]: 'car_garage_opener_uptime': Sending state 118.86400 s with 0 decimals of accuracy
> [10:22:26][D][sensor:092]: 'car_garage_opener_uptime': Sending state 119.86400 s with 0 decimals of accuracy
> [10:22:26][I][ota:046]: Boot seems successful, resetting boot loop counter.
> [10:22:27][D][sensor:092]: 'car_garage_opener_uptime': Sending state 120.86400 s with 0 decimals of accuracy
> [10:22:28][D][sensor:092]: 'car_garage_opener_uptime': Sending state 121.86400 s with 0 decimals of accuracy
> [10:22:28]
> Serial port closed!

Activating `captive_portal`:

> [10:26:25][D][sensor:092]: 'car_garage_opener_uptime': Sending state 94.24400 s with 0 decimals of accuracy
> [10:26:26][D][sensor:092]: 'car_garage_opener_uptime': Sending state 95.24400 s with 0 decimals of accuracy
> [10:26:27][I][wifi:193]: WiFi Connecting to 'guestwifi'...
> [10:26:28][D][sensor:092]: 'car_garage_opener_uptime': Sending state 97.06900 s with 0 decimals of accuracy
> [10:26:28][D][sensor:092]: 'car_garage_opener_uptime': Sending state 97.24400 s with 0 decimals of accuracy
> **[10:26:29][D][sensor:092]: 'car_garage_opener_uptime': Sending state 98.24400 s with 0 decimals of accuracy**
> **[10:26:30][I][wifi:423]: WiFi Connected!**
> [10:26:30][C][wifi:283]:   SSID: 'guestwifi'
> [10:26:30][C][wifi:284]:   IP Address: 192.168.XXX.XXX
> [10:26:30][C][wifi:286]:   BSSID: XXXX
> [10:26:30][C][wifi:287]:   Hostname: 'car_opener'
> [10:26:30][C][wifi:291]:   Signal strength: -38 dB ▂▄▆█
> [10:26:30][C][wifi:295]:   Channel: 6
> [10:26:30][C][wifi:296]:   Subnet: 255.255.XXX.XXX
> [10:26:30][C][wifi:297]:   Gateway: 192.168.XXX.XXX
> [10:26:30][C][wifi:298]:   DNS1: 192.168.XXX.XXX
> [10:26:30][C][wifi:299]:   DNS2: (IP unset)
> [10:26:30][D][wifi:432]: Disabling AP...
> [10:26:30][C][ota:029]: Over-The-Air Updates:
> [10:26:30][C][ota:030]:   Address: car_opener.local:8266
> [10:26:30][C][ota:032]:   Using Password.
> [10:26:30][C][api:022]: Setting up Home Assistant API server...
> [10:26:30][C][mqtt:025]: Setting up MQTT...
> [10:26:30][I][mqtt:162]: Connecting to MQTT...
> [10:26:30]
> **[10:26:30]Exception (9):
> [10:26:30]epc1=0x4020d7cb epc2=0x00000000 epc3=0x00000000 excvaddr=0x0188019e depc=0x00000000**
[... 35 lines cut ...]
> [10:26:31]3fffff80:  3fffff34 3fff2f74 3fff2d6c feefeffe  
> [10:26:31]3fffff90:  feefeffe feefeffe feefeffe 3fff12ac  
> [10:26:31]3fffffa0:  3fffdad0 00000000 3fff127c 40221654  
> [10:26:31]3fffffb0:  feefeffe feefeffe 3ffe8584 4010086d  
> [10:26:31]<<<stack<<<
> [10:26:31]
> [10:26:31] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> [10:26:31]
> [10:26:31]load 0x4010f000, len 1384, room 16 
> [10:26:31]tail 8
> [10:26:31]chksum 0x2d
> [10:26:31]csum 0x2d
> [10:26:31]v8b899c12
> [10:26:31]~ld
> [10:26:31][I][logger:156]: Log initialized
> [10:26:31][C][ota:364]: There have been 2 suspected unsuccessful boot attempts.
[... cut ...]
> **[10:26:31][D][sensor:092]: 'car_garage_opener_uptime': Sending state 0.10000 s with 0 decimals of accuracy**
> [10:26:31][D][binary_sensor:036]: 'uptime_grt240s': Sending state OFF
> [10:26:31][D][sensor:092]: 'car_garage_opener_uptime': Sending state 0.89300 s with 0 decimals of accuracy
> [10:26:32][D][sensor:092]: 'car_garage_opener_uptime': Sending state 1.89300 s with 0 decimals of accuracy
> [10:26:33][D][sensor:092]: 'car_garage_opener_uptime': Sending state 2.89300 s with 0 decimals of accuracy

closed 2020-12-05 by stale[bot]