esphome/issues #1000: Tuya MCU light toggle

wlatic, 2020-01-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Bass.ii

**ESP (ESP32/ESP8266, Board/Sonoff):**

Esp8266
**Affected component:**
tuya MCU light

**Description of problem:**
When the tuya MCU is used for a 3 way switch (rather than dimmer) a switch toggle happens on device restart/wifi reconnect. If the light is on the toggle is off, if the light is off the toggle is on.

**Problem-relevant YAML-configuration entries:**
```yaml
Will edit with paste from laptop

```

**Logs (if applicable):**

```
N/A

```

**Additional information and things you've tried:**
Tried adjusting the dimmer and switch values without success.

wlatic, 2020-01-22:
Have done testing using:
```
on_boot:
    priority: -1000.1
    then:
    - light.turn_off: bedroomlight
```
When flashing the light while its on the light will then turn off and then come back on just as the log says:
```
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.7.51: timed out). Re-Trying in 1 seconds
INFO Connecting to 192.168.7.51:6053 (192.168.7.51)
```

wlatic, 2020-01-22:
When removing:
```
# Register the Tuya MCU connection
tuya:
```
From the config the toggle no longer happens.

wlatic, 2020-01-22:
Currently put the Switch onto Tasmota which works without the light toggle.

ssieb, 2020-08-07:
I don't understand what you mean.  Can you please provide step by step descriptions of what you're doing and what is happening and what you expect to happen?

wlatic, 2020-12-18:
The problem is the light will toggle on wifi connection and also on the reboot of the switch itself.

This means if you have a scheduled reboot for the wifi in the middle of the night the light will trigger.

ssieb, 2020-12-18:
If you don't want it rebooting when the wifi goes off, then disable the wifi_timeout.  But does your wifi go off for more than 15 minutes?
Rereading the summary, I'm a little confused.  Do you have a device that's both a dimmer *and* a 3-way switch?

wlatic, 2020-12-18:
Rather than not wanting the device not to reboot etc. its the problem of the switch toggling.

Should the light be on and the switch reboots the light goes off, should the light be off and the switch reboots the light goes on.

The element of toggling light state causes issues.

This is just a 3-way switch.

Hopefully example will provide more understanding:

Manually press the switch to toggle the light to off state
reboot the switch via web interface
Switch reboots and while rebooting automatically toggles the state to on (the light is lit).

ssieb, 2020-12-18:
If it's a 3-way switch, then whether the light is on or not would depend on the other switch.  Maybe there could be an option to keep the state from the MCU instead of initializing it at boot.
Btw, in the time since you opened this issue, there is now a Tuya switch that you can use instead of the dimmer.

wlatic, 2020-12-18:
I feel really bad bringing this back up without actually looking through to see this update.

This light setup is working exactly how it should be!

https://esphome.io/components/switch/tuya.html

It was the final device i had which isnt Tuyalocal or ESPhome and now its sorted! Im glad i brought it up again, but embarrassed i didnt see this sooner.

closed 2020-12-18 by wlatic