esphome/issues #999: Warning! You are not connected to the Internet.

antaljani, 2020-01-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
HassOS 3.8
Hass.io 195 on RPi3+

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESPHome 1.14.3 used together with SSH & Web Terminal, Configurator, DcukDNS, TasmoAdmin and Samba add-ons.

**Affected component:**
Sonoff Basic setting, for Sonoff Mini. 

**Description of problem:**
I created a General ESP8266 configuration for Sonoff Mini. 
It was validated and it is valid.
During compile, I getting the error message "Warning! You are not connected to the Internet." 
and the compilation is terminated. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: nappali_vilagitas
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "DIGI-02259068"
  password: "**********"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Nappali Vilagitas"
    password: "ZMoW7wexuPx3"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

```

**Logs (if applicable):**
```
INFO Reading configuration /config/esphome/nappali_vilagitas.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/nappali_vilagitas
Processing nappali_vilagitas (board: esp01_1m; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Looking for ESPAsyncTCP-esphome library in registry
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Looking for ESPAsyncWebServer-esphome library in registry
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/nappali_vilagitas/.pioenvs/nappali_vilagitas/src/esphome/components/api/api_connection.cpp.o
Compiling /data/nappali_vilagitas/.pioenvs/nappali_vilagitas/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/nappali_vilagitas/.pioenvs/nappali_vilagitas/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/nappali_vilagitas/.pioenvs/nappali_vilagitas/src/esphome/components/api/api_server.cpp.o
In file included from src/esphome/components/api/api_server.cpp:1:0:
src/esphome/components/api/api_server.h:19:25: fatal error: ESPAsyncTCP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncTCP.h
*
*********************************************************************

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
In file included from src/esphome/components/api/api_connection.h:7:0,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/api_server.h:19:25: fatal error: ESPAsyncTCP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncTCP.h
*
*********************************************************************

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
*** [/data/nappali_vilagitas/.pioenvs/nappali_vilagitas/src/esphome/components/api/api_server.cpp.o] Error 1
*** [/data/nappali_vilagitas/.pioenvs/nappali_vilagitas/src/esphome/components/api/api_connection.cpp.o] Error 1
```

**Additional information and things you've tried:**
Tried to reboot, and use protection mode off. 
This is a completely fresh install.

antaljani, 2020-01-20:
One more information what I forgot to write down, I turned on WIFI on RPI3+ and I try to use it on WIFI, not wired LAN cable. Also static IP address was implemented for WIFI.

brandond, 2020-01-20:
This is a platformio error. Are you able to successfully run `platformio update`?

antaljani, 2020-01-22:
That’s correct, platformio link also not reachable. 
So I cannot run the platformio update.

antaljani, 2020-01-22:
In the meantime I reinstall the whole stuff on my SD card and created a completely new instance on LAN cable. It is working properly, however I cannot setup WIFI.
I fI follow the guide how to set WIFI and static IP with nmcli commands, I just loosing all connection I have with RPI (both LAN and WIFI). There is no any other chance than image to the SD card again.

brandond, 2020-01-22:
Ok. If there's no issue with ESPHome, I'm going to close this. Feel free to reopen if the error continues once you can get online.

closed 2020-01-22 by brandond