esphome/issues #1002: Cannot build a config for esp32 because of mdns and wrong reference to WiFi.h

dchaplinsky, 2020-01-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
PIP

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32 cam

**Affected component:**
MDNS, WiFi

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32testbed
  platform: ESP32
  board: esp32cam
wifi:
  ssid: XXX
  password: !secret wifi_password

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  safe_mode: True
  password: !secret ota_password
```

**Logs (if applicable):**

```
$ esphome esp32_testbed.yaml run
INFO Reading configuration esp32_testbed.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d esp32testbed
************************************************************************************************************************************************************************************
Obsolete PIO Core v4.0.3 is used (previous was 4.1.0)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
************************************************************************************************************************************************************************************
Processing esp32testbed (board: esp32cam; framework: arduino; platform: espressif32@1.11.0)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <esp8266_mdns>
|   |   |-- <WiFi> 1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling .pioenvs/esp32testbed/src/esphome/core/util.cpp.o
Compiling .pioenvs/esp32testbed/src/main.cpp.o
In file included from /Users/dchaplinsky/.platformio/packages/framework-arduinoespressif32/libraries/ESPmDNS/src/ESPmDNS.h:46:0,
                 from src/esphome/core/util.cpp:20:
/Users/dchaplinsky/.platformio/lib/esp8266_mdns_ID1091/mdns.h:5:25: fatal error: ESP8266WiFi.h: No such file or directory

*********************************************************************
* Looking for ESP8266WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266WiFi.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266WiFi.h
*
*********************************************************************

compilation terminated.
Compiling .pioenvs/esp32testbed/libccc/WiFi/WiFiUdp.cpp.o
Compiling .pioenvs/esp32testbed/lib45e/esp8266_mdns_ID1091/mdns.cpp.o
Compiling .pioenvs/esp32testbed/libbb8/ESPmDNS/ESPmDNS.cpp.o
In file included from /Users/dchaplinsky/.platformio/lib/esp8266_mdns_ID1091/mdns.cpp:2:0:
/Users/dchaplinsky/.platformio/lib/esp8266_mdns_ID1091/mdns.h:5:25: fatal error: ESP8266WiFi.h: No such file or directory

*********************************************************************
* Looking for ESP8266WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266WiFi.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266WiFi.h
*
*********************************************************************

compilation terminated.
Compiling .pioenvs/esp32testbed/libb71/Update/Updater.cpp.o
Archiving .pioenvs/esp32testbed/libFrameworkArduinoVariant.a
Indexing .pioenvs/esp32testbed/libFrameworkArduinoVariant.a
In file included from /Users/dchaplinsky/.platformio/packages/framework-arduinoespressif32/libraries/ESPmDNS/src/ESPmDNS.h:46:0,
                 from /Users/dchaplinsky/.platformio/packages/framework-arduinoespressif32/libraries/ESPmDNS/src/ESPmDNS.cpp:41:
/Users/dchaplinsky/.platformio/lib/esp8266_mdns_ID1091/mdns.h:5:25: fatal error: ESP8266WiFi.h: No such file or directory

*********************************************************************
* Looking for ESP8266WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266WiFi.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266WiFi.h
*
*********************************************************************

compilation terminated.
*** [.pioenvs/esp32testbed/src/esphome/core/util.cpp.o] Error 1
Compiling .pioenvs/esp32testbed/FrameworkArduino/Esp.cpp.o
*** [.pioenvs/esp32testbed/lib45e/esp8266_mdns_ID1091/mdns.cpp.o] Error 1
*** [.pioenvs/esp32testbed/libbb8/ESPmDNS/ESPmDNS.cpp.o] Error 1
```

I've tried manually change `#include <ESP8266WiFi.h>` to `#include <WiFi.h>`in mdns.h but that didn't help

dchaplinsky, 2020-01-20:
I've also tried to perform `platformio update` and install esphome from scratch in a fresh venv — nothing helped

dchaplinsky, 2020-01-20:
`platformio lib --global uninstall esp8266_mdns` helped tho.

OttoWinter, 2020-07-24:
Thanks for the update @dchaplinsky - this does however look like an issue with your platformio install (you installed esp8266_mdns globally before). So I will close this issue unless more people encounter this.

closed 2020-07-24 by OttoWinter