esphome/issues #1003: Request: Energy value for Pzem004-T Pzemac

ASL07, 2020-01-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

https://esphome.io/components/sensor/pzemac.html
https://innovatorsguru.com/wp-content/uploads/2019/06/PZEM-004T-V3.0-Datasheet-User-Manual.pdf

**Description of problem:**
Hello,
The total energy value for this device is not available in esphome. It is stored in a couple of modbus register. Please see doc above.
If we can read every other value, then we should be able to read the total energy one quite easily.
I can find some time to implement it if anyone is kind enough to give me some guidance regarding to what files need modified (apart from pzemac.h and pzemac.cpp)

Thanks 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

rradar, 2020-01-25:
@ASL07 there is a special tracker for feature-requests and this one exists already here:

https://github.com/esphome/feature-requests/issues/457

Looks like the coding is also done and it maybe get's integrated in the next release.

This issue here can be therefor closed

ASL07, 2020-01-25:
@rradar 
Thanks, this is great
Looking forward to the next release then

closed 2020-01-25 by ASL07