esphome/issues #1004: MS5611 calibration errors

truglodite, 2020-01-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hassos 3.8, hassio 0.104.3, pi3b+

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP12F module (esp8266), 5V usb-LM1117 powered

**Affected component:**
MS5611 i2c barometer

**Description of problem:**
Pressure measurements randomly jump up and down by 10 to 30hpa.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: garage
  platform: ESP8266
  board: nodemcuv2
  on_boot:
    then:
      - switch.turn_off: beeper

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Garage Fallback Hotspot"
    password: "password"

captive_portal:

# Enable logging to get dallas address
logger:
  
# Enable HA API
api:
  password: !secret garage_password
# Enable ota
ota:
  password: !secret ota_password

#define pirPin              13     // PIR output
#define dhtPin              12     // DHT sensor data pin (default io12)
#define dallasPin           14     // Dallas sensor data pin
#define sclPin              4      // i2c clock pin
#define sdaPin              5      // i2c data pin
#define beepPin             16     // Beeper signal

dallas:
  - pin: GPIO14
  
i2c:  
  sda: GPIO5
  scl: GPIO4

sensor:
  - platform: dallas
    index: 0
    resolution: 12
    accuracy_decimals: 1
    name: "Garage Temperature"
    id: celsius
    unit_of_measurement: "°F"
    filters:
      - lambda: return x * (9.0 / 5.0) + 32.0; # c to f
    
  - platform: dht
    pin: GPIO12
    model: DHT22
    humidity:
      accuracy_decimals: 0
      name: "Garage Humidity"
    update_interval: 60s
    
  - platform: wifi_signal
    name: "Garage WiFi Signal"
    update_interval: 60s
    
  - platform: ms5611
    address: 0x77
    temperature:
      name: "Garage Baro Temperature"
    pressure:
      name: "Garage Pressure"
      filters:
        - lambda: return x - 8.6; # calibration
        
  - platform: bh1750
    name: "Garage Luminance"
    address: 0x23
    update_interval: 10s

binary_sensor:
  - platform: status
    name: "Garage Status"
    
  - platform: gpio
    pin:
      number: GPIO13
      mode: INPUT
    name: "Garage Motion"
    device_class: motion
    
switch:
  - platform: gpio
    name: "Garage Beeper"
    pin: GPIO16
    id: beeper
    icon: mdi:volume-low
    
  - platform: template
    name: "Garage Beeps 2"
    turn_on_action:
    - switch.turn_on: beeper
    - delay: 50ms
    - switch.turn_off: beeper
    - delay: 50ms
    - switch.turn_on: beeper
    - delay: 50ms
    - switch.turn_off: beeper
    
  - platform: template
    name: "Garage Beeps 3"
    turn_on_action:
    - switch.turn_on: beeper
    - delay: 50ms
    - switch.turn_off: beeper
    - delay: 50ms
    - switch.turn_on: beeper
    - delay: 50ms
    - switch.turn_off: beeper
    - delay: 50ms
    - switch.turn_on: beeper
    - delay: 50ms
    - switch.turn_off: beeper
    
  - platform: template
    name: "Garage Beeps 5"
    turn_on_action:
    - switch.turn_on: beeper
    - delay: 50ms
    - switch.turn_off: beeper
    - delay: 50ms
    - switch.turn_on: beeper
    - delay: 50ms
    - switch.turn_off: beeper
    - delay: 50ms
    - switch.turn_on: beeper
    - delay: 50ms
    - switch.turn_off: beeper
    - delay: 50ms
    - switch.turn_on: beeper
    - delay: 50ms
    - switch.turn_off: beeper
    - delay: 50ms
    - switch.turn_on: beeper
    - delay: 50ms
    - switch.turn_off: beeper

```

**Logs (if applicable):**

I posted graphs displaying the behavior here:
https://community.home-assistant.io/t/ms5611-going-bad-or-i2c-issue/164240/4

**Additional information and things you've tried:**
At first, I thought it was a 4 year old ms5611 that was just going bad. The sensor was running on the same esp device for a very long time with zero problems, using my own arduino code compiled with PIO. After switching to esphome, within a day these errors started to pop up. I swapped in a brand new ms5611, and the issue remains. I have also scoped out the i2c lines and verified decent looking traces (300/350ns rise time for sda/scl). I'm running at 50kHz only, so the 350ns should not be an issue.

I lost the stackexchange link I read yesterday, where some folks were dealing with a very similar problem on an ms5611. Their debugging had not concluded, but they were focusing on the possibility that the ms5611 calibration prom data is not being read correctly. Not sure if this info will be of use, so figured I'd mention that.

truglodite, 2020-01-31:
After lots of patient help from Simon, I ended up giving up on my attempt to get the known working ms5xxx library working using the custom sensor component:
https://community.home-assistant.io/t/help-with-a-custom-sensor/167750/7

That was a path likely destined for failure, since the device also has a bh1750 on the same i2c bus (so likely my usage of Wire.h would bork the i2c that esphome sets up for that). That leaves me with fixing the existing esphome library as the most likely solution. Reading more in the link above, I mention one difference I found between the libraries; esphome delays 100ms after resetting (in the 'setup' function), and ms5xxx delays only 3ms after resetting (in the 'ReadProm' function). For reference here, the working ms5xxx library is located here:

https://github.com/Schm1tz1/arduino-ms5xxx

Unfortunately I'm a newb and running in to problems with my changes to esphome/device/src/esphome sticking when compiling. The readme mentions this, I'm just not there yet and have some learning to do.

truglodite, 2020-02-09:
So I figured out how to get the edited library to compile. Unfortunately reducing the setup delay from 100ms to 3ms did not work. I also tried 4ms without success, as I found another working library using that. In addition, I noticed a 2000 in the calculations that was not casted to float like the rest of the calcs. Changing that of course didn't help. 

Again, I'm not familiar with the i2c comms or the bitshift maths used in the esphome library. I tried implementing the maths using pow() functions instead of bitshifts with no success. The resulting values are way off (like 200x higher than nominal). Here is my failed attempt at recoding calcs:
```
// d2 = raw temp, d1 = raw press
void MS5611Component::calculate_values_(uint32_t d2, uint32_t d1) {
	// Truglodite: adapted calcs from ms5xxx.cpp
	double dT;
	double OFF;
	double SENS;
	unsigned long D1=d1;
	unsigned long D2=d2;

	dT=D2 - double(this->prom_[4]) * pow(2.0,8.0);
	OFF=double(this->prom_[1])*pow(2.0,17.0)+dT*double(this->prom_[3])/pow(2.0,6.0);
	SENS=double(this->prom_[0])*pow(2.0,16.0)+dT*double(this->prom_[2])/pow(2.0,7.0);
	double TEMP=(2000.0+(dT*double(this->prom_[5]))/pow(2.0,23.0));
	double P=(((D1*SENS)/pow(2.0,21.0)-OFF)/pow(2.0,15.0));

	double T2=0., OFF2=0., SENS2=0.;
	if(TEMP<2000.0) {
	  T2=dT*dT/pow(2.0,31.0);
	  OFF2=61.0*(TEMP-2000.0)*(TEMP-2000.0)/pow(2.0,4.0);
	  SENS2=2.0*(TEMP-2000.0)*(TEMP-2000.0);
	  if(TEMP<-1500.0) {
		OFF2+=15.0*(TEMP+1500.0)*(TEMP+1500.0);
		SENS2+=8.0*(TEMP+1500.0)*(TEMP+1500.0);
	  }
	}

	TEMP-=T2;
	OFF-=OFF2;
	SENS-=SENS2;
	P=(((D1*SENS)/pow(2.0,21.0)-OFF)/pow(2.0,15.0));

	ESP_LOGD(TAG, "Got temperature=%0.02f°C pressure=%0.01fhPa", float(TEMP), float(P));

	if (this->temperature_sensor_ != nullptr)
	this->temperature_sensor_->publish_state(float(TEMP));
	if (this->pressure_sensor_ != nullptr)
	this->pressure_sensor_->publish_state(float(P));  // hPa
	this->status_clear_warning();
}
```
I am hitting the end of what my skills can offer on this fix. I suspect either A) something in the way esphome is reading prom values is off timing or something (perhaps a fix outside of the library?), or B) there may be some mishandled overflow in the maths that I haven't located yet.

Dudu67, 2020-05-31:
I do have the exact same issue with this sensor.

closed 2020-10-12 by stale[bot]

Roglogg, 2020-10-25:
I have the same problem with this sensor