esphome/issues #1005: Wifi connection issues. Git bisected.

skarcha, 2020-01-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Ubuntu 18.04, Kubuntu 19.10, several versions of Python, pip, esphome, etc.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266. Board D1 Mini.

**Affected component:**

Wifi

**Description of problem:**

I have an ESP8266 configured as deep sleep with an BME280 and a ds18b20. All is powered by a 18650 battery and a solar panel.

The system works like a charm with v1.13.6, but it fails with v1.14.3 and 'dev'. It takes several minutes to connect to wifi (if connect), so it continues reading sensors, etc. but doesn't begin deep sleep because it can't connect to wifi. Sometimes it connects, enter deep sleep and when wakes up, the same cycle.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: azotea
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  manual_ip:
      static_ip: 192.168.20.151
      gateway: 192.168.20.254
      subnet: 255.255.255.0

deep_sleep:
  run_duration: 1s
  sleep_duration: 3min
  id: deep_sleep_1

mqtt:
  broker: 192.168.20.200
  on_message:
    topic: azotea/ota_mode
    payload: 'ON'
    then:
      - deep_sleep.prevent: deep_sleep_1

logger:

ota:

i2c:

dallas:
  - pin: D3

sensor:
  - platform: dallas
    address: 0x85021563DF47FF28
    name: "Temperatura Placas"

  - platform: adc
    pin: VCC
    name: "VCC pila"

  - platform: bme280
    address: 0x76
    temperature:
      name: "Temperatura ambiente"
    pressure:
[... 1513 lines cut ...]
[V][sensor:013]: 'Temperatura ambiente': Received new state 9.820000
[D][sensor:092]: 'Temperatura ambiente': Sending state 9.82000 °C with 1 decimals of accuracy
[V][sensor:013]: 'Presión': Received new state 1002.808777
[D][sensor:092]: 'Presión': Sending state 1002.80878 hPa with 1 decimals of accuracy
[V][sensor:013]: 'Humedad': Received new state 82.796875
[D][sensor:092]: 'Humedad': Sending state 82.79688 % with 1 decimals of accuracy

[CANCELED BY ME]
```

**Additional information and things you've tried:**

I decided to git bisect and after "some" work I found the last commit that works without problems: 7a895adec9dcf49c245ea2cfabab461d348ba2d7

After it, new scheduler was introduced in this commit: b51cbc420766a008d0ed55973898df5a143a38b3 (with other changes, so more dificult to git bisec and isolate problems). After it, I tried this commit 8db6f3129cb35a9a0fe8c5c4977af5be9521c2db (no logs provided, sorry) and Wifi connection works fine but not MQTT.

BUT, the key is in the next commit, 36f47ade70351e8d5ddbc6ec608b3092427f54bb, where captive portal is introduced, and with that commit starts the connection problems.

I've read some closed issues, like #455, even tried changing the output power, but I'm pretty sure the problem is the code changed to add captive portal. In git bisect I trust.. :-) 

Hope this logs and info helps to solve the problem. If you need more, I'm glad to provide more logs or info.

P.S: I LOVE this software. A great idea and simple to configure and maintain. Congrats and thanks!

P.S.2: I'd suggest do more specific commit, avoid squash unrelated commits and create and merge hotfix branches instead cherry-picking to help with git bisect. Also use `pip freeze` every now and then; I had to edit requirements_test.txt in some commits to force `voluptuous` and `platformio` versions.

abmantis, 2020-03-10:
I was also trying to revert the commit, but indeed it has a lot of changes. It would be good to have smaller commits with individual changes :/

skarcha, 2020-03-11:
Thanks for trying, @abmantis 👍

OttoWinter, 2020-07-24:
So the scheduler that was introduced in the mentioned commit first had an issue (that would only show up in certain configurations, thus it was not caught immediately). See also git history here: https://github.com/esphome/esphome/commits/dev/esphome/core/scheduler.cpp

That probably means the git bisect is tainted (or at least doesn't give accurate result) because the pattern is not `fail, fail, fail, ..., good, good, good`.

Thanks for trying to bisect it. As for the big commits: We squash commits to keep the git history cleaner, obviously that makes git bisecting harder (but it should be possible if you switch to the branch of the PR).

closed 2020-08-07 by stale[bot]