esphome/issues #1006: OTA update failed in TYWE3S/ESP8266-powered switch

mizhgan, 2020-01-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HA in Docker, ESPHome in Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

TYWE3S (it's basically ESP8266 used in Tuya-compatible products, two channel switch in my case)

**Affected component:**

https://esphome.io/components/ota.html

**Description of problem:**
OTA update failed in the beginning of the process.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: parents_room_switch
  platform: ESP8266
  board: esp_wroom_02

wifi:
  ssid: 'ssid'
  password: 'pass'
  manual_ip:
    static_ip: 192.168.8.38
    gateway: 192.168.8.1
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

status_led:
  pin: GPIO16
  
output:
  - platform: gpio
    pin: 4
    id: relay1
  - platform: gpio
    pin: 13
    id: relay2

light:
  - platform: binary
    name: 'light_parents_room_relay1'
    id: light_parents_room_relay1
    output: relay1
    restore_mode: ALWAYS_OFF
  - platform: binary
    name: 'light_parents_room_relay2'
    id: light_parents_room_relay2
    output: relay2
    restore_mode: ALWAYS_OFF

binary_sensor:
  - platform: gpio
    pin: 14
    filters:
      - invert:
      - delayed_on: 20ms
    name: parents_room_button1
    id: parents_room_button1
    on_release:
      then:
        - light.toggle: light_parents_room_relay1
  - platform: gpio
    pin: 12
    filters:
      - invert:
      - delayed_on: 20ms
    name: parents_room_button2
    id: parents_room_button2
    on_release:
      then:
        - light.toggle: light_parents_room_relay2

sensor:
  - platform: wifi_signal
    name: 'parents_room_switch_wifi'
    update_interval: 60s

```

**Logs (if applicable):**

```
INFO Reading configuration /config/parents_room_switch.yaml...
INFO Starting log output from 192.168.8.38 using esphome API
INFO Connecting to 192.168.8.38:6053 (192.168.8.38)
INFO Successfully connected to 192.168.8.38
[16:10:06][I][app:100]: ESPHome version 1.14.3 compiled on Jan 15 2020, 13:39:18
[16:10:06][C][status_led:019]: Status LED:
[16:10:06][C][status_led:020]:   Pin: GPIO16 (Mode: OUTPUT)
[16:10:06][C][wifi:415]: WiFi:
[16:10:06][C][wifi:283]:   SSID: [redacted]
[16:10:06][C][wifi:284]:   IP Address: 192.168.8.38
[16:10:06][C][wifi:286]:   BSSID: [redacted]
[16:10:06][C][wifi:287]:   Hostname: 'parents_room_switch'
[16:10:06][C][wifi:291]:   Signal strength: -70 dB ▂▄▆█
[16:10:06][C][wifi:295]:   Channel: 5
[16:10:06][C][wifi:296]:   Subnet: 255.255.255.0
[16:10:06][C][wifi:297]:   Gateway: 192.168.8.1
[16:10:06][C][wifi:298]:   DNS1: (IP unset)
[16:10:06][C][wifi:299]:   DNS2: (IP unset)
[16:10:06][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'parents_room_button1'
[16:10:06][C][gpio.binary_sensor:016]:   Pin: GPIO14 (Mode: INPUT)
[16:10:06][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'parents_room_button2'
[16:10:06][C][gpio.binary_sensor:016]:   Pin: GPIO12 (Mode: INPUT)
[16:10:06][C][gpio.output:010]: GPIO Binary Output:
[16:10:06][C][gpio.output:011]:   Pin: GPIO4 (Mode: OUTPUT)
[16:10:06][C][gpio.output:010]: GPIO Binary Output:
[16:10:06][C][gpio.output:011]:   Pin: GPIO13 (Mode: OUTPUT)
[16:10:06][C][logger:175]: Logger:
[16:10:06][C][logger:176]:   Level: DEBUG
[16:10:06][C][logger:177]:   Log Baud Rate: 115200
[16:10:06][C][logger:178]:   Hardware UART: UART0
[16:10:06][C][light:177]: Light 'light_parents_room_relay1'
[16:10:06][C][light:177]: Light 'light_parents_room_relay2'
[16:10:06][C][ota:029]: Over-The-Air Updates:
[16:10:06][C][ota:030]:   Address: 192.168.8.38:8266
[16:10:06][C][api:095]: API Server:
[16:10:06][C][api:096]:   Address: 192.168.8.38:6053
[16:10:06][C][wifi_signal.sensor:009]: WiFi Signal 'parents_room_switch_wifi'
[16:10:06][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[16:10:06][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[16:10:06][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[16:10:21][D][ota:072]: Starting OTA Update from 192.168.1.1...
[16:10:21][D][ota:243]: OTA in progress: 0.1%
[16:10:21][W][ota:233]: Error writing binary data to flash: 0 != 536!
[16:10:21][W][ota:276]: Update end failed! Error: ERROR[2]: Flash Erase Failed

[16:10:22][D][sensor:092]: 'parents_room_switch_wifi': Sending state -69.00000 dB with 0 decimals of accuracy

```

```
INFO Reading configuration /config/parents_room_switch.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/parents_room_switch
Processing parents_room_switch (board: esp_wroom_02; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 2MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/parents_room_switch/src/main.cpp.o
Linking .pioenvs/parents_room_switch/firmware.elf
Building .pioenvs/parents_room_switch/firmware.bin
Retrieving maximum program size .pioenvs/parents_room_switch/firmware.elf
Checking size .pioenvs/parents_room_switch/firmware.elf
DATA:    [====      ]  42.6% (used 34872 bytes from 81920 bytes)
PROGRAM: [====      ]  37.1% (used 387232 bytes from 1044464 bytes)
Creating BIN file ".pioenvs/parents_room_switch/firmware.bin" using ".pioenvs/parents_room_switch/firmware.elf"
========================= [SUCCESS] Took 8.02 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.8.38
INFO Uploading /config/parents_room_switch/.pioenvs/parents_room_switch/firmware.bin (391392 bytes)
Uploading: [=                                                           ] 2% 
ERROR Error sending data: [Errno 104] Connection reset by peer
```

**Additional information and things you've tried:**

brandond, 2020-01-22:
You've most likely got the wrong flash size. `esp_wroom_02` has 2mb flash. Most of these Tuya devices only have 1mb. Try with `esp01_1m` instead.

Just out of curiosity, how did you decide to use `esp_wroom_02` to begin with?

mizhgan, 2020-01-22:
According to this datasheet it has 2Mbyte flash memory. https://docs.tuya.com/en/iot/device-development/module/wifi-module/wifie3smodule
But i'll try with `esp01_1m`, thanks.

evadim, 2020-02-29:
Got same issue and there is a thing:
Compiled firmware with **esp_wroom_02**, flash by com port = OK.
Try to update via OTA = fail.
Changed type to **esp01_1m**, flash via OTA = fail. Flash via com works fine.
Now, OTA works fine vith **esp01_1m**
Changed back to **esp_wroom_02**, updated via OTA works fine (did several times, turn of power, etc).

@brandond 
> Just out of curiosity, how did you decide to use esp_wroom_02 to begin with?

I found datasheet above and the only module with 2M flash and similar pinout is **esp_wroom_02**

Side note: how to add new board type?

Chuckame, 2020-03-25:
Hi,

This is working with my tuya switch (2 way) with this config:
```
esphome:
  name: switch_tuya2_1
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "xxx"
  password: "xxx"

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

ota:
```

closed 2020-08-07 by stale[bot]