esphome/issues #1007: status_led component prevents state led to stay on.

Lovely-Maisonette, 2020-01-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

  platform: ESP8266
  board: esp01_1m

Tuya smart plug generic, One LED, one button, Power metering. 
**Affected component:**

https://esphome.io/components/status_led.html
https://esphome.io/components/output/esp8266_pwm.html
https://esphome.io/components/light/monochromatic.html

**Description of problem:**
The plug works as expected if the status_led component is not used
relay and led turn on/off by pressing the button or by turning the entity on/off in hass.io

When I enable  the status_led component and configure it to use the same and only led (same GPIO), after turning the led on/off with the aforementioned methods, the led gets one more action: to restore the status_led status. Example:

if status_led was LOW, and the plug/relay/led was off, by toggle on his happens:
LED turns on/Relay switches on, led turns off
pressing the button again (toggle off) this happens
led turns off (by first turning itself on)

We can remove the relay and button from the problem, sames happens by turning the led on and off with hass.io
turn on: turns on and off
turn off: turns off by turning it firs on (turn on immediately, and the off by transition, 1s) and stays off

by giving the component an Invert: True, it resets to the high position instead.

But this only happens if the brightness is 100%!
If I slide it down to 99% this reset to state_led state does not happen, but this is an ugly workaround, and returns the light to 99%, even if previously set to 50%

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

# Code is intended to be used with plugs with one or two LEDs alike
substitutions:
  gpio_led1:   "13"
  gpio_led2:   "13"
  gpio_relay:  "14"
  gpio_button:  "3"

status_led:
  pin:
    number: $gpio_led2
    inverted: true

output:
  - platform: esp8266_pwm
    id: led2
    pin:
      number: $gpio_led2
      inverted: true
  - platform: esp8266_pwm
    id: led1
    pin:
      number: $gpio_led1
      inverted: true
light:
  - platform: monochromatic
    id: red_led
    name: $fname Red LED
    output: led2
    default_transition_length: 500ms
  - platform: monochromatic
    id: blue_led
    name: $fname Blue LED
    output: led1
    default_transition_length: 500ms
switch:
  # Switch to toggle the relay
  - platform: gpio
    id: relay
    pin: $gpio_relay
  - platform: template
    id: plug
    name: Plug
    lambda: |-
      if (id(relay).state) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
    - light.turn_on:
        id: blue_led
        brightness: 99%
    - switch.turn_on: relay
    turn_off_action:
    - light.turn_off: blue_led
    - switch.turn_off: relay

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

It seems to be working (needs more testing, I hadn't notice a problem there) without the 99% workaround on other similar plugs with two LEDs where the second (red) LED is shared between the state_led component and the off  status (while blue is on)

brandond, 2020-01-22:
The status LED component does not expect to have anything else controlling the output, so I'd say what you're observing is somewhat expected. You'll run into the same thing any time you point multiple components at the same pin.

Lovely-Maisonette, 2020-01-22:
So this would be a Feature Request rather than a bug?
It is exclusively a software implementation, to let a LED be on/off and additionally blink at different speeds or patterns under certain conditions  A quite common feature on the sock firmwares. In any case, this component should not do anything at all if there are no warnings or errors, and do not interfere if other components are changing the LED's properties.

It seems the code is checking and turning this led to the default status_led state permanently (not efficient) if the brightness is 0 or 100

Lovely-Maisonette, 2020-02-01:
Then I will post a feature request instead.
I found a partial workaround, I can let a led blink by checking some internal sensors, in this case wifi.connected (if someone comes here through google)
```
      - if:
          condition:
            not:
              wifi.connected:
          then:
            - light.turn_off:
                id: blue_led
                transition_length: 0s
            - light.turn_on:
                id: red_led
                brightness: 100%
                transition_length: 0s
            - delay: 250ms
```
however I have no way to to alert if there are some warnings, errors, if a firmware is being uploaded, etc.

OttoWinter, 2020-07-24:
Yes, feature request not a bug.

closed 2020-07-24 by OttoWinter