esphome/issues #1010: ERROR Circular dependency when refering another value in payload lambda

pfichtner, 2020-01-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, Board huzzah

**Affected component:**

https://esphome.io/components/mqtt.html#mqtt-publish-action

**Description of problem:**
When accessing the y value (``` id(hmc5883l_y)```) within the lambda of x the code does not compile (```ERROR Circular dependency```)

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: node2
  platform: ESP8266
  board: huzzah

<<: !include .wifi.yaml
<<: !include .mqtt.yaml
<<: !include .management.yaml

i2c:
  sda: SDA
  scl: SCL

sensor:
  - platform: hmc5883l
    address: 0x1E
    field_strength_x:
      id: hmc5883l_x
      name: "Field Strength X"
      on_value:
        - mqtt.publish_json:
            topic: foo
            payload: |-
              root["bar"] = id(hmc5883l_y).state;
    field_strength_y:
      id: hmc5883l_y
      name: "Field Strength Y"
```

**Logs (if applicable):**

```
INFO Reading configuration /config/xxxxx.yaml...
INFO Generating C++ source...
ERROR Circular dependency detected!
```

**Additional information and things you've tried:**

pfichtner, 2020-01-23:
What do I want to achieve? 

Publish all values (x,y,z) of the sensor at once in one JSON message
```
on_value:
  - mqtt.publish_json:
      topic: foobar/all
      payload: |-
        root["x"] = id(hmc5883l_x).state;
        root["y"] = id(hmc5883l_y).state;
        root["z"] = id(hmc5883l_z).state;
```

brandond, 2020-01-23:
From [the docs](https://esphome.io/components/sensor/index.html#on-value):
> In Lambdas you can get the value from the trigger with x.

So in the on_value for `hmc5883l_x` you can avoid the circular reference by simply using `x` instead of `id(hmc5883l_x).state`, and then do as you currently do for the other two values.

However, for the sake of clean code I would probably move the whole thing out to a separate [interval](https://esphome.io/guides/automations.html#interval)

```yaml
interval:
  - interval: 30s
    then:
      - mqtt.publish_json:
          topic: foobar/all
          payload: |-
            root["x"] = id(hmc5883l_x).state;
            root["y"] = id(hmc5883l_y).state;
            root["z"] = id(hmc5883l_z).state;

pfichtner, 2020-01-23:
@brandond thanks for your reply. I will try changing the structure as suggested. 
But I still don't understand the reason for the circular dependency
```

sensor:
  - platform: hmc5883l
    address: 0x1E
    field_strength_x:
      id: hmc5883l_x
      name: "Field Strength X"
      on_value:
        - mqtt.publish_json:
            topic: foo
            payload: |-
              root["bar"] = id(hmc5883l_y).state;
    field_strength_y:
      id: hmc5883l_y
      name: "Field Strength Y"
```
Can you explain where the circular dependency is? Thanks!

brandond, 2020-01-23:
I'm not sure what the scope is on resolving IDs, but I think it's at the platform entity level, so any IDs created within `platform: hmc5883` can't be referenced within it. I haven't actually looked at the code though.

pfichtner, 2020-01-25:
Did it using the suggested `interval`. 
So if the original desired access (access y within lambda of x) is nothing that should be guaranteed to be allowed the issue can be closed.

closed 2020-12-06 by stale[bot]