esphome/issues #1011: YAML anchors undefined in included yaml

DerFlob, 2020-01-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Generic Linux distro, Python 3.8, pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8285-based switch power sockets, Gosund SP111

**Affected component:**

None, general configuration mechanism

**Description of problem:**
I have several switched power sockets (Gosund SP111), which have a HLW8012 built-in to measure output voltage/current/power and so on. For them to show sensible values, they have to be adjusted. I use a `calibrate_linear`-filter for that. 

Since the filter values are the only part the individual plugs' config don't share, I would like to have a base-yaml, which I include in a device-specific yaml and somehow substitute the filter values. The `substitutions` mechanism does not work, as it does not allow to specify lists as its value.

Instead I was trying to use yaml anchors and references to solve the problem. See the (shortened) example configuration. 

**Problem-relevant YAML-configuration entries:**
base.yaml
```yaml
esphome:
  name: ${name}
  platform: ESP8266
  board: esp8285

wifi:
  networks: !secret known_networks

time:
  - platform: sntp

sensor:
  - platform: hlw8012
    sel_pin:
      number: GPIO12
      inverted: true
    cf_pin: GPIO05
    cf1_pin: GPIO04
    change_mode_every: 2
    update_interval: 1s
    voltage:
      name: ${name}_voltage
      unit_of_measurement: V
      accuracy_decimals: 1
      filters:
        - calibrate_linear: *voltage_adjustment
        - delta: 0.2
```

plug00.yaml
```yaml
substitutions:
  name: plug00

.voltage: &voltage_adjustment
  - 709.80157 -> 231
  - 719.60889 -> 234
  - 721.24341 -> 235

<<: !include base.yaml
```

**Logs (if applicable):**

```
$ esphome plug00.yaml compile
INFO Reading configuration plug00.yaml...
ERROR Error while reading config: Invalid YAML syntax:

found undefined alias 'voltage_adjustment'
  in "sp111.yaml", line 66, column 29:
            - calibrate_linear: *voltage_adjustment
```

**Additional information and things you've tried:**
The `name` substitutions works fine, but `voltage_adjustment` is undefined. 
I tried pasting the base.yaml into the plug00.yaml, which works fine and die references are substituted with the values for the `calibrate_linear`-filter.

I don't prefer substitutions or yaml anchors. But having at least one of the mechanisms work, would help quite a bit in reducing duplicated configuration. Right now I have to change ten files, whenever I want to do a change and have them for all sockets.

I had a quick look into the source and it seems currently included yamls, are loaded independent of the context from the original file.

brandond, 2020-01-24:
I have a bunch of them too and I gave up on include and just run them through jinja2 templating to generate device-specific calibration stanzas :(

mweinelt, 2020-07-24:
I'm currently also using jinja2 templating, but it's just sad that his is even necessary. How else does one keep a generic template per model and mix in per device filters?

OttoWinter, 2020-07-24:
The new `packages` feature in the next version should help here.

As for the issue here about YAML anchors not being inserted from included files. Yes that's due to how the parsing works, and there's no solution to it (other than rewriting the entire YAML parser library pyYAML, which will *not* happen). So I will close this issue.

closed 2020-07-24 by OttoWinter