esphome/issues #1012: LEDs flickering when using fastled and display component

RomainD2, 2020-01-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio under Docker on OMV server
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**Affected component:**

FastLED Light and Display Core

**Description of problem:**
I have a problem when using both component at the same time, I have a lot of lag on fastled effect. I also had some freeze of my esp32 when using the touch button to toggle other led on my project, that needed a hard reboot.

When I remove the Display component, fastled effect become smooth.

Is it possible to find a way to use both component at the same time, without having lag on fastled ?

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: lampe_salon
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: "HR"
  password: "RD19marsHF"
  manual_ip:
    static_ip: 192.168.1.65
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Lampe Salon Fallback Hotspot"
    password: "yd8tINOQotEX"

captive_portal:

# Pour pouvoir afficher l'heure
time:
  - platform: homeassistant
    id: horloge
    
# Enable logging
#logger:

# Enable Home Assistant API
api:

ota:

esp32_touch:
  #setup_mode: True

i2c:
  sda: 21
  scl: 22
  scan: True

# A fichiers font à télécharger et importer dans le dossier /config/esphome directement sur HomeAssistant dans Configurator
#font:
#  - file: 'slkscr.ttf'
#    id: font1
#    size: 14
#  - file: 'BebasNeue-Regular.ttf'
#    id: font2
#    size: 40
#  - file: 'arial.ttf'
#    id: font3
#    size: 14

# L'adresse 0x76 est importante, sinon, ne fonctionne pas!
#sensor:
#  - platform: bme280
#    temperature:
#      name: "Lampe Salon Température"
#      oversampling: 16x
#      id: temperature
#    pressure:
#      name: "Lampe Salon Pression"
#      id: pression
#    humidity:
#      name: "Lampe Salon Humidité"
#      id: humidite
#    address: 0x76
#    update_interval: 60s
#  - platform: wifi_signal
#    name: "WiFi Signal Sensor"
#    update_interval: 60s

binary_sensor:
  - platform: esp32_touch
    name: "Lampe Salon BTN 1"
    id: btn_1
    pin: GPIO15
    threshold: 600
    on_multi_click:
    - timing:
        - ON for at most 1s
        - OFF for at most 1s
        - ON for at most 1s
        - OFF for at least 0.2s
      then:
        - logger.log: "Double Clicked"
    - timing:
        - ON for 1s to 50s
        - OFF for at least 0.5s
      then:
        - logger.log: "Single Long Clicked"
        - while:
            condition:
            - binary_sensor.is_on: btn_1
            then:
            - light.dim_relative:
                id: lampe_salon_1
                relative_brightness: 2%
                transition_length: 0.01s
            - delay: 0.01s
    - timing:
        - ON for at most 1s
        - OFF for at least 0.5s
      then:
        - logger.log: "Single Short Clicked"
        - if:
            condition:
            - light.is_on: lampe_salon_1
            then:
            - light.turn_off: lampe_salon_1
            - light.turn_on:
                id: led_btn_1
                brightness: 100%
                red: 0.0
                green: 0.0
                blue: 1.0
        - if:
            condition:
            - light.is_off: lampe_salon_1
            then:
            - light.turn_on: lampe_salon_1
            - light.turn_on:
                id: led_btn_1
                brightness: 100%
                red: 1.0
                green: 1.0
                blue: 1.0
        - if:
            condition:
            - light.is_on: lampe_salon_2
            then:
            - light.turn_off: lampe_salon_2
            - light.turn_on:
                id: led_btn_1
                brightness: 100%
                red: 0.0
                green: 0.0
                blue: 1.0
        - if:
            condition:
            - light.is_off: lampe_salon_2
            then:
            - light.turn_on: lampe_salon_2
            - light.turn_on:
                id: led_btn_1
                brightness: 100%
                red: 1.0
                green: 1.0
                blue: 1.0
  - platform: esp32_touch
    name: "Lampe Salon BTN 2"
    id: btn_2
    pin: GPIO12
    threshold: 600
    on_multi_click:
    - timing:
        - ON for at most 1s
        - OFF for at most 1s
        - ON for at most 1s
        - OFF for at least 0.2s
      then:
        - logger.log: "Double Clicked"
    - timing:
        - ON for 1s to 2s
        - OFF for at least 0.5s
      then:
        - logger.log: "Single Long Clicked"
    - timing:
        - ON for at most 1s
        - OFF for at least 0.5s
      then:
        - logger.log: "Single Short Clicked"
        #- output.turn_on: led1_cold
  - platform: esp32_touch
    name: "Lampe Salon BTN 3"
    id: btn_3
    pin: GPIO4
    threshold: 600
    on_multi_click:
    - timing:
        - ON for at most 1s
        - OFF for at most 1s
        - ON for at most 1s
        - OFF for at least 0.2s
      then:
        - logger.log: "Double Clicked"
    - timing:
        - ON for 1s to 2s
        - OFF for at least 0.5s
      then:
        - logger.log: "Single Long Clicked"
    - timing:
        - ON for at most 1s
        - OFF for at least 0.5s
      then:
        - logger.log: "Single Short Clicked"
        - light.toggle: led_ambiance

# Example configuration entry
remote_transmitter:
  pin: GPIO13
  carrier_duty_percent: 50%

# Individual switches
#switch:
#  - platform: template
#    name: "Panasonic TV Off"
#    turn_on_action:
#      remote_transmitter.transmit_panasonic:
#        address: 0x4004
#        command: 0x100BCBD
      
output:
  - platform: ledc
    pin: GPIO25
    frequency: 20kHz
    inverted: False
    id: led1_warm
  - platform: ledc
    pin: GPIO33
    frequency: 20kHz
    inverted: False
    id: led1_cold
  - platform: ledc
    pin: GPIO26
    inverted: False
    id: led2_warm
    frequency: 20kHz
  - platform: ledc
    pin: GPIO27
    frequency: 20kHz
    inverted: False
    id: led2_cold

light:
  - platform: cwww
    name: "Lampe Salon 1"
    id: lampe_salon_1
    cold_white: led1_cold
    warm_white: led1_warm
    cold_white_color_temperature: 6000 K
    warm_white_color_temperature: 3500 K
  - platform: cwww
    name: "Lampe Salon 2"
    id: lampe_salon_2
    cold_white: led2_cold
    warm_white: led2_warm
    cold_white_color_temperature: 6000 K
    warm_white_color_temperature: 3500 K
  - platform: fastled_clockless
    chipset: WS2811
    pin: GPIO05
    num_leds: 1
    rgb_order: RGB
    name: "Lampe Salon Led BTN 3"
    id: led_btn_3
  - platform: fastled_clockless
    chipset: WS2811
    pin: GPIO16
    num_leds: 1
    rgb_order: RGB
    name: "Lampe Salon Led BTN 2"
    id: led_btn_2
  - platform: fastled_clockless
    chipset: WS2811
    pin: GPIO17
    num_leds: 1
    rgb_order: RGB
    name: "Lampe Salon Led BTN 1"
    id: led_btn_1
  - platform: fastled_clockless
    chipset: WS2811
    pin: GPIO18
    num_leds: 61
    rgb_order: GRB
    id: led_ambiance
    name: "Lampe Salon Ambiance"
    effects:
      - random:
      - strobe:
      - flicker:
      - addressable_rainbow:
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - addressable_flicker:

# L'adresse 0x3C est importante, sinon, ne fonctionne pas!
display:
  - platform: ssd1306_i2c
    model: "SH1106 128x64"
    address: 0x3C
#    lambda: |-
#      // Afficher texte en haut
#      it.printf(64, 0, id(font3), TextAlign::TOP_CENTER, "Lampe Salon");
#      
#      // Print time in HH:MM format
#      it.strftime(0, 60, id(font2), TextAlign::BASELINE_LEFT, "%H:%M", id(horloge).now());
#
#      // Afficher la température
#      if (id(temperature).has_state()) {
#        it.printf(127, 23, id(font3), TextAlign::TOP_RIGHT , "%.1f°C", id(temperature).state);
#      }
      
#      // Afficher l'humidité
#      if (id(humidite).has_state()) {
#        it.printf(127, 60, id(font3), TextAlign::BASELINE_RIGHT , "%.0f%%", id(humidite).state);
#      }

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
Adding and remove the display component. I didn't find any problem with the time component or the temperature sensor component, it is only the display component which give me this issue.

rradar, 2020-01-25:
I think this flickering with a lot's (and heavy) components is somehow well known for the fastled component. One thing you could try is to use the neopixel integration instead. It's supposed to be more robust in this cases as I heard.

gibranZawahra, 2020-01-25:
hello, i can confirm that i have the same problem,the leds lag in animation but only occurs on screen refresh so once every 5s by default depending on screen refresh interval, unfortunatly even the neopixelsbus does ... does anyone knows of any workarounds? esp32 wrover by the way. thanks

gibranZawahra, 2020-01-26:
i think i found the solution the problem is not fastled nor neopixelbus not even the display per say but your i2c frequency it's at 50khz by default but i think that's way too low for the display redraw and takes way too much time, i set to 800khz and am testing stability, you can look at the i2c component doc page on esphome

closed 2020-12-05 by stale[bot]