esphome/issues #1013: WiFi Disconnects (again) - OpenWrt router

borpin, 2020-01-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Compiled in HassIO on Generic install DietPi VM

**ESP (ESP32/ESP8266, Board/Sonoff):**

WEMOS Mini D1

**Affected component:**

Wi-Fi & mhz19

**Description of problem:**
The ESP device continually disconnects from the AP. This can be seen both in the ESPHome logs and the OpenWrt router logs. During the period up to when the OpenWrt router disconnects, data has been received by the sensor on the MQTT topic specified.

**Problem-relevant YAML-configuration entries:**
```yaml
# Example configuration entry
uart:
  rx_pin: D2
  tx_pin: D1
  baud_rate: 9600
  id: uart_co2 

sensor:
  - platform: mhz19
    co2:
      name: "MH-Z19 CO2 Value"
      state_topic: emon/CO2/CO2_4
      retain: false
      discovery: false
    temperature:
      name: "MH-Z19 Temperature"
    update_interval: 30s
    uart_id: uart_co2
    automatic_baseline_calibration: false
```

**Logs (if applicable):**

From ESPHome, the usual Home Assistant disconnects message.

OpenWrt logs
```
Fri Jan 24 07:32:48 2020 daemon.info hostapd: wlan0: STA xx:xx:xx:4a:0a:a3 IEEE 802.11: authenticated
Fri Jan 24 07:32:48 2020 daemon.info hostapd: wlan0: STA xx:xx:xx:4a:0a:a3 IEEE 802.11: associated (aid 15)
Fri Jan 24 07:32:48 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED xx:xx:xx:4a:0a:a3
Fri Jan 24 07:32:48 2020 daemon.info hostapd: wlan0: STA xx:xx:xx:4a:0a:a3 WPA: pairwise key handshake completed (RSN)
Fri Jan 24 07:40:58 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED xx:xx:xx:4a:0a:a3
Fri Jan 24 07:40:58 2020 daemon.info hostapd: wlan0: STA xx:xx:xx:4a:0a:a3 IEEE 802.11: disassociated
Fri Jan 24 07:40:58 2020 daemon.info hostapd: wlan0: STA xx:xx:xx:4a:0a:a3 IEEE 802.11: disassociated
Fri Jan 24 07:40:58 2020 daemon.info hostapd: wlan0: STA xx:xx:xx:4a:0a:a3 IEEE 802.11: disassociated
Fri Jan 24 07:40:58 2020 daemon.info hostapd: wlan0: STA xx:xx:xx:4a:0a:a3 IEEE 802.11: disassociated
Fri Jan 24 07:40:58 2020 daemon.info hostapd: wlan0: STA xx:xx:xx:4a:0a:a3 IEEE 802.11: disassociated
Fri Jan 24 07:40:58 2020 daemon.info hostapd: wlan0: STA xx:xx:xx:4a:0a:a3 IEEE 802.11: disassociated
Fri Jan 24 07:40:58 2020 daemon.info hostapd: wlan0: STA xx:xx:xx:4a:0a:a3 IEEE 802.11: disassociated
Fri Jan 24 07:40:59 2020 daemon.info hostapd: wlan0: STA xx:xx:xx:4a:0a:a3 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Jan 24 07:41:05 2020 daemon.info hostapd: wlan0: STA xx:xx:xx:4a:0a:a3 IEEE 802.11: authenticated
Fri Jan 24 07:41:05 2020 daemon.info hostapd: wlan0: STA xx:xx:xx:4a:0a:a3 IEEE 802.11: associated (aid 15)
```

**Additional information and things you've tried:**

Set OpenWrt to not disconnect on low ack, tried various different combinations of the arduino environment and ESPHome versions.

I do have another WEMOS D1 Mini using ESPHome and the BME680 sensor that does not exhibit this problem of disconnecting.

I have reflashed the problem device with Tasmota, and that does not now exhibit the same problem (i.e. does not disconnect from the router).

rradar, 2020-01-25:
Was this wemos d1 mini flashed with another firmware before or was it virgin?

Could you try a erase_flash and than flash the esphome firmware again?

borpin, 2020-01-25:
> Was this wemos d1 mini flashed with another firmware before or was it virgin?

Virgin.

> Could you try a erase_flash and than flash the esphome firmware again?

I did, multiple times.

rradar, 2020-01-26:
Could you paste the esphome logs when it disconnects? Best with very_verbose logging if possible for you.

borpin, 2020-01-26:
Verbose logging doesn't work with this sensor (or is it VV - can't remember but it did fox me for a while). the error is just the 'API disconnected' error that is regularly reported, nothing else of use. Sadly I have moved the device/sensor to Tasmota (which is rock solid) so I don't now have the ability to generate fresh logs.

I do like ESPHome, the integration with HA and the simple method of configuration is excellent and easy from the add-on, if only the Wi-Fi was more reliable.

andriej, 2020-02-11:
Have exactly the same problems except my nodes loose connectivity to API (worked stable over MQTT) after I've moved to API. I've "Cleared MQTT" before the process. Routers on Openwrt.

closed 2020-12-05 by stale[bot]