esphome/issues #1014: PCF8574 only works when address is 0x20

bradphillips921, 2020-01-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**
When connecting a PCF8574 to a Wemos D1 Mini or NodeMCU you can only use the address 0x20 on the pcf8574. Any other address and the ESP8266 board won't boot. I have tried with all address 0x21-0x27 as well as two different pcf8574 boards, two Wemos D1 Mini boards, and a NodeMCU board.

**Problem-relevant YAML-configuration entries:**
```yaml
# pcf8574 configuration entry
i2c:
  sda: D2
  scl: D1

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
Forum topic I discussed this on
https://community.home-assistant.io/t/need-help-working-with-wemos-d1-mini-and-pcf8574-expansion-board/165995

S-Przybylski, 2020-01-29:
Dear @bradphillips921 
i use seven PCF8574 on a Wemos D1 mini at the same time without any issue on the same pins 4=D2 and 5=D1 like you did!
Here is my configuration:

````
i2c:
  sda: 4
  scl: 5
  scan: True

pcf8574:
  - id: 'pcf8574_hub1'
    address: 0x20
    pcf8575: False

  - id: 'pcf8574_hub2'
    address: 0x21
    pcf8575: False

  - id: 'pcf8574_hub3'
    address: 0x22
    pcf8575: False

  - id: 'pcf8574_hub4'
    address: 0x23
    pcf8575: False

  - id: 'pcf8574_hub5'
    address: 0x24
    pcf8575: False

  - id: 'pcf8574_hub6'
    address: 0x25
    pcf8575: False

  - id: 'pcf8574_hub7'
    address: 0x26
    pcf8575: False

````

randybb, 2020-01-29:
I believe the issue is only in missing pull-up resistors on sda/scl. the internal ones are not enough for multiple devices on i2c bus..

betyar, 2020-02-03:
Did you check where the A0-A2 pins are connected? If they are all to GND, that is 0x20; if A0 is to VCC and the other two are to GND, you should get 0x21.

I also never had any problems using different addresses.

bradphillips921, 2020-02-03:
Yep, I have the A0-A2 pins correct for the addresses.

chalabit, 2020-03-02:
HELP... When I change any of the address jumpers, the device shorts across Vcc and Gnd.   I confimed this multi meter continuity test.   Also happens with the pcf8575 chip.  Am I doing something wrong?

amishv, 2020-06-11:
> HELP... When I change any of the address jumpers, the device shorts across Vcc and Gnd. I confimed this multi meter continuity test. Also happens with the pcf8575 chip. Am I doing something wrong?

Do you mean that when you change any jumper the VCC and Gnd gets short circuit? If yes, can you remove all the jumpers and verify if the pins are not soldered to the voltage rails?

Also, did you try pull up resistance (4.7k ~ 10k) in the SDA&SCL rails?

chalabit, 2020-06-12:
Thanks Amish for your help.

As you suggested, I removed all the jumper and did a continuity test.  Unfortunately, all the rails were solderedto VCC, causing a short.
I returned the defective chips and have been using PCF9685 chips instead.
Thanks again,

Tom

closed 2020-12-05 by stale[bot]