esphome/issues #1015: SIM800l wrong message decoding

Basanon, 2020-01-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
ESPhome 1.14.3, Hassos 3.5, hassio 0.104.2, pi4

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP12, D1 Mini

**Affected component:**
https://esphome.io/components/sim800l.html

**Description of problem:**
Seems to be SIM800l works fine with ESP and HA, as soon I able to see incoming SMS messages in ESPHome node log and in Home Assistant, but..
I able to receive message only once. Second one message didn't show up in a log and HA
To be able to receive new text message I have to put SIM card to a phone, read and delete all new messages (may be just read will be enough, did not test it). Than put it back to SIM800l, start ESP

Also, I see lot of strange 2 years old messages :) Have no idea what is that :)
_[23:03:57][D][sim800l:216]: Unhandled: +CMGL: 41,"REC READ","1064899010367","","18/03/26,21:37:27+32" - 0
[23:03:57][D][sim800l:216]: Unhandled: 962DC79D162803011 - 0_

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp12gsm
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "XXX"
  password: "XXX"
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.1.73
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.1.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp12Gsm Fallback Hotspot"
    password: "XXX"

captive_portal:

# Enable logging
logger:
   baud_rate: 0

ota:

# Enable Home Assistant API
api:
  services:
  - service: send_sms
    variables:
      recipient: string
      message: string
    then:
    - sim800l.send_sms:
        recipient: !lambda 'return recipient;'
        message: !lambda 'return message;'

uart:
  baud_rate: 9600
  tx_pin: TX
  rx_pin: RX

sim800l:
  on_sms_received:
    - lambda: |-
        id(sms_sender).publish_state(sender);
        id(sms_message).publish_state(message);
    - logger.log:
        format: "Received '%s' from %s"
        args: [ 'message.c_str()', 'sender.c_str()' ]

text_sensor:
- platform: template
  id: sms_sender
  name: "Sms Sender"
- platform: template
  id: sms_message
  name: "Sms Message"
```

**Logs (if applicable):**

```
[22:37:14][D][sim800l:216]: Unhandled: OK - 0
[22:37:16][D][sim800l:129]: RSSI: 15
[22:37:17][D][sim800l:174]: Received SMS from: +XXXXXXXX263
[22:37:17][D][sim800l:175]: 04100430
[22:37:17][D][text_sensor:015]: 'Sms Sender': Sending state '+XXXXXXXX263'
[22:37:17][D][text_sensor:015]: 'Sms Message': Sending state '04100430'
[22:37:17][D][main:229]: Received '04100430' from +XXXXXXXX263
```

**Additional information and things you've tried:**
Change TX,RX pin did not help

Basanon, 2020-01-26:
Seems to be I fixed it :) 
My very first, and very short issue post on GitHub

So
After changing bound rate from 9600 to 115200 

```yaml
uart:
  baud_rate: 115200 #9600
  tx_pin: TX
  rx_pin: RX
```
log is changed, much less messages, and I see now my last received SMS, not first one only with multiple strange old messages (I unable to see them on phone)

```[00:58:28][D][sim800l:175]: 962DC79D162803011
[00:58:28][D][text_sensor:015]: 'Sms Sender': Sending state '1064899010367'
[00:58:28][D][text_sensor:015]: 'Sms Message': Sending state '962DC79D162803011'
[00:58:28][D][main:229]: Received '962DC79D162803011' from 1064899010367
[00:58:33][D][sim800l:174]: Received SMS from: 1064899010367
[00:58:33][D][sim800l:175]: 962DC79D162803011
[00:58:33][D][text_sensor:015]: 'Sms Sender': Sending state '1064899010367'
[00:58:33][D][text_sensor:015]: 'Sms Message': Sending state '962DC79D162803011'
[00:58:33][D][main:229]: Received '962DC79D162803011' from 1064899010367
[00:58:43][D][sim800l:129]: RSSI: 14
[00:58:48][D][sim800l:129]: RSSI: 14
[00:58:53][D][sim800l:129]: RSSI: 15
[00:58:58][D][sim800l:129]: RSSI: 15
[00:59:03][D][sim800l:129]: RSSI: 15
[00:59:04][D][sim800l:174]: Received SMS from: +XXXXXXXX263
[00:59:04][D][sim800l:175]: Abcd
[00:59:04][D][text_sensor:015]: 'Sms Sender': Sending state '+XXXXXXXX263'
[00:59:04][D][text_sensor:015]: 'Sms Message': Sending state 'Abcd'
[00:59:04][D][main:229]: Received 'Abcd' from +XXXXXXXX263
[00:59:13][D][sim800l:129]: RSSI: 14
[00:59:18][D][sim800l:129]: RSSI: 14
[00:59:23][D][sim800l:129]: RSSI: 14
[00:59:28][D][sim800l:129]: RSSI: 14
[00:59:33][D][sim800l:129]: RSSI: 14
[00:59:38][D][sim800l:129]: RSSI: 14
[00:59:43][D][sim800l:129]: RSSI: 13
```
However, is there any way to delete old SMS, to see list of them?

glmnet, 2020-01-27:
There’s no much to tell about the logs in debug. 
Setup the logger to verbose and post all the logs that you can

Basanon, 2020-02-01:
A little update
I had issue with this kind of SIM800l module [[image]
](url)
After I made some successful tests with sending SMS from HA, I left my "GSM Modem" project powered on battery only, to test how long it will keep alive (result is 22 hours)
After couple days I got new SIM card and powered project on, this time SIM800l still blinking with old and new SIM. Have no idea what happen, nothing been changed.
Fine, I have a replacement [image] 
But, this time, I got another problem. This module start to response on 9600 bound rate only with changing UART pins from RX\TX to different

Basanon, 2020-02-01:
> Setup the logger to verbose and post all the logs that you can

Actually, after 20 minutes running, all that strange messages in log just gone
Now I see something like "RSSI: 19 dB" only

glmnet, 2020-02-01:
SMS messages are stored in the sim card in older phones, these modules can read messages stored in the card, so the software tries to always remove messages from the sim card, your issues might be related to the software being unable to remove all messages 

I have done all testing with the module in your first picture, they are power hungry, they need a stable supply. usually sending sms or registering to the network failures are sign of poor power supply of the module, ideally about 4v, make sure connections are pretty solid too.

Basanon, 2020-02-01:
Yes, I know that. Thats why I opened that SIM in my old phone and delete all SMS what I found. However, SIM800l showed me that strange messages again.
Never mind, that was just strange log thing, which is gone :)

Yes, that modules pretty hungry, I just notice that with second one, then sending SMS from HA while that thing was on battery, ESP module even rebooted twice :)

Couple days ago, that red module works just fine with the same power supply, 4V and 2A. Now it's not...

Now I'm using blue one, connected directly to battery with stepUp module (this one required 5V, and, as you correctly said, did not registered in network due 3.7V from 18650). 
This module charting battery and providing power for ESM module
[http://mh.nodebb.com/topic/48/mh-cd42-dc-5v-2-1a-mobile-power-diy-board](url)
If I connect SIM800l to power module directly, it start suffer from lack of power, and unable to connect to network even with 5V

glmnet, 2020-02-08:
Closing as seems solved.

closed 2020-02-08 by glmnet