esphome/issues #1017: Base Sensor Configuration not comply with Template Sensor

Heiner1967, 2020-01-28:
As per ESPHome the Base Sensor Configuration allws for example the setting of unit_of_measurement.

This unfortunate does not work for the template sensor.

brandond, 2020-01-30:
Template sensor should extend the base sensor schema. Do you get an error when using it?

brandond, 2020-01-30:
Ah I see, it extends them with the icon and unit specifically unset:
https://github.com/esphome/esphome/blob/dev/esphome/components/template/sensor/__init__.py#L10

Heiner1967, 2020-02-25:
Hi sorry to ask, but what does this now mean. You identified and corrected the code with the link above? And how could i test if it works in my installation or when will it be part of an update? Just to understand how it runs?

However to answer your previous question, yes I get a Error if I try to add the core command
"unit_of_measurement: " to the template Sensor as not available option.

glmnet, 2020-02-28:
Seems like @brandond just confirmed what you reported, now I don't understand why this is this way.

Heiner1967, 2020-02-29:
I'm sorry, my fault tried to define the unit inside the lamda where I set the value. Which does not work. No doing it within the templatesensor definition all is good.

closed 2020-02-29 by Heiner1967