esphome/issues #1018: fallback hotspot not starting on sonoff basic

BelgarionNL, 2020-01-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
[22:28:18][D][wifi:304]: Starting scan...
[22:28:24][D][wifi:319]: Found networks:
[22:28:24][D][wifi:369]: - 'gasten' (FE:EC:DA:8A:30:ED) ▂▄▆█
[22:28:24][W][wifi:374]: No matching network found!
[22:28:29][D][wifi:304]: Starting scan...
[22:28:35][D][wifi:319]: Found networks:
[22:28:35][D][wifi:369]: - 'gasten' (FE:EC:DA:8A:30:ED) ▂▄▆█
[22:28:35][D][wifi:369]: - 'H369A8D46FD' (70:2E:22:8D:46:FD) ▂▄▆█
[22:28:35][W][wifi:374]: No matching network found!
[22:28:40][D][wifi:304]: Starting scan...
[22:28:46][D][wifi:319]: Found networks:
[22:28:46][D][wifi:369]: - 'gasten' (FE:EC:DA:8A:30:ED) ▂▄▆█
[22:28:46][W][wifi:374]: No matching network found!
[22:28:51][D][wifi:304]: Starting scan...
[22:28:57][D][wifi:319]: Found networks:
[22:28:57][D][wifi:369]: - 'gasten' (FE:EC:DA:8A:30:ED) ▂▄▆█
[22:28:57][W][wifi:374]: No matching network found!
[22:29:02][D][wifi:304]: Starting scan...
[22:29:07][I][wifi:115]: Starting fallback AP!
[22:29:07][C][wifi:160]: Setting up AP...
[22:29:07][C][wifi:162]:   AP SSID: 'Sonoff 2 Fallback Hotspot'
[22:29:07][C][wifi:163]:   AP Password: 'YpvXyG4UksDs'
[22:29:08][C][wifi:172]:   IP Address: 192.168.4.1

```

**Additional information and things you've tried:**

reflashing is not working. I have tried 3 sonoff's with 2 different flashers on 2 windows machines but they refuse the flash again so I am stuck getting this portal to work.

fabaff, 2020-02-01:
Works for me with an ESP32

```bash
[16:12:19][I][wifi:115]: Starting fallback AP!
[16:12:19][V][wifi_esp32:038]: Enabling AP.
[16:12:19][V][esp-idf:000]: I (60354) wifi: mode : sta (3c:71:bf:9d:bd:f0) + softAP (3c:71:bf:9d:bd:f1)
[16:12:19][V][esp-idf:000]: I (60361) wifi: Total power save buffer number: 16
[16:12:19][C][wifi:160]: Setting up AP...[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
[16:12:19][V][wifi_esp32:357]: Event: WiFi AP start
[16:12:19]
[16:12:19][C][wifi:162]:   AP SSID: 'Fallback Hotspot'
[16:12:19][C][wifi:163]:   AP Password: 'test1234'
[16:12:20][V][esp-idf:000]: I (61125) wifi: Total power save buffer number: 16
[16:12:20][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
[16:12:20][V][wifi_esp32:357]: Event: WiFi AP start
[16:12:20][V][wifi_esp32:484]: DHCP server IP lease start: 192.168.4.100[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
[16:12:20]
[16:12:20]
[16:12:20][V][wifi_esp32:487]: DHCP server IP lease end: 192.168.4.200
[16:12:20][C][wifi:172]:   IP Address: 192.168.4.1
[16:12:21][D][wifi:304]: Starting scan...
[16:12:23][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
```

```yaml
esphome:
  name: test_cap_portal
  platform: ESP32
  board: lolin32

logger:
  level: VERBOSE

wifi:
  ssid: "doesn_t_exist"
  password: "who_knows"
  ap:
    ssid: "Fallback Hotspot"
    password: "test1234"

captive_portal:

binary_sensor:
  - platform: status
    name: "Status"
```
Works as well for Esp8266

```bash
[16:18:52][I][wifi:115]: Starting fallback AP!
[16:18:52][V][wifi_esp8266:041]: Enabling AP.
[16:18:52]mode : sta(60:01:94:5d:3c:a2) + softAP(62:01:94:5d:3c:a2)
[16:18:52]add if1
[16:18:52]dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
[16:18:52]bcn 100
[16:18:52][C][wifi:160]: Setting up AP...
[16:18:52][C][wifi:162]:   AP SSID: 'Fallback Hotspot'
[16:18:53][C][wifi:163]:   AP Password: 'test1234'
[16:18:53]bcn 0
[16:18:53]del if1
[16:18:53]add if1
[16:18:53]dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
[16:18:53]bcn 100
[16:18:53][V][wifi_esp8266:540]: DHCP server IP lease start: 192.168.4.100
[16:18:53][V][wifi_esp8266:543]: DHCP server IP lease end: 192.168.4.200
[16:18:53][C][wifi:172]:   IP Address: 192.168.4.1
[16:18:53][V][wifi_esp8266:376]: Event: Changed Mode old=STA new=AP+STA
```

OttoWinter, 2020-07-24:
At the end of your logs it *does* start:

```
[22:29:07][C][wifi:160]: Setting up AP...
[22:29:07][C][wifi:162]:   AP SSID: 'Sonoff 2 Fallback Hotspot'
[22:29:07][C][wifi:163]:   AP Password: 'YpvXyG4UksDs'
[22:29:08][C][wifi:172]:   IP Address: 192.168.4.1
```

closed 2020-07-24 by OttoWinter