esphome/issues #1019: MQTT client doesn't work with SSL

oholovko, 2020-01-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

platform: ESP32
board: esp32doit-devkit-v1

**Affected component:**

https://esphome.io/components/mqtt.html

**Description of problem:**
I found, that MQTT does not work with SSL. I tested it with a few MQTT Cloud services and every time I had the same errors (other clients work well with those MQTT services.)

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

esphome:
  name: cont_main_esp32
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: !secret wifi_name_mobile
  password: !secret wifi_password_mobile

# Debug
debug:

# Enable logging
logger:
  level: VERBOSE

ota:
  password: !secret ota_password

mqtt:
  broker: !secret mqtt_broker
  port: !secret mqtt_port
  username: !secret mqtt_user
  password: !secret mqtt_password
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

[21:37:48][C][mqtt:025]: Setting up MQTT...
[21:37:48][D][mqtt:103]: Resolving MQTT broker IP address...
[21:37:48][D][mqtt:136]: Resolved broker IP address to <<IP>>
[21:37:48][I][mqtt:162]: Connecting to MQTT...
[21:37:48][W][AsyncTCP.cpp:950] _poll(): pcb is NULL
[21:37:48][W][mqtt:247]: MQTT Disconnected: TCP disconnected.
[21:37:53][I][mqtt:162]: Connecting to MQTT...
[21:37:53][W][mqtt:247]: MQTT Disconnected: TCP disconnected.
[21:37:58][I][mqtt:162]: Connecting to MQTT...
[21:37:58][W][mqtt:247]: MQTT Disconnected: TCP disconnected.
[21:38:03][I][mqtt:162]: Connecting to MQTT...
[21:38:03][W][AsyncTCP.cpp:950] _poll(): pcb is NULL
[21:38:03][W][mqtt:247]: MQTT Disconnected: TCP disconnected.
[21:38:08][I][mqtt:162]: Connecting to MQTT...
[21:38:08][W][mqtt:247]: MQTT Disconnected: TCP disconnected.
[21:38:13][I][mqtt:162]: Connecting to MQTT...
[21:38:13][W][AsyncTCP.cpp:950] _poll(): pcb is NULL
[21:38:13][W][mqtt:247]: MQTT Disconnected: TCP disconnected.

```

**Additional information and things you've tried:**
I hope, those logs from the MQTT broker will be useful:
```
2020-01-28 21:41:43: OpenSSL Error: error:1408F10B:SSL routines:ssl3_get_record:wrong version number
2020-01-28 21:41:43: Socket error on client <unknown>, disconnecting.
2020-01-28 21:42:05: New connection from <<IP>> on port 28224.
2020-01-28 21:42:05: OpenSSL Error: error:1408F10B:SSL routines:ssl3_get_record:wrong version number
2020-01-28 21:42:05: Socket error on client <unknown>, disconnecting.
2020-01-28 21:42:10: Client connection from <<IP>> failed: error:1408F10B:SSL routines:ssl3_get_record:wrong version number.
2020-01-28 21:42:15: Client connection from <<IP>> failed: error:1408F10B:SSL routines:ssl3_get_record:wrong version number.
```

OttoWinter, 2020-07-24:
That's because SSL is not supported currently by the MQTT stack. That would be more of a feature request then.

closed 2020-07-24 by OttoWinter

Puntoboy, 2021-05-18:
Is this still the case that SSL is not supported? I have a new device I want to connect to a public MQTT broker as it's not on my local network. But I can't get it to connect with the same `[mqtt:247]: MQTT Disconnected: TCP disconnected.` error message.

lifesetter, 2021-10-01:
Hi, is anyone aware about changes about SSL support? Until now i was not able to find updates regarding this. I tried to wire up an MQTT brocker at hivemq to connect clients which are not part of my home network. But... i still got the same error as shown above in the comment.

fi-sch, 2021-11-11:
Hi!
+1 here. Any updates on this issue? Shouldn't it be reopened?

fi-sch, 2021-11-17:
> That's because SSL is not supported currently by the MQTT stack. That would be more of a feature request then.

Is it still the case? I am not sure when [SSL Fingerprints](https://esphome.io/components/mqtt.html#ssl-fingerprints) were introduced but it seems that SLL should work using fingerprints. Nevertheless, it seems to not work as it should.
Should I create a separate issue for this?

yaleh, 2022-01-07:
Actually, I cannot build with mqtt and ssl_fingerprints now.

`
In file included from .piolibdeps\d1\ESPAsyncTCP-esphome\src\ESPAsyncTCP.cpp:84:0:
.piolibdeps\d1\ESPAsyncTCP-esphome\src/tcp_axtls.h:44:25: fatal error: include/ssl.h: No such file or directory
 #include "include/ssl.h"
`

michaelpayne02, 2022-02-02:
This issue needs to be reopened. I remember reading that the ssl.h file is no longer included in some Arduino libraries.