esphome/issues #1021: Weird I2C behaviour on Sonoff RX/TX pins

morph027, 2020-01-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

* Python 3.8.8 virtualenv
* esphome 1.14.3 via pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

* Sonoff Dual R2

```yaml
esphome:
  platform: ESP8266
  board: esp01_1m
```

**Affected component:**

* I2C

**Description of problem:**

I'm trying to use I2C sensor (BH1750) on the Sonoff Dual R2. As the other GPIOs are used already, i wanted to use RX/TX for I2C. I've disabled UART and setup I2C plus the sensor as follows:

**Problem-relevant YAML-configuration entries:**

```yaml
logger:
  baud_rate: 0

i2c:
  sda: 3
  scl: 1

sensor:
  - platform: bh1750
    name: 'Illuminance'
    address: 0x23
```

**Logs (if applicable):**

First boot when powering on:

```
[08:39:37][C][i2c:028]: I2C Bus:
[08:39:37][C][i2c:029]:   SDA Pin: GPIO3
[08:39:37][C][i2c:030]:   SCL Pin: GPIO1
[08:39:37][C][i2c:031]:   Frequency: 50000 Hz
[08:39:37][I][i2c:033]: Scanning i2c bus for active devices...
[08:39:37][I][i2c:040]: Found i2c device at address 0x23
[08:39:38][C][logger:175]: Logger:
[08:39:38][C][logger:176]:   Level: DEBUG
[08:39:38][C][logger:177]:   Log Baud Rate: 0
[08:39:38][C][logger:178]:   Hardware UART: UART0
[08:39:38][C][bh1750.sensor:019]: BH1750 'dual_r2_1 Illuminance'
[08:39:38][C][bh1750.sensor:019]:   Unit of Measurement: 'lx'
[08:39:38][C][bh1750.sensor:019]:   Accuracy Decimals: 1
[08:39:38][C][bh1750.sensor:019]:   Icon: 'mdi:brightness-5'
[08:39:38][C][bh1750.sensor:020]:   Address: 0x23
[08:39:38][E][bh1750.sensor:022]: Communication with BH1750 failed!
```

Soft restart:

```
[08:44:57][C][i2c:028]: I2C Bus:
[08:44:57][C][i2c:029]:   SDA Pin: GPIO3
[08:44:57][C][i2c:030]:   SCL Pin: GPIO1
[08:44:57][C][i2c:031]:   Frequency: 50000 Hz
[08:44:57][I][i2c:033]: Scanning i2c bus for active devices...
[08:44:57][I][i2c:049]: Found no i2c devices!
[08:45:23][W][i2c:070]: Received NACK on transmit of address 0x23
[08:46:23][W][i2c:070]: Received NACK on transmit of address 0x23
[08:46:46][I][ota:046]: Boot seems successful, resetting boot loop counter.
[08:47:23][W][i2c:070]: Received NACK on transmit of address 0x23
[08:48:23][W][i2c:070]: Received NACK on transmit of address 0x23
[08:49:23][W][i2c:070]: Received NACK on transmit of address 0x23
[08:50:23][W][i2c:070]: Received NACK on transmit of address 0x23
```

**Additional information and things you've tried:**

I've tried simple binary sensors on the GPIOs to get the state, it looks like they are pulled up high all the time (FTDI ?)

```yaml
binary_sensor:
  - platform: gpio
    pin: 1
    name: '${device_name}_tx'
  - platform: gpio
    pin: 3
    name: '${device_name}_rx'
```

[image]

brandond, 2020-01-29:
Even if you disable the UART, the esp8266 bootloader outputs a bunch of stuff that cannot be disabled. Not saying that's the cause of your problems, but I wouldn't be surprised if it was interfering with things.

morph027, 2020-01-29:
Jeah, quite sure, read somewhere else that the FTDI chip itself is probably causing something ...

I do have a PIR on 9 right now (wanted to add the illuminance sensor to mimic a traditional PIR), the PIR right now is pulling the input low on motion. Let's try if the PIR works on 1 or 3....

closed 2020-12-05 by stale[bot]