esphome/issues #1022: DeepSleep Mode on Wemos D1 Mini Boards (and/or clones) not working

Basti-Fantasti, 2020-01-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

hass.io in a VM on ProxMox

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini compatible (clone) with ESP8266 12E Module

**Affected component:**

https://esphome.io/components/deep_sleep.html

**Description of problem:**

A lot of D1 mini boards (or clones) available do not wake up after deep sleep, when the deepsleep method of the ESP library is used. (Please check the first link in the additional info section below)

I've tried it on 8 different boards of the same type (D1 mini clone).
Reset and D0 (GPIO16) are bridged as needed, without success. 

The wake up signal is triggered, but the board is not booting up anymore. The board seems to be in a "floating" or zombie mode. The current consumption goes up to approx 4mA but that's it.

Setup() and loop() methods are not executed anymore.

**Problem-relevant YAML-configuration entries:**
```yaml

# Example configuration entry
deep_sleep:
   run_duration: 30s
   sleep_duration: 10min

```

**Logs (if applicable):**

```
No logs available

```

**Additional information and things you've tried:**

here's the original thread on github where the problem is discussed: 

https://github.com/esp8266/Arduino/issues/6007#issuecomment-487470442

and here's the link containing the solution with the code snippet on how to fix that problem:

https://github.com/esp8266/Arduino/issues/6007#issuecomment-578811346

I've tried the code snippet mentioned above in a small plattformio project on the same hardware and it worked flawlessly. The board went to sleep and came back several times without any issues.

w1ll1am23, 2020-07-26:
I can confirm this same issue, I have having this problem on 2 different d1 mini clones.

rmrfus, 2020-09-04:
I'm affected too. Reproduced on 3 different Wemos D1 mini boards.

kokein, 2020-11-08:
It should be written somwhere... 

When using WemosD1 & DHT22 You should avoid GPIO0, GPIO2 and GPIO15 for their special boot properties Using D7/GPIO13 as a communication port worked for me. And remember - you also need to "short" RST and D0 pins to make deep_speer work. I've connected DHT22 to 5V pin additionally - and it work like a charm :)

skipper22hassio, 2021-01-11:
My D1 mini does wake up, but goes to sleep again in the same second even though it's supposed to run for 50s. 

```YAML
  deep_sleep:
  run_duration: 50s 
  sleep_duration: 1min 
  id: deep_sleep_1

[10:17:18][I][deep_sleep:067]: Beginning Deep Sleep
[10:18:23][I][mqtt:202]: MQTT Connected!
[10:18:23][C][deep_sleep:013]: Setting up Deep Sleep...
[10:18:23][I][app:059]: setup() finished successfully!
[10:18:23][I][app:105]: ESPHome version 1.15.3 compiled on Jan 11 2021, 08:56:13
[10:18:23][C][wifi:443]: WiFi:
...
[10:18:23][C][wifi:317]:   Gateway: 192.168.88.1
[10:18:23][C][wifi:318]:   DNS1: 192.168.88.1
[10:18:23][C][wifi:319]:   DNS2: 195.234.128.139
[10:18:23][I][deep_sleep:067]: Beginning Deep Sleep
```

It reacts to MQTT messages to prevent deep sleep and then runs as expected. 
```YAML
  'on_message:
    - topic: d1_lr_temp_2/ota_mode
      payload: 'ON'
      then:
        - deep_sleep.prevent: deep_sleep_1'

[10:22:52][C][wifi:319]:   DNS2: 195.234.128.139
[10:23:56][I][mqtt:202]: MQTT Connected!
[10:23:57][C][deep_sleep:013]: Setting up Deep Sleep...
[10:23:57][I][app:059]: setup() finished successfully!
[10:23:57][I][app:105]: ESPHome version 1.15.3 compiled on Jan 11 2021, 08:56:13
[10:23:57][C][wifi:443]: WiFi:
...
[10:23:57][C][wifi:317]:   Gateway: 192.168.88.1
[10:23:57][C][wifi:318]:   DNS1: 192.168.88.1
[10:23:57][C][wifi:319]:   DNS2: 195.234.128.139
[10:23:57][C][logger:185]: Logger:
[10:23:57][C][logger:186]:   Level: DEBUG
[10:23:57][C][logger:187]:   Log Baud Rate: 115200
[10:23:57][C][logger:188]:   Hardware UART: UART0
      ```

Is that the same Problem? Hope that helps to fix our problem.

bbr111, 2021-04-04:
same problem here. I can confirm the issue.
Is there any workaround?

closed 2021-08-17 by stale[bot]