esphome/issues #1024: light.name is optional despite documentation stating Required

akasma74, 2020-01-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
Sonoff RF

**Affected component:**
[light](https://esphome.io/components/light/binary.html)

**Description of problem:**
If `name` configuration variable of the `light` is omitted, there is no errors (despite it being Required) and the module works BUT there is no light entity in HA.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
output:
  - platform: gpio
    pin: GPIO12
    id: relay

light:
  - platform: binary
    id: ${device_id}
    #name: my light
    output: relay
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2020-01-31:
Ahhh well. This will require a whole docs review I guess. I’m not so familiar with the beginnings but nowadays name is never required and neither is id, but you have to provide at least one of two. I don’t know if that is documented somewhere. 
You can provide name and thus it will appear in home assistant unless you set internal to true.

akasma74, 2020-01-31:
> nowadays name is never required and neither is id, but you have to provide at least one of two

I also thought that. The issue is, to see your `light.whatever` in HA one **must** declare `name: whatever`, just `id: whatever` **is not** enough.

glmnet, 2020-01-31:
Id is used internally, and you definitely want it when you have to reference the component in automations and lambdas. 

`light.turn_on: light_id`

If you don’t provide an id esphome will create one based on the name.

akasma74, 2020-01-31:
> d is used internally, and you definitely want it when you have to reference the component in automations and lambdas.

I agree.

> If you don’t provide an id esphome will create one based on the name.

The question is what should happen if I don't provide a `name`, only an `id`?
Current result is undocumented as well - you'd have no light endity in HA pite having a `light` section in esphome config.

Is that ok?

glmnet, 2020-01-31:
It is well known that it works that way, rest assured that won't change.

It's somewhat documented, all core components documents `internal`  as this:

> internal (Optional, boolean): Mark this component as internal. Internal components will not be exposed to the frontend (like Home Assistant). Only specifying an id without a name will implicitly set this to true.

However I agree with you in that all components documents `name` as required and `id` as optional, and as I said before the real thing is both are optional but you must declare at least one of the two. And as you've figured out already it does not behaves the same. Due to this I usually don't use the `internal` option, however I do use internal components widely.

* If I am using the component internally, e.g. in an automation, I always set the `id`
* If I am viewing the component in HA, I always set the `name`

So, for some components I set `id` only, others `name` only, and others both of them.

For some others components `name` does not make sense and `id` is usually managed internally by esphome, e.g. `i2c` component.

akasma74, 2020-01-31:
> It is well known that it works that way

I'm a new user and I expected it to be documented but I didn't see anything about that.

> It's somewhat documented, all core components documents `internal` as this

I didn't know I need to read the whole set of docs if I only use one core component.

Thanks for explaining your use cases.
I think apart from the documentation bit it's generally a bit unclear how `id`s are used (I hope it **is** documented somewhere and it was just me assuming that esphome `id` and HA (entity_)`id` are related.

Anyway, should I close the issue? ;)

glmnet, 2020-01-31:
We take docs very seriously, however it is understandable you'll not read all docs the first time you use it.

It's not too harsh on new users, if you set name and don't set id (as required and optional documented) you'll end up with what you expect. :)

If you are using any light like binary light or a rgb one, it is expected you are familiar with core light too, this same applies for every component (sensor, output, climate, etc). That's why I said all core components documents `internal`

> p.s how do you do that, by specifying id only?

yes

---
all this I am saying is relevant when using `api` and not using `mqtt` originally esphome did not have `api` and it only had `mqtt`.
I really don't know the details of how this behaves with `mqtt` as I never used mqtt with esphome.

akasma74, 2020-01-31:
No problem, I never used MQTT for ESPHome.
Thanks.

closed 2020-01-31 by akasma74