esphome/issues #1025: [update_interval] is an invalid option for [sensor.dallas]

forhuan, 2020-02-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

dallas
https://esphome.io/components/sensor/dallas.html

**Description of problem:**
update_interval is an invalid option for sensor.dallas.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
sensor:
  - platform: dallas
    address: 0xC90218608FD4FF28
    name: "Water Temperature"
    update_interval: 5s
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

ryrzy, 2020-02-10:
update_interval for dallas should be here:

`# Example configuration entry
dallas:
  - pin: 
        number: GPIOxx
    update_interval: xx
`

xaviersand, 2020-12-05:
i would upvote this problem.

forhuan, 2020-12-15:
@xaviersand
try this
```yaml
# Example configuration entry
dallas:
  - pin: 23
    update_interval: 10s

# Individual sensors
sensor:
  - platform: dallas
    address: 0x1c0000031edd2a28
    name: "Livingroom Temperature"
```

closed 2020-12-15 by forhuan

xaviersand, 2020-12-17:
thank you captian obvious,
but the reason this ticket is there is because the update_itnerval has no impact.

[update_interval] is an invalid option for [sensor.dallas]. Did you mean [internal]?

> 
> 
> @xaviersand
> try this
> 
> ```yaml
> # Example configuration entry
> dallas:
>   - pin: 23
>     update_interval: 10s
> 
> # Individual sensors
> sensor:
>   - platform: dallas
>     address: 0x1c0000031edd2a28
>     name: "Livingroom Temperature"
> ```