esphome/issues #1032: WaveShare e-paper 1.54in v2 display does not work

Basanon, 2020-02-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
ESPhome 1.14.3, Hassos 3.5, hassio 0.104.2, pi4

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32, Dev Module

**Affected component:**
Waveshare E-Paper Display 1.54in **v2**

**Description of problem:**
Display just shows nothing

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32test
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "X"
  password: "X"
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.1.74
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.1.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "X"
    password: "X"

captive_portal:

# Enable logging
logger:
#  level: very_verbose
# Enable Home Assistant API

api:

ota:

# Example configuration entry
spi:
  clk_pin: GPIO18
  mosi_pin: GPIO23

display:
  - platform: waveshare_epaper
    cs_pin: GPIO5
    dc_pin: GPIO17
    busy_pin: GPIO4
    reset_pin: GPIO16
    model: 1.54in
    full_update_every: 5
    lambda: |-
      it.fill(COLOR_OFF);
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
Firs time I thought I have defected display module, since I have two of them, 2.9 and 1.54 size.
2.9 size display work just fine, 1.54 don't even blink. 
Tried many tests from GitHub, nothing happen. 
Then I contacted WaveShare, and they ask me "Does your module have a sticker with v2?" 
Yes, I have it. Then I found this example of code - [https://github.com/waveshare/e-Paper/tree/master/Arduino/epd1in54_V2](url), and finally, my 1.54in display start show picture like here [image]

k-bronowski, 2020-02-06:
I have the same problem. 
I join the request.

kradcomtec, 2020-02-18:
The problem is, for the 1.54 inch BW display gives a new hardware SPI, and this SPI has another code. This is NOT compatible to the old version. We need new code for that :-(

hervema, 2020-02-26:
Hello, I have the same problem and didn't find any relevant code to work with this version 2. The support of WaveShare didn't respond to me also... :( Please :-(

Basanon, 2020-02-26:
> Hello, I have the same problem and didn't find any relevant code to work with this version 2. The support of WaveShare didn't respond to me also... :( Please :-(

> Then I found this example of code - [https://github.com/waveshare/e-Paper/tree/master/Arduino/epd1in54_V2](url), and finally, my 1.54in display start show picture like here

hervema, 2020-02-26:
Yes I got your link and the example but it's a very basic sample showing an image.

What about partial updates and some old functions

1)	Before to init module in full update I used :

//start EPD in full update first
if (epd.Init(lut_full_update) != 0) {

and for partial update I used :

//put EPD in partial update to update local data
if (epd.Init(lut_partial_update) != 0) {
  
epd.init() still exists this function no longer accepts parameter, so what is the way to active full and partial update in V2 ?

2)	Previously the screen was with two memories, the function SetFrameMemory is not present what is the new one, the way to do the same ?

3)	Also DisplayFrame is not present, was is the new one ?

Code example in V1

  /* 
  *  there are 2 memory areas embedded in the e-paper display
  *  and once the display is refreshed, the memory area will be auto-toggled,
  *  i.e. the next action of SetFrameMemory will set the other memory area
  *  therefore you have to set the frame memory and refresh the display twice.
  */

  //load image data
  epd.SetFrameMemory(IMAGE_DATA);
  epd.DisplayFrame();
  epd.SetFrameMemory(IMAGE_DATA);
  epd.DisplayFrame();

4)	To print something in partial update, this code example worked previously, how now do the same in V2 ?

  paint.SetWidth(16);
  paint.SetHeight(200);
  paint.SetRotate(ROTATE_90);
  paint.Clear(UNCOLORED);

  paint.DrawStringAt(0, 0, Message, &Font16, COLORED);
  epd.SetFrameMemory(paint.GetImage(),100,0, paint.GetWidth(), paint.GetHeight());
  epd.DisplayFrame();

Thanks.

hervema, 2020-03-02:
I contacted the support about a sample code to activate and do a partial update, but the only answer I got was "sorry for the inconvenience, we don't have a such code" !!

hervema, 2020-03-04:
Hello,

For information, if you're looking for a serious E-Paper Displays library, use : https://github.com/ZinggJM/GxEPD2

With this library all works perfectly with the **V2** of 1.54in display from waveshare.

I used to demo a Wemos D1, the screen is connected as suggested : // BUSY -> D2, RST -> D4, DC -> D3, CS -> D8, CLK -> D5, DIN -> D7, GND -> GND, 3.3V -> 3.3V

and I used this definition : GxEPD2_BW<GxEPD2_154_D67, GxEPD2_154_D67::HEIGHT> display(GxEPD2_154_D67(/*CS=D8*/ SS, /*DC=D3*/ 0, /*RST=D4*/ 2, /*BUSY=D2*/ 4)); // GDEH0154D67 <== this one is the equivalent of 1.54 V2

Full, partial, rotate image, all is working perfectly.

I had to add also ADAFRUIT_GFX Library

HTH

nodoubtman, 2020-04-11:
Finally i got it working thank you so mucch! :-))

pulimento, 2020-05-26:
> Finally i got it working thank you so mucch! :-))

Hi! Did you use ESPHome, or anything else?

CMarjoribanks, 2020-08-10:
Can I clarify please? The above guide helped me get this working in the Arduino IDE, which is great. *I think* the 1.54in Waveshare v2 (sticker says V2, board says "Rev2.1") epaper screen isn't supported in ESPHome at this stage? 
Arduino > MQQT > HAss is clunky compared to the excellent functionality of ESPHome - the people behind ESPHome have done amazing work and this is a query from a new player and not a moan.

stevenjev, 2020-11-25:
Just want to clarify, this Waveshare 1.54 V2 epaper screen is still unsupported in ESPHome and anyone getting it working have used the GxEPD2 library mentioned above (i.e. not ESPHome)?

Misiu, 2020-12-29:
There is an updated code in the waveshare repo (https://github.com/waveshare/e-Paper/tree/master/Arduino/epd1in54_V2).
anyone had success with the new (v2) display?

velaar, 2021-02-03:
@Misiu I'm using a fork of GxEPD with GxGDEH0154D67.h which works well with this display. It seems it is using SSD1681 controller. 

No luck with ESPHome yet but will try to play with the waveshare component to see if there is a way of implementing it.

nuttytree, 2021-02-16:
@velaar I can work on adding support for this display.  I did commit to working on another Waveshare display first but then I can work on this one.  Might be a week or two.

velaar, 2021-02-16:
@nuttytree thank you so much! You're very kind. Let me know if there is any way I can help with. In my own tests I was able to initialize the display but was unable to clear it. (Tested with Waveshare's code and it works fine). I can publish some very messy code if it helps.

nuttytree, 2021-02-20:
@velaar Looking on the Waveshare web site I am seeing the 1.54in, the 1.54in B, and the 1.54in C (3 color) but not a "V2", is your display actually the B model?

Misiu, 2021-02-20:
@nuttytree  take a look at their wiki: https://www.waveshare.com/wiki/1.54inch_e-Paper_Module
V2 means revision.

velaar, 2021-02-20:
@nuttytree there was a link a couple messages above to waveshere sample code for 154_v2. I guess this is where the name is coming from: https://github.com/waveshare/e-Paper/tree/master/Arduino/epd1in54_V2

cicinovec, 2021-03-31:
Hello, is there any update on this topic? I have now 1.54 v2 display and absolutely no success with it.
@hervema , do you have step by step documentation how you got the GxEPD2 working?

Jasper-Hoek, 2021-04-01:
Yeah I second @cicinovec his question! @nuttytree did you or @velaar have a look to implement the 1.54inchV2 into esphome?

nuttytree, 2021-04-01:
I have not had a chance to work on this, been busy with other things.  Hoping to get back to this soon.

willem-pennings, 2021-04-09:
To those who are having issues getting their display to work at all. I have a working code example to get this display (1.54in v2) running with the Adafruit EPD Arduino library which is available [here](https://github.com/adafruit/Adafruit_EPD).

The Adafruit EPD library is very easy to use and leverages the Adafruit GFX library which contains a lot of drawing functionality. I will post a piece of example code below, in which you only need to modify the pin numbers. I tested this with an ESP32 Pico kit but it should also work with regular Arduino boards as long as they have enough RAM. I simplified the example to leave out some demo effects.

`/***************************************************
  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.
  MIT license, all text above must be included in any redistribution
 ****************************************************/

#include "Adafruit_ThinkInk.h"

#define EPD_CS      33
#define EPD_DC      27
#define SRAM_CS     -1 // Set to -1 to not use SRAM chip
#define EPD_RESET   4 
#define EPD_BUSY    14 
#define EPD_SID     32 // a.k.a. as MOSI ("Master Out Slave In") or DIN on the Waveshare module
#define EPD_CLK     25
#define EPD_MISO    -1 // "Master In Slave Out" a.k.a. data out, not used for this display

ThinkInk_154_Mono_D67 display(EPD_SID, EPD_CLK, EPD_DC, EPD_RESET, EPD_CS, SRAM_CS, EPD_MISO, EPD_BUSY);

void setup() {
  Serial.begin(115200);
  while (!Serial) { delay(10); }
  Serial.println("Adafruit EPD full update test in mono");
  display.begin(THINKINK_MONO);
  display.setRotation(0);
}

void loop() {
  Serial.println("Banner demo");
  display.clearBuffer();
  display.setTextSize(3);
  display.setCursor((display.width() - 180)/2, (display.height() - 24)/2);
  display.setTextColor(EPD_BLACK);
  display.print("Monochrome");
  display.display();
  delay(2000);
  
  Serial.println("B/W rectangle demo");
  display.clearBuffer();
  display.fillRect(display.width()/2, 0, display.width()/2, display.height(), EPD_BLACK);
  display.display();
  delay(2000);
}`

The library also provides partial update functionality, but I think the library does not support partial update without an SRAM chip. The Waveshare modules do not contain such SRAM chips. I tried it and though the screen did update only a part of the screen, the pixels were displaying garbage data (the example that Adafruit provides increments a counter). I'll try and see if I can get this library to do partial updates without the SRAM chip.

Hope this helps anyone in getting their display to work and perhaps it helps those who are working to fix the ESPHome implementation.

comatose-tortoise, 2021-04-14:
What's the status on this? As far as I understand, the solutions posted here have nothing to do with ESPHome?

Jasper-Hoek, 2021-05-28:
> Here's a link to the [datasheet for that V2 display](https://www.waveshare.com/w/upload/e/e5/1.54inch_e-paper_V2_Datasheet.pdf), in case someone wants to take a stab at updating the [component](https://github.com/esphome/esphome/tree/dev/esphome/components/waveshare_epaper).

It should just be added by somebody just like the other second versions of epaper displays by waveshare!

geoffdavis, 2021-05-29:
> It should just be added by somebody just like the other second versions of epaper displays by waveshare!

Done. See esphome/esphome#1843

sconkdk, 2021-05-30:
Thanks works for me.

Jasper-Hoek, 2021-05-31:
> Done. See [esphome/esphome#1843](https://github.com/esphome/esphome/pull/1843)

Cheers! Great work, thank you for adding this.

closed 2021-06-10 by jesserockz

Jasper-Hoek, 2021-08-19:
There appears to be some problem with my waveshare 1.54v2, when I update and upload my yaml file I get the following error:

[E][waveshare_epaper:119]: Timeout while displaying image!

@geoffdavis do you have any suggestion where this error might come from? See below the yaml part of the waveshare screen (yes I did thoroughly check the pin numbers):

```
spi:
  clk_pin: 32
  mosi_pin: 25

display:
  - platform: waveshare_epaper
    cs_pin: 33
    dc_pin: 27
    busy_pin: 4
    reset_pin: 14
    model: 1.54inv2
    full_update_every: 30
    lambda: |-
      it.line(1, 1, 5, 5);