esphome/issues #1033: can't use GPIO16 as binary sensor

betyar, 2020-02-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

esphome command line

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp8266-12f module

**Affected component:**

https://esphome.io/index.html#binary-sensor-components

**Description of problem:**

I am trying to use GPIO16 as a binary sensor to switch a led on and off. I have similar configurations on other pins so I have no others available (I even use a PCF8574 to expand the amount of pins I can use). Unlike all other pins, I can't get it to work. I tried every possible combination: the best I can do is that the led will switch on, and then it stays on (other pins still work meanwhile). I know GPIO16 has a special function, that it can be tied to the EN pin and used when necessary to bring the module out of deep sleep. Does this mean GPIO16 can't be used for anything else? If it can't be used for input can it be used for output?

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:
  - platform: gpio
    name: "Button 1"
    id: button_1
    pin:
      number: GPIO16
      mode: INPUT_PULLUP
      inverted: True
    on_press:
        - switch.turn_on: led_green
    on_release:
        - switch.turn_off: led_green
```

**Logs (if applicable):**

```
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from esp12f/debug
INFO Connected to MQTT broker!
[20:26:18][I][ota:267]: OTA update finished!
[20:26:29][I][mqtt:202]: MQTT Connected!
[20:26:29][I][app:058]: setup() finished successfully!
[20:26:29][I][app:100]: ESPHome version 1.14.3 compiled on Feb  3 2020, 20:26:01
[20:26:29][C][status_led:019]: Status LED:
[20:26:29][C][status_led:020]:   Pin: GPIO15 (Mode: OUTPUT, INVERTED)
[20:26:30][C][wifi:415]: WiFi:
[20:26:30][C][wifi:283]:   SSID: 'betyar0'
[20:26:30][C][wifi:284]:   IP Address: 192.168.1.95
[20:26:30][C][wifi:286]:   BSSID: 00:23:CD:11:B1:9C
[20:26:30][C][wifi:287]:   Hostname: 'esp12f'
[20:26:30][C][wifi:291]:   Signal strength: -56 dB ▂▄▆█
[20:26:30][C][wifi:295]:   Channel: 6
[20:26:30][C][wifi:296]:   Subnet: 255.255.255.0
[20:26:30][C][wifi:297]:   Gateway: 192.168.1.1
[20:26:30][C][wifi:298]:   DNS1: (IP unset)
[20:26:30][C][wifi:299]:   DNS2: (IP unset)
[20:26:30][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Button 1'
[20:26:30][C][gpio.binary_sensor:016]:   Pin: GPIO16 (Mode: INPUT_PULLUP, INVERTED)
[20:26:30][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Button 2'
[20:26:30][C][switch.gpio:042]: GPIO Switch 'led_green'
[20:26:30][C][mqtt.switch:038]: MQTT Switch 'led_red': 
[20:26:30][C][mqtt.switch:039]:   State Topic: 'esp12f/switch/led_red/state'
[20:26:30][C][mqtt.switch:039]:   Command Topic: 'esp12f/switch/led_red/command'
[20:26:30][C][mqtt.binary_sensor:018]: MQTT Binary Sensor 'Button 1':
[20:26:30][C][mqtt.binary_sensor:019]:   State Topic: 'esp12f/binary_sensor/button_1/state'
[20:26:30][C][mqtt.binary_sensor:018]: MQTT Binary Sensor 'Button 2':
[20:26:36][D][binary_sensor:036]: 'Button 1': Sending state ON
[20:26:36][D][switch:029]: 'led_green' Toggling ON.
[20:26:36][D][switch:045]: 'led_green': Sending state ON
[20:28:22][I][ota:046]: Boot seems successful, resetting boot loop counter.

```

**Additional information and things you've tried:**

I've tried all possible combinations of INPUT_PULLUP and INPUT_PULLDOWN_16 together with on_state switch.toggle and on_press/on_release as well as inverted true/false. Either the led would not switch on at all or would switch on and always stay on. The log would not show any entries after that. The config file above is the best combination I could get (switch on then stays on).

brandond, 2020-02-04:
https://www.esp8266.com/wiki/doku.php?id=esp8266_gpio_pin_allocations

GPIO16 can't do pwm, interrupts, and a few other things. Any reason you can't pick a different pin?

glmnet, 2020-02-04:
According to esphome docs https://esphome.io/devices/esp8266.html the pin only has a weak pull down and no pull up can be configured, so you might try to high side switch it, i.e. connect it to +3.3v to turn the input on and disconnect (or connecto to 0v) to turn it off.

glmnet, 2020-02-04:
It works well as an output for me, there are some esp8266 boards like the one in the esphome docs https://esphome.io/devices/nodemcu_esp8266.html which have a LED connected to the GPIO16 (D0) and you can easily turn it on and off using a light or switch component.
I would try to use it as output only, specially if it already has a LED there as it might interfere with an input signal.

betyar, 2020-02-04:
Many thanks @brandond and @glmnet for your informative comments. Unfortunately all my pins are allocated so it would have been nice if I could have used GPIO16 as an input because the wire would run parallel to others (I am using a simple esp8266-12F module with 2mm pins soldered on). Wiring is an issue since the 2mm wires are in pairs. However, since it's clear that I can't use it as originally planned I was able to workaround the problem by using GPIO16 as the status led (works perfectly) and shifting around some components.

closed 2020-02-04 by betyar