esphome/issues #1034: TM1637 Platform Not Found on Dev 1.15.0

secretanybody, 2020-02-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio, but also verified by a fellow esphome discord member using CLI.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32/ESP8266

**Affected component:**

TM1637 - https://next.esphome.io/components/display/tm1637.html

**Description of problem:**
Using the ESPHome-DEV (1.15.0-dev) environment, I am trying to add the TM1637 component to my yaml for an esp32 or esp8266, but I am unable to use it. When pasting the example code below into my .yaml, the platform comes up as 'Not Found'. 
I have tried this one a couple instances of ESPHome-Dev 1.15.0, and all show it as not found. 
I can confirm that other 'new' components do work, so that should confirm i am in fact on Dev, for example slow_pwm, qmc58831 are found. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: spitest
  platform: ESP32
  board: nodemcu-32s

...

display:
    platform: tm1637
    id: tm1637_display
    clk_pin: D6
    dio_pin: D5
    lambda: |-
      it.print("0123");

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```
**Additional information and things you've tried:**
Confirmed it on HomeAssistant, as well as CLI (Discord Member confirmed)
Tried other new components, those do work, only this component is 'Not Found'

ssieb, 2020-02-04:
The problem is that there is no `__init__.py` file in the `components/tm1637` directory.

glmnet, 2020-02-04:
Hi, I pushed that component recently.
sb correct me if I am wrong, but I believed this __init__.py was needed only for Python 2 and we are using Python 3 now?
Actually the test are compiling it no problem. If it's needed for some reason I can quickly add it.

ssieb, 2020-02-04:
`__init__.py` is for any Python.  How does the testing work?  Does it use the installer or does it run from in the tree directly.  The problem is that if the `__init__.py` file doesn't exist, then the installer doesn't copy the other `display.py` file either, which is the real reason compiling fails.

closed 2020-02-05 by glmnet