esphome/issues #1035: ESPHOME and NGINX HA SSL proxy addon don't play nicely

dedepene, 2020-02-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

hassio running on Ubuntu 18.04.2 LTS
**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**
ESPHome loads the web UI but whatever i click from there I get "terminated process" and nothing happens. Getting these in the meantime:

```
2020-02-04 16:10:28 WARNING (MainThread) [aiohttp.websocket] websocket connection is closing.
2020-02-04 16:10:57 WARNING (MainThread) [aiohttp.websocket] websocket connection is closing.
2020-02-04 16:11:03 WARNING (MainThread) [aiohttp.websocket] websocket connection is closing.
```

```
20-02-04 14:00:07 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:00:09 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:00:13 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:00:15 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:00:19 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:00:21 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:00:25 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:00:27 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:00:31 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:00:33 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:00:37 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:00:39 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:00:43 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:00:45 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:00:49 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:00:51 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:00:55 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:00:57 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:01:01 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:01:03 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace
20-02-04 14:01:07 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status', url='http://172.30.32.1:62520/ace 
```

from the ESPHOME addon log:

```
WARNING 403 GET /ace (0.0.0.0) 3.11ms
WARNING 403 GET /ace (0.0.0.0) 2.01ms
WARNING 403 GET /ace (0.0.0.0) 2.61ms
WARNING 403 GET /ace (0.0.0.0) 1.50ms
WARNING 403 GET /ace (0.0.0.0) 1.81ms
WARNING 403 GET /ace (0.0.0.0) 2.45ms
```

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

dedepene, 2020-02-10:
seriously? am i the only one with this issue?

when trying to interact with esphome addon over ssl connection i keep getting "terminated process" and this is what shows up in the nginx proxy addon log:

```
2020/02/10 20:35:09 [error] 34#34: *122547 recv() failed (104: Connection reset by peer) while proxying upgraded connection, client: 192.168.1.1, server: domain.duckdns.org, request: "GET /api/hassio_ingress/CL8o3ozNBPdXQ-NmyQnYZlWXbf9wqTPqWYvxgVPsSz8/ace HTTP/1.1", upstream: "http://172.30.32.1:8123/api/hassio_ingress/CL8o3ozNBPdXQ-NmyQnYZlWXbf9wqTPqWYvxgVPsSz8/ace", host: "domain.duckdns.org:8124"
2020/02/10 20:35:14 [error] 34#34: *122553 recv() failed (104: Connection reset by peer) while proxying upgraded connection, client: 192.168.1.1, server: domain.duckdns.org, request: "GET /api/hassio_ingress/CL8o3ozNBPdXQ-NmyQnYZlWXbf9wqTPqWYvxgVPsSz8/ace HTTP/1.1", upstream: "http://172.30.32.1:8123/api/hassio_ingress/CL8o3ozNBPdXQ-NmyQnYZlWXbf9wqTPqWYvxgVPsSz8/ace", host: "toranaga.duckdns.org:8124"
```

DaAwesomeP, 2020-03-18:
I am having this issue as well. Is this the same as #372?

KiLLeRRaT, 2020-03-25:
I also have this issue but using IIS Reverse Proxy. I think it is related to https://github.com/esphome/issues/issues/372 yes.

jodur, 2020-05-24:
I just switched recently to NGINX because of the latest update concerning internal and external adress and noticed the black terminal screens in ESPHOME Dashboard without any output. Indeed only the sessions that are proxified apears to have this problem. Accessing tough ingress with normal http adres of the dashboard directly don't suffer this problem. It appears to be an old problem without any solution yet?

billchurch, 2020-06-02:
Same here, /ace and /logs result in 403 error on the esphome addon logs
```
WARNING 403 GET /logs (0.0.0.0) 0.31ms
WARNING 403 GET /ace (0.0.0.0) 0.30ms
```
Hassio v 0.110.3
Esphome Addon v 1.14.3

alex18881, 2020-09-23:
Hello every one. I'm using the docker installation of Homeassistant core and a separate ESPHome installation in docker as well. As a proxy I use NGiNX to make everything be available over SSL. Until now I've experienced the same issue. But thanks to @zvldz's  [comment](https://github.com/esphome/issues/issues/372#issuecomment-521621395) now I have solution of this issue for me. May be it'll help someone.
ESPHome uses the tornado.websocket which requires origin to be checked. As it says in comment in code of check_origin method in tornado documentation [here](https://www.tornadoweb.org/en/stable/_modules/tornado/websocket.html#WebSocketHandler.check_origin):
```
    # Check to see that origin matches host directly, including ports
    return origin == host
```
So the solution of this issue for me is to add **port** to the `host` header in nginx configuration. Here's the location section for ESPHome from my config:
```
location /esphome-web/ {
	proxy_pass	http://esphomeupstream/;
		
	proxy_set_header Host		$host:$server_port;

	proxy_http_version 1.1;
	proxy_set_header X-Real-IP $remote_addr;
	proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection "upgrade";

	proxy_set_header X-Hassio-Ingress YES;
	proxy_redirect off;
}
```

mario-tux, 2020-09-23:
Ok, I can confirm that manually adding the port used in main URL of hassio (a higher non standard port un my case) in the `proxt_set_header` specification, the esphome addon works using ingress. Now I have to figure out how to insert this custom configuration in the standard nginx proxy addon of HassOS. 

@alex18881, thanks for the hint!

mario-tux, 2020-09-23:
Ok, I've found a "clean way" to fix the use of ingress with esphome using NGINX addon jointly with a non-standard public port (instead of the standard 443 https one), `12345` in the example below.
I added a file `/share/nginx_proxy_default_fix_ingress.conf`  in the hassio file-system with the following content:
```
location /api/hassio_ingress {
    proxy_pass http://homeassistant.local.hass.io:8123/api/hassio_ingress;
    proxy_set_header Host $host:12345;
    proxy_redirect http:// https://;
    proxy_http_version 1.1;
    proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection $connection_upgrade;
}
```
You have to customize the port `12345` in your case: the rest should be fine in any other scenario. 

In the addon configuration tab in the HA interface you have to set the `customize.activate` flag; the configuration should looks like this:
```
domain: <YOUR PUBLIC DOMAIN WITHOUT PORT>
certfile: fullchain.pem
keyfile: privkey.pem
hsts: max-age=31536000; includeSubDomains
cloudflare: false
customize:
  active: true
  default: nginx_proxy_default*.conf
  servers: nginx_proxy/*.conf
```
I hope this help someone with this long lasting problem.

rccoleman, 2020-10-24:
I can confirm that this works for me.  Thanks!

I'm using the Nginx Proxy Manager to manage my hosts and ended up creating `server_proxy.conf` in `data/nginx/custom` with the following content:

```
location /api/hassio_ingress {
    proxy_pass http://192.168.1.169:8123/api/hassio_ingress;
    proxy_set_header Host $host:8444;
    proxy_redirect http:// https://;
    proxy_http_version 1.1;
    proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection "upgrade";
}
```

`192.168.1.169` is my local host and `8444` is the public port.

akennerly, 2020-11-10:
I'm saw the same issue with the ESPhome add-on.  Same log errors. Same "process terminated" symptom when NGINX is front-ending the TLS for HA.

I took @mario-tux 's config but had trouble at first, I still saw the same "Process terminated" issue after restarting NGINX.
I'm was using standard port 443 on my NGINX add-on. What I failed to do at first was set the ``` proxy_set_header ``` port to the ports used by the HomeAssistant clients and not the port NGINX was using for TLS termination which was 443.  

```
domain: ha.example.com
certfile: fullchain.pem
keyfile: privkey.pem
hsts: max-age=31536000; includeSubDomains
cloudflare: false
customize:
  active: true
  default: nginx_proxy_default*.conf
  servers: nginx_proxy/*.conf
```
```
location /api/hassio_ingress {
    proxy_pass http://homeassistant.local.hass.io:8123/api/hassio_ingress;
    proxy_set_header Host $host:<port your HA client points too>;
    proxy_redirect http:// https://;
    proxy_http_version 1.1;
    proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
    proxy_set_header Upgrade $http_upgrade;
    proxy_set_header Connection $connection_upgrade;
}
```

ErikThorsell, 2021-01-09:
In my case, the problem was that I had `/api/websocket` as my location. I changed to `/api/hassio_ingress` and now it works. See my config below:
```
    server {
        access_log /var/log/nginx/$redacted_domain.access.log;
        error_log /var/log/nginx/$redacted_domain.error.log;
        listen 443 ssl;
        server_name  ha.$redacted_domain;

	ssl_certificate /usr/local/etc/letsencrypt/live/$redacted_domain/fullchain.pem;
	ssl_certificate_key /usr/local/etc/letsencrypt/live/$redacted_domain/privkey.pem;
	ssl_session_cache builtin:1000 shared:SSL:10m;
	ssl_protocols TLSv1 TLSv1.1 TLSv1.2;
	ssl_ciphers HIGH:!aNULL:!eNULL:!EXPORT:!CAMELLIA:!DES:!MD5:!PSK:!RC4;
	ssl_prefer_server_ciphers on;

        location / {
		proxy_pass http://192.168.1.20:8123;
		proxy_set_header Upgrade $http_upgrade;
		proxy_set_header Connection "upgrade";
	}

	location /api/hassio_ingress {
		proxy_pass http://192.168.1.20:8123/api/hassio_ingress;
		proxy_set_header Host $host;
		proxy_http_version 1.1;
		proxy_set_header Upgrade $http_upgrade;
		proxy_set_header Connection "upgrade";
	}

    }
```

closed 2021-05-23 by stale[bot]

thefunkygibbon, 2021-09-02:
i get this same issue.  is it an 'issue'? or is it something people in this thread (and me) are doing wrong?
its driving me mad.   
is there an alternative to using nginx to get ssl working?

nohn, 2021-09-02:
@thefunkygibbon in Apache, it looks like this (1.2.3.4:5678 is native ip and port of esphome):

      ProxyRequests Off
      RewriteEngine on

      ProxyPass /esphome http://1.2.3.4:5678
      ProxyPassReverse /esphome http://1.2.3.4:5678
      RewriteCond %{HTTP:Upgrade} websocket [NC]
      RewriteCond %{HTTP:Connection} upgrade [NC]
      RewriteRule ^/esphome/(.*) "ws://1.2.3.4:5678/$1" [P,L]

bigmike613, 2021-12-19:
I am having the same issue. I have esphome running in a docker container, separate from homeassistant. I then have nginx running on a Ubuntu vm. the proxy for http to https works fine. I can view the esphome dashboard and can click all links but the console is always blank when I try to build or view logs. Also, the remote serial connection fails "Failed to prepare configuration" (the main reason I am setting up the reverse proxy). The chrome dev. console shows "WebSocket connection to 'wss://10.10.10.251/ace' failed: ". Any help is greatly appreciated.

nginx config:
```
upstream websocket{
        server 10.10.10.250:6052;
}

server {

    listen 443;
    server_name default_server;

    ssl_certificate           /etc/nginx/cert.crt;
    ssl_certificate_key       /etc/nginx/cert.key;

    ssl on;
    ssl_session_cache  builtin:1000  shared:SSL:10m;
    ssl_protocols  TLSv1 TLSv1.1 TLSv1.2;
    ssl_ciphers HIGH:!aNULL:!eNULL:!EXPORT:!CAMELLIA:!DES:!MD5:!PSK:!RC4;
    ssl_prefer_server_ciphers on;

    access_log            /var/log/nginx/esphome.access.log;
    error_log            /var/log/nginx/esphome.error.log;
    location / {

      proxy_pass          http://10.10.10.250:6052;
      proxy_set_header Upgrade $http_upgrade;
      proxy_set_header Connection "upgrade";
      proxy_set_header Host $host:$server_port;
      proxy_set_header X-Hassio-Ingress YES;
      proxy_read_timeout 90;
}
```

  }

theautomation, 2021-12-22:
I have same issue as @bigmike613

koying, 2021-12-28:
Looks like the trick is to add a specific header:

```
proxy_set_header X-NginX-Proxy true;
```

bigmike613, 2021-12-28:
see my new default file below. I still cannot see any text in the terminal. 

```
upstream websocket{
        server 10.10.10.250:6052;
}

server {

    listen 443;
    server_name default_server;

    ssl_certificate           /etc/nginx/cert.crt;
    ssl_certificate_key       /etc/nginx/cert.key;

    ssl on;
    ssl_session_cache  builtin:1000  shared:SSL:10m;
    ssl_protocols  TLSv1 TLSv1.1 TLSv1.2;
    ssl_ciphers HIGH:!aNULL:!eNULL:!EXPORT:!CAMELLIA:!DES:!MD5:!PSK:!RC4;
    ssl_prefer_server_ciphers on;

    access_log            /var/log/nginx/esphome.access.log;
    error_log            /var/log/nginx/esphome.error.log;
    location / {

      proxy_pass          http://10.10.10.250:6052;
      proxy_set_header Upgrade $http_upgrade;
      proxy_set_header Connection "upgrade";
      proxy_set_header Host $host:$server_port;
      proxy_set_header X-NginX-Proxy true;
      proxy_read_timeout 90;
}

  }
```

koying, 2021-12-29:
Her is my working config:
```
upstream backend_esphome.semperpax.local {
      # esphome
      server uranus.lan:6052;
}
server {
  listen 443;
  listen [::]:443;
  include global/ssl_local.conf;
  server_name esphome.semperpax.local;
  location / {
    proxy_pass http://backend_esphome.semperpax.local;
    include /etc/nginx/proxy_params;
    proxy_set_header X-NginX-Proxy true;
  }
}
```

proxy_params
```
proxy_set_header Host $http_host;
proxy_set_header X-Real-IP $remote_addr;
proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
proxy_set_header X-Forwarded-Proto $scheme;
proxy_ssl_name $host;
proxy_ssl_server_name on;
```

theautomation, 2021-12-29:
No success when adding `proxy_set_header X-NginX-Proxy true;`
```
# ESPhome
upstream esphome {
    server docker-host.lan:6052;
}

server {
    listen 80;
    server_name esphome.docker;
    server_tokens off;

    location / {
        proxy_pass http://esphome;
        proxy_set_header Host $http_host;
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header X-Forwarded-Proto $scheme;
        proxy_ssl_name $host;
        proxy_ssl_server_name on;
        proxy_set_header X-NginX-Proxy true;
    }
}
```

bigmike613, 2021-12-29:
> Her is my working config:
>
> ```
> upstream backend_esphome.semperpax.local {
>       # esphome
>       server uranus.lan:6052;
> }
> server {
>   listen 443;
>   listen [::]:443;
>   include global/ssl_local.conf; [...]

you done have the port on "proxy_pass http://backend_esphome.semperpax.local;". are you hosting it on port 6052?

theautomation, 2021-12-30:
@bigmike613 What do you mean? there is no port seen in the proxy_pass in the working configuration examples here. the port is already set in the upstream section. I see this error in the browser console 
```javascript
index-7db316db.js:1 WebSocket connection to 'ws://esphome.docker/ace' failed: 
f @ index-7db316db.js:1
```

koying, 2021-12-30:
> Looks like the trick is to add a specific header

I take that off.
Actually, I cannot make it fail anymore. Strange behavior...

bigmike613, 2021-12-30:
> 

If I remove :6052 from the proxy_pass I cannot get to the website at all. I have all other lines matching @koying working config but it still does not work.