esphome/issues #1039: Docs wrong on valid characters for name of

jimmypuckett, 2020-02-07:
The docs here...

https://esphome.io/components/esphome.html

says that the `name` key use upper case...

```
name (Required, string): This is the name of the node. It should always be unique in your esphome network. May only contain upper/lowercase characters, digits and underscores. See Changing ESPHome Node Name.
```

However, this is the error that you get when using an uppercase...

```
'G' is an invalid character for names. Valid characters are: abcdefghijklmnopqrstuvwxyz0123456789_ (lowercase, no spaces).
  name: Gate
```

As you can see I was using "Gate" for the `name`, but only "gate" works.

It appears that the text should say...

```
name (Required, string): This is the name of the node. It should always be unique in your esphome network. May only contain lowercase characters, digits and underscores. See Changing ESPHome Node Name.
```

DanielWinks, 2020-03-20:
Additionally, the hostname character validation checks are wrong.

See [here](http://man7.org/linux/man-pages/man7/hostname.7.html), [here](https://tools.ietf.org/html/rfc952), and [here](https://en.wikipedia.org/wiki/Hostname#Restrictions_on_valid_hostnames).

Hostnames can NOT have underscores. ESPHome allows them, this should be fixed. Additionally, ESPHome prevents the use of dash/minus characters, which ARE allowed, this should also be fixed.

Technically, the RFC regarding this allows for upper and lower case, which is then supposed to be used case-insensitively, however simply enforcing all lower case is more compatible. That said, allowing underscores in hostnames does break DNS resolution in many cases, and dashes should be allowed, just not as the first or last character.

DanielWinks, 2020-03-20:
The ESPHome code forcibly prevents the use of perfectly valid dash characters:

`'-' is an invalid character for names. Valid characters are: abcdefghijklmnopqrstuvwxyz0123456789_ (lowercase, no spaces).`

This should be changed to ``Valid characters are: abcdefghijklmnopqrstuvwxyz0123456789- (lowercase, no spaces, - may not be first or last character).``

OttoWinter, 2020-07-24:
Fixed in https://github.com/esphome/esphome-docs/commit/9b6170ac6

closed 2020-07-24 by OttoWinter