esphome/issues #1040: http_request in dev branch is sending corrupted (header) data

bjobo, 2020-02-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Windows 10 pip install with git clone (2020-02-07)

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 nodemcuv2

**Affected component:**

http_request

**Description of problem:**
I tried using the new http_request component via the dev channel. The request is send and "something" is received, however the header data seems broken, so that the request can not be handled. In the esphome log there is only one line related to the component.

**Problem-relevant YAML-configuration entries:**
```yaml
http_request:
  useragent: esphome/device
  timeout: 10s

switch:
  - platform: gpio
    pin: 10
    name: "Test http_request"
    on_turn_on:
    - http_request.send: 
        method: GET
        url: http://192.168.179.101:3001/
        headers:
          Content-Type: application/json
        verify_ssl: false
```

**Logs (if applicable):**

ESPHome Log:
```
[22:40:39][I][app:058]: setup() finished successfully!
[22:40:39][I][app:101]: ESPHome version 1.15.0-dev compiled on Feb  7 2020, 22:40:06
...
[22:42:10][D][switch:029]: 'Test http_request' Toggling ON.
[22:42:10][D][switch:045]: 'Test http_request': Sending state ON
[22:45:05][W][http_request:047]: HTTP Request failed; URL: http://192.168.179.101:3001/; Error: connection refused
```

The http request as it is received (send by esphome):
```
pi@Hassio:~ $ nc -l 3001
▒▒8▒▒▒▒▒R▒▒▒▒▒nc▒▒▒$▒▒1IZ̨̩▒+▒/▒,▒0▒▒▒▒▒#▒'▒$▒(▒  ▒▒
▒▒-▒1▒.▒2▒%▒)▒&▒*▒▒▒▒▒▒▒▒▒▒<=/5▒▒▒
P▒192.168.179.101

                                                                                 PuTTYPuTTYPuTTY
```
The same url send by a browser:
```
pi@Hassio:~ $ nc -l 3001
GET /api/v1/commands/?cmd=toggle HTTP/1.1
Host: 192.168.179.101:3001
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:72.0) Gecko/20100101 Firefox/72.0
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8
Accept-Language: de,en-US;q=0.7,en;q=0.3
Accept-Encoding: gzip, deflate
Connection: keep-alive
Upgrade-Insecure-Requests: 1
Pragma: no-cache
Cache-Control: no-cache
```

**Additional information and things you've tried:**
- Tried differen Content-Types: text/plain as well as application/json
- Tried different Methods (GET, PUT, POST)
- Tried different

OttoWinter, 2020-02-09:
CC @Anonym-tsk

Anonym-tsk, 2020-02-09:
Please check stable version too.

bjobo, 2020-02-09:
Which stable version? I can't find http_request in the stable (=master?) branch nor in a clean pip install.

Do you mean to use the master branch and add the component manually? I tried adding the http_request folder with its 3 files from the dev branch (I did nothing else). It gives me the following error.

```
esphome esp_rfid_wz2.yaml clean
...
Failed config

http_request: [source esp_rfid_wz2.yaml:62]

  Component not found: http_request.
  useragent: esphome/device
  timeout: 10s
switch.gpio: [source esp_rfid_wz2.yaml:110]
  platform: gpio
  pin: 10
  name: Test http_request
  on_turn_on:
    - [source esp_rfid_wz2.yaml:114]

      Unable to find action with the name 'http_request.send'.
      http_request.send:  [source esp_rfid_wz2.yaml:115]
        method: GET
        url: http://192.168.179.101:3001/
        headers:
          Content-Type: application/json
        verify_ssl: False
```

Anonym-tsk, 2020-02-10:
Sorry, http_request is not available in stable. I'm checking problem.

closed 2020-03-12 by brandond