esphome/issues #1043: dynamic component variables (This option is not templatable)

Bierchermuesli, 2020-02-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** docker
**ESP (ESP32/ESP8266, Board/Sonoff):** wemos_d1_mini32
**Affected component:** tm1637 (from dev) and/or maybe others
**Description of problem:**
obiously it is not possible to use lambdas for component variables (probably not made for this?)
In this case I would like to change a component variable during runtime (display brightness). 

**Problem-relevant YAML-configuration entries:**
```yaml
display:
    platform: tm1637
    id: tm1637_display
    clk_pin: GPIO23
    dio_pin: GPIO05
    intensity: !lambda return id(brightness_value).state;
    ...
```

**Logs (if applicable):**
```
  This option is not templatable!.
  intensity: !lambda |-
    return id(brightness_value).state
```

**Additional information and things you've tried:**
globals vars won't work either
substitutions would work but can't be changed during runtime IMHO.

Any advice?
btw. tm1637 from current dev works great! Thanks!

OttoWinter, 2020-02-09:
That's to be expected.

There's a lot of effort involved in making settings runtime-configurable. There's no way to make all static options "magically" be dynamically adjustable.

In this case, you can call the `set_intensity` method of the display to manually change it (although that should be documented, because this will not work for other components).

closed 2020-02-09 by OttoWinter

jumblies, 2020-02-13:
I am trying what I think is the same thing with max7219.  

Is the right way to access class methods in a lambda like so?
```
    lambda: |-
      auto time = id(ha_time).now();
      if ((time.second % 7) == 0) {
        it.set_intensity(15);
        it.print("        ");
        it.printf(0, "Out %.0f F", id(dstemp).state);
      }
```

glmnet, 2020-02-19:
yes, it should work that way

jumblies, 2020-02-19:
Thanks! I appreciate the response.