esphome/issues #1046: Sonoff TH16, DS18B20, crash while booting when four GPIO are in use

Variour, 2020-02-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, Sonoff TH16, Sonoff DS18B20

**Affected component:**

status_led https://esphome.io/components/status_led.html  
dallas https://esphome.io/components/sensor/dallas.html  
binary_sensor https://esphome.io/components/binary_sensor/index.html  
switch https://esphome.io/components/switch/index.html

**Description of problem:**

Sonoff TH16 with the Sonoff DS18B20 sensor  
The Sonoff crashes after connecting to wifi and receiving dns server configuration (receiving logs over wifi is not possible in this case). I assume the next step would be setting up of the other components, which is the point at which it crashes.  
I have two TH16 and two Sonoff DS18B20, the error persits with both devices and while switching the sensors between the TH16.  
I have tested one of the TH16 with the sensor on the stock eWeLink firmware. It works and reports the temperature.  
I tried replacing the dallas sensor address with index, no difference.  
Dev version is not working as well.  
Whenever i disable one of the components (or physically disconnect the sensor) the sonoff works, see table below. (Each row a seperate test, Tick enabled, cross commented out)

|status_led|dallas|sensor|sensor connected|binary_sensor|switch|works?|
|---       |---   |---   |---|---          |---   |---   |
|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|:x:|
|:x:|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|
|:heavy_check_mark:|:heavy_check_mark:|:x:|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|
|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|:x:|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|
|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|:x:|:heavy_check_mark:|:heavy_check_mark:|
|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|:x:|:heavy_check_mark:|

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: th16_prototype
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "SpeedIOT"
  password: "xxx"

logger:

ota:
  password: "xxx"
  
api:
  password: "xxx"

status_led:
  pin: GPIO13
  
dallas:
  - pin: GPIO14

sensor:
  - platform: dallas
    address: 0x6B0300A279C23328
    #index: 0
    id: temp
      
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    id: button
      
switch:
  - platform: gpio
    id: relay
    pin: GPIO12

```

**Logs (if applicable):**

non working example
```
[22:51:00][I][logger:156]: Log initialized
[22:51:00][C][status_led:014]: Setting up Status LED...
[22:51:00][C][ota:364]: There have been 1 suspected unsuccessful boot attempts.
[22:51:00][I][app:028]: Running through setup()...
[22:51:00][D][binary_sensor:034]: 'button': Sending initial state OFF
[22:51:00][C][switch.gpio:011]: Setting up GPIO Switch 'relay'...
[22:51:00][D][switch:025]: 'relay' Turning OFF.
[22:51:00][D][switch:045]: 'relay': Sending state OFF
[22:51:00][D][switch:025]: 'relay' Turning OFF.
[22:51:00][C][dallas.sensor:032]: Setting up DallasComponent...
[22:51:00][C][wifi:033]: Setting up WiFi...
[22:51:00][D][wifi:304]: Starting scan...
[22:51:01][D][dallas.sensor:148]: 'temp': Got Temperature=22.3°C
[22:51:01][D][sensor:092]: 'temp': Sending state 22.31250 °C with 1 decimals of accuracy
[22:51:07][D][wifi:319]: Found networks:
[22:51:07][I][wifi:365]: - 'SpeedIOT' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:366]:     Channel: 1
[22:51:07][D][wifi:367]:     RSSI: -23 dB
[22:51:07][D][wifi:369]: - 'Speed2.1' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'Speed5' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - '' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'SpeedGuest' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'lodhianetwork' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'Telekom_FON' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'WLAN-MRCG58' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'Telekom_FON' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'Vodafone-07EC' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'KarlvomKiosk' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'WLAN-LH5Y7D' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'KDG-53345' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'Ha Vu' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'Telekom_FON' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'FRITZ!Box 7412' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'o2-WLAN27' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'WLAN-793757' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'WLAN-895163' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'Telekom_FON' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'hoddel' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'EasyBox-098978' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][D][wifi:369]: - 'Vodafone-BB95' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[22:51:07][I][wifi:193]: WiFi Connecting to 'SpeedIOT'...
[22:51:08][I][wifi:423]: WiFi Connected!
[22:51:08][C][wifi:283]:   SSID: 'SpeedIOT'
[22:51:08][C][wifi:284]:   IP Address: 192.168.9.124
[22:51:08][C][wifi:286]:   BSSID: xx:xx:xx:xx:xx:xx
[22:51:08][C][wifi:287]:   Hostname: 'th16_prototype'
[22:51:08][C][wifi:291]:   Signal strength: -29 dB ▂▄▆█
[22:51:08][C][wifi:295]:   Channel: 1
[22:51:08][C][wifi:296]:   Subnet: 255.255.255.0
[22:51:08][C][wifi:297]:   Gateway: 192.168.9.1
[22:51:08][C][wifi:298]:   DNS1: 192.168.11.61
[22:51:08][C][wifi:299]:   DNS2: 192.168.11.62
[22:51:15]
[22:51:15] ets Jan  8 2013,rst cause:4, boot mode:(3,7)
[22:51:15]
[22:51:15]wdt reset
[22:51:15]load 0x4010f000, len 1384, room 16 
[22:51:15]tail 8
[22:51:15]chksum 0x2d
[22:51:15]csum 0x2d
[22:51:15]v8b899c12
[22:51:15]~ld
[22:51:15][I][logger:156]: Log initialized

```
working example, status_led disabled
```
[23:12:03][I][app:100]: ESPHome version 1.14.3 compiled on Feb 10 2020, 23:07:14
[23:12:03][C][wifi:415]: WiFi:
[23:12:03][C][wifi:283]:   SSID: 'SpeedIOT'
[23:12:03][C][wifi:284]:   IP Address: 192.168.9.124
[23:12:03][C][wifi:286]:   BSSID: xx:xx:xx:xx:xx:xx
[23:12:03][C][wifi:287]:   Hostname: 'th16_prototype'
[23:12:03][C][wifi:291]:   Signal strength: -31 dB ▂▄▆█
[23:12:03][C][wifi:295]:   Channel: 1
[23:12:03][C][wifi:296]:   Subnet: 255.255.255.0
[23:12:03][C][wifi:297]:   Gateway: 192.168.9.1
[23:12:03][C][wifi:298]:   DNS1: 192.168.11.61
[23:12:03][C][wifi:299]:   DNS2: 192.168.11.62
[23:12:03][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'button'
[23:12:03][C][gpio.binary_sensor:016]:   Pin: GPIO0 (Mode: INPUT_PULLUP, INVERTED)
[23:12:03][C][switch.gpio:042]: GPIO Switch 'relay'
[23:12:03][C][switch.gpio:043]:   Pin: GPIO12 (Mode: OUTPUT)
[23:12:03][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[23:12:03][C][logger:175]: Logger:
[23:12:03][C][logger:176]:   Level: DEBUG
[23:12:03][C][logger:177]:   Log Baud Rate: 115200
[23:12:03][C][logger:178]:   Hardware UART: UART0
[23:12:03][C][dallas.sensor:070]: DallasComponent:
[23:12:03][C][dallas.sensor:071]:   Pin: GPIO14 (Mode: INPUT)
[23:12:03][C][dallas.sensor:072]:   Update Interval: 60.0s
[23:12:03][D][dallas.sensor:077]:   Found sensors:
[23:12:03][D][dallas.sensor:080]:     0x6B0300A279C23328
[23:12:03][C][dallas.sensor:085]:   Device 'temp'
[23:12:03][C][dallas.sensor:085]:     Unit of Measurement: '°C'
[23:12:03][C][dallas.sensor:085]:     Accuracy Decimals: 1
[23:12:03][C][dallas.sensor:085]:     Icon: 'mdi:thermometer'
[23:12:03][C][dallas.sensor:093]:     Address: 0x6B0300A279C23328
[23:12:03][C][dallas.sensor:094]:     Resolution: 12
[23:12:03][C][ota:029]: Over-The-Air Updates:
[23:12:03][C][ota:030]:   Address: th16_prototype.local:8266
[23:12:03](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[23:12:03][C][api:095]: API Server:
[23:12:03][C][api:096]:   Address: th16_prototype.local:6053
[23:12:46][D][dallas.sensor:148]: 'temp': Got Temperature=23.1°C
[23:12:46][D][sensor:092]: 'temp': Sending state 23.06250 °C with 1 decimals of accuracy
```

**Additional information and things you've tried:**

Variour, 2020-07-24:
This is, as far as I know, still happening. But I will try again with the latest release and update the issue accordingly.

coogle, 2020-07-25:
I can say that I am also experiencing crashes with the TH16. It doesn't matter if the sensor is plugged in or not, but if I leave it plugged in at my desk for awhile eventually the button stops responding and the device doesn't seem to be able to accept updates either.

closed 2020-12-06 by stale[bot]

ptr727, 2021-08-13:
Looks similar to an issue I have with the TH10 when using multiple Dallas sensors: https://github.com/esphome/issues/issues/1548
I have multiple TH10's, as soon as I add more than 1 dallas sensor I get occasional watchdog resets.
I suspect something with the dallas code not keeping the watchdog happy / taking too long hen more than one sensor is in play?