esphome/issues #1049: Platformio 4.2.0 is out and breaks esphome-dev

middelink, 2020-02-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

native linux, venv

This is an esphome-dev instance with `esphome.egg-info/requires.txt` edited to "platformio==4.2.0" to make it even start compiling.

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32

**Affected component:**

ble_tracker, but I don't think thats relevant.

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ble_tracker_${chipid}
  platform: ESP32
  board: nodemcu-32s
  arduino_version: latest
  build_path: .build

substitutions:
  chipid: "042a30"
  mqtt_prefix: esp8266/${chipid}

wifi: !include common/wifi-ch.yaml

mqtt: !include common/mqtt-ch.yaml

# Enable logging
logger:
  level: INFO

# Enable OTA
ota:

# Enable BLE scanning
esp32_ble_tracker:

sensor:
  - platform: xiaomi_hhccjcy01
    mac_address: 80:EA:CA:88:EF:2D
    temperature:
      name: "${chipid} Xiaomi MiFlora Temperature"
    moisture:
      name: "${chipid} Xiaomi MiFlora Moisture"
    illuminance:
      name: "${chipid} Xiaomi MiFlora Illuminance"
    conductivity:
      name: "${chipid} Xiaomi MiFlora Soil Conductivity"
    battery_level:
      name: "${chipid} Xiaomi MiFlora Battery Level"

  - platform: wifi_signal
    name: "${chipid} WiFi Signal"
```

**Logs (if applicable):**

```
$ esphome ble_tracker_042a30.yaml run
INFO Reading configuration ble_tracker_042a30.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d .build
Processing ble_tracker_042a30 (board: nodemcu-32s; framework: arduino; platform: espressif32)
[... 199 lines cut ...]
                         ^
.piolibdeps/ble_tracker_042a30/ESPAsyncTCP-esphome_ID6757/src/ESPAsyncTCP.cpp: In member function 'void AsyncServer::begin()':
.piolibdeps/ble_tracker_042a30/ESPAsyncTCP-esphome_ID6757/src/ESPAsyncTCP.cpp:1122:14: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   local_addr.addr = (uint32_t) _addr;
              ^
*** [.pioenvs/ble_tracker_042a30/lib783/ESPAsyncTCP-esphome_ID6757/AsyncPrinter.cpp.o] Error 1
.piolibdeps/ble_tracker_042a30/ESPAsyncTCP-esphome_ID6757/src/ESPAsyncTCPbuffer.cpp: In constructor 'AsyncTCPbuffer::AsyncTCPbuffer(AsyncClient*)':
.piolibdeps/ble_tracker_042a30/ESPAsyncTCP-esphome_ID6757/src/ESPAsyncTCPbuffer.cpp:35:15: error: 'panic' was not declared in this scope
         panic();
               ^
.piolibdeps/ble_tracker_042a30/ESPAsyncTCP-esphome_ID6757/src/ESPAsyncTCPbuffer.cpp: In member function 'virtual size_t AsyncTCPbuffer::write(const uint8_t*, size_t)':
.piolibdeps/ble_tracker_042a30/ESPAsyncTCP-esphome_ID6757/src/ESPAsyncTCPbuffer.cpp:127:23: error: 'panic' was not declared in this scope
                 panic();
                       ^
.piolibdeps/ble_tracker_042a30/ESPAsyncTCP-esphome_ID6757/src/ESPAsyncTCPbuffer.cpp: In member function 'size_t AsyncTCPbuffer::_handleRxBuffer(uint8_t*, size_t)':
.piolibdeps/ble_tracker_042a30/ESPAsyncTCP-esphome_ID6757/src/ESPAsyncTCPbuffer.cpp:469:21: error: 'panic' was not declared in this scope
               panic(); //TODO: What action should this be ?
                     ^
*** [.pioenvs/ble_tracker_042a30/lib783/ESPAsyncTCP-esphome_ID6757/ESPAsyncTCP.cpp.o] Error 1
*** [.pioenvs/ble_tracker_042a30/lib783/ESPAsyncTCP-esphome_ID6757/ESPAsyncTCPbuffer.cpp.o] Error 1
=============================================================== [FAILED] Took 10.76 seconds ===============================================================

```

**Additional information and things you've tried:**

middelink, 2020-02-13:
Addendum: going back to platformio-4.1.0 it compiles again, but I noticed the dependencies changes as well:

```
$ esphome ble_tracker_042a30.yaml compile
INFO Reading configuration ble_tracker_042a30.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d .build
**************************************************************************************************************************************************************
Obsolete PIO Core v4.1.0 is used (previous was 4.2.0)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
**************************************************************************************************************************************************************
Processing ble_tracker_042a30 (board: nodemcu-32s; framework: arduino; platform: espressif32)
--------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <AsyncMqttClient-esphome> 0.8.4
|   |-- <AsyncTCP-esphome> 1.1.1
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling .pioenvs/ble_tracker_042a30/src/main.cpp.o
Linking .pioenvs/ble_tracker_042a30/firmware.elf
Retrieving maximum program size .pioenvs/ble_tracker_042a30/firmware.elf
Checking size .pioenvs/ble_tracker_042a30/firmware.elf
Building .pioenvs/ble_tracker_042a30/firmware.bin
DATA:    [==        ]  18.6% (used 60812 bytes from 327680 bytes)
PROGRAM: [========= ]  88.7% (used 1452710 bytes from 1638400 bytes)
================================================================ [SUCCESS] Took 10.75 seconds ================================================================
INFO Successfully compiled program.

```

```
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <AsyncMqttClient-esphome> 0.8.4
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <AsyncTCP-esphome> 1.1.1
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <Update> 1.0
|-- <WiFi> 1.0

vs

Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <AsyncMqttClient-esphome> 0.8.4
|   |-- <AsyncTCP-esphome> 1.1.1
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <Update> 1.0
|-- <WiFi> 1.0
```
Note the extra `<ESPAsyncTCP-esphome> 1.2.2` when using 4.2.0?

glmnet, 2020-02-19:
you shouldn't edit that file and then comply it does not work, it should work as is, thus you cannot upgrade platformio and expect it to work, it's like an art to have it working finding all versions matching, etc.

I mean, if the requirements specifies a version, it means it should work with that version, otherwise I don't understand the issue here

middelink, 2020-02-22:
Right. And from your words I can conclude you are not interested in other people helping? Never mind then, I'll wait till you discover platformio 4.2.0 is out yourself and run into the same problems.

closed 2020-02-22 by middelink