esphome/issues #1052: Sonoff device will kill power when home assistant is offline

ajfriesen, 2020-02-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant install method: https://www.home-assistant.io/hassio/installation/#alternative-install-home-assistant-supervised-on-a-generic-linux-host

Host:
```
Linux server 5.3.0-28-generic #30~18.04.1-Ubuntu SMP Fri Jan 17 06:14:09 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

esphome version: 
```
Add-on version: 1.14.3
 You are running the latest version of this add-on.
 System: Ubuntu 18.04.4 LTS  (amd64 / qemux86-64)
 Home Assistant version: 0.105.4
 Supervisor version: 200
```

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff S20
Sonoff Pow

**Affected component:**

**Description of problem:**
When home assistant is gone the relays in my sonoff s20 and pow turn off and on very 
quickly in some time interval (did not measure yet).
This behaviour would shutdown my pc for example which I would like to attach to a Sonoff Pow.
Is this my yaml configuration?
I thought the ```  reboot_timeout: 0s``` would fix this.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: sonoff_s20_02
  platform: ESP8266
  board: esp01_1m

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
    manual_ip:
      static_ip: !secret sonoff_s20_02_static_ip
      gateway: !secret wifi_gateway
      subnet: !secret wifi_subnet
  reboot_timeout: 0s
  ap:
    ssid: ${devicename}
    password: !secret esphome_ap_password

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      then:
        - switch.toggle: relay
    name: "sonoff_s20_02_button"
  - platform: status
    name: "sonoff_s20_02_status"

switch:
  - platform: gpio
    name: "sonoff_s20_02_relay"
    pin: GPIO12
    id: relay
    on_turn_on:
      - light.turn_on: led
    on_turn_off:
      - light.turn_off: led

output:
  # Register the green LED as a dimmable output ....
  - platform: esp8266_pwm
    id: s20_green_led
    pin:
      number: GPIO13
      inverted: True

light:
  # ... and then make a light out of it.
  - platform: monochromatic
    name: "sonoff_s20_02_led"
    output: s20_green_led
    id: led

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

ryrzy, 2020-02-17:
try to add   reboot_timeout: 0s to api (api, wifi and mqtt use this setting)

ajfriesen, 2020-02-17:
thanks, i think that was it!

closed 2020-02-17 by ajfriesen