esphome/issues #1053: Sensor value  on web sensor

jarcikw, 2020-02-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant / py. env

**ESP (ESP32/ESP8266, Board/Sonoff):**

Custom board based on ESP8266-12S (https://github.com/jarcikw/Puszka-v6.1)
**Affected component:**
Web Server

**Description of problem:**
When configure sensor dallas data displayed are wrongly displayed. Similar to issue https://github.com/esphome/issues/issues/806#issue-516738640 ,
Please see attached screenshots

**1. Configuration without dallas sensor**
[image]

**2. Configuration with dallas sensor**
[image]

**3. In Home Assistant is everything OK**
[image]

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  name: korytarz_sdm630
  ds18b20: '0xBF031840D35EFF28'
esphome:
  name: ${name}
  platform: ESP8266
  board: esp01_1m
  includes:
    - sdm/SDM_Config_User.h
    - sdm/SDM.h
    - sdm/sdm_esphome.h

uart:
  tx_pin: TX
  rx_pin: RX
  baud_rate: 9600
  stop_bits: 1

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass

# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${name}
    password: ""

captive_portal:

web_server:
  port: 80

# Enable logging
logger:
  baud_rate: 0
  level: ERROR
  
  esp8266_store_log_strings_in_flash: False
  
# Enable Home Assistant API
api:
  password: !secret api_pass
  reboot_timeout: 0s 

dallas:
  - pin: GPIO2
    update_interval: 60s

ota:
  password: !secret api_pass

sensor:

- platform: custom
  lambda: |-
    auto sdm_sensors = new SDMComponent();
    App.register_component(sdm_sensors);
    return {sdm_sensors->voltage_sensor, sdm_sensors->current_sensor, sdm_sensors->power_sensor, sdm_sensors->energy_import_sensor, sdm_sensors->voltage1_sensor, sdm_sensors->voltage2_sensor, sdm_sensors->voltage3_sensor,};
  sensors:
    - name: "${name}_Voltage"
      unit_of_measurement: V
      accuracy_decimals: 1
    - name: "Current"
      unit_of_measurement: A
      accuracy_decimals: 1
    - name: "Power"
      unit_of_measurement: W
      accuracy_decimals: 1
    - name: "Energy import"
      unit_of_measurement: kWh
      accuracy_decimals: 1
    - name: "Voltage1"
      unit_of_measurement: V
      accuracy_decimals: 1
    - name: "Voltage2"
      unit_of_measurement: V
      accuracy_decimals: 1      
    - name: "Voltage3"
      unit_of_measurement: V
      accuracy_decimals: 1     

 
#- platform: dallas
#  address: ${ds18b20}
#  name: ${name}_Temp         

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[13:28:34][I][app:100]: ESPHome version 1.14.3 compiled on Feb 18 2020, 13:27:38
[13:28:34][C][wifi:415]: WiFi:
[13:28:34][C][wifi:283]:   SSID: [redacted]
[13:28:34][C][wifi:284]:   IP Address: 192.168.10.248
[13:28:34][C][wifi:286]:   BSSID: [redacted]
[13:28:34][C][wifi:287]:   Hostname: 'korytarz_sdm630'
[13:28:34][C][wifi:291]:   Signal strength: -68 dB ▂▄▆█
[13:28:34][V][wifi:293]:   Priority: 0.0
[13:28:34][C][wifi:295]:   Channel: 3
[13:28:34][C][wifi:296]:   Subnet: 255.255.0.0
[13:28:34][C][wifi:297]:   Gateway: 192.168.1.1
[13:28:34][C][wifi:298]:   DNS1: 192.168.1.2
[13:28:34][C][wifi:299]:   DNS2: (IP unset)
[13:28:34][C][uart:140]: UART Bus:
[13:28:34][C][uart:142]:   TX Pin: GPIO1
[13:28:34][C][uart:145]:   RX Pin: GPIO3
[13:28:34][C][uart:147]:   Baud Rate: 9600 baud
[13:28:34][C][uart:148]:   Stop bits: 1
[13:28:34][C][uart:150]:   Using hardware serial interface.
[13:28:34][C][logger:175]: Logger:
[13:28:34][C][logger:176]:   Level: VERY_VERBOSE
[13:28:34][C][logger:177]:   Log Baud Rate: 0
[13:28:34][C][logger:178]:   Hardware UART: UART0
[13:28:34][C][dallas.sensor:070]: DallasComponent:
[13:28:34][C][dallas.sensor:071]:   Pin: GPIO2 (Mode: INPUT)
[13:28:34][C][dallas.sensor:072]:   Update Interval: 60.0s
[13:28:34][D][dallas.sensor:077]:   Found sensors:
[13:28:34][D][dallas.sensor:080]:     0xBF031840D35EFF28
[13:28:34][C][dallas.sensor:085]:   Device 'korytarz_sdm630_Temp'
[13:28:34][C][dallas.sensor:085]:     Unit of Measurement: '°C'
[13:28:34][C][dallas.sensor:085]:     Accuracy Decimals: 1
[13:28:34][C][dallas.sensor:085]:     Icon: 'mdi:thermometer'
[13:28:34][V][dallas.sensor:085]:     Unique ID: 'dallas-BF031840D35EFF28'
[13:28:34][C][dallas.sensor:093]:     Address: 0xBF031840D35EFF28
[13:28:34][C][dallas.sensor:094]:     Resolution: 12
[13:28:34][C][captive_portal:169]: Captive Portal:
[13:28:34][C][web_server:122]: Web Server:
[13:28:34][C][web_server:123]:   Address: korytarz_sdm630.local:80
[13:28:34][C][ota:029]: Over-The-Air Updates:
[13:28:34][C][ota:030]:   Address: korytarz_sdm630.local:8266
[13:28:34][C][ota:032]:   Using Password.
[13:28:34][C][api:095]: API Server:
[13:28:34][C][api:096]:   Address: korytarz_sdm630.local:6053
[13:28:39][VV][api.service:250]: on_ping_request: PingRequest {}
[13:28:39][VV][api.service:035]: send_ping_response: PingResponse {}
[13:28:40][VV][scheduler:131]: Running interval 'update' with interval=10000 last_execution=11778 (now=21778)
[13:28:40][D][SDM:025]: Start reading
[13:28:40][D][SDM:029]: Voltages 240.256195 241.460648 240.637619
[13:28:40][V][sensor:013]: 'korytarz_sdm630_Voltage': Received new state 240.784836
[13:28:40][D][sensor:092]: 'korytarz_sdm630_Voltage': Sending state 240.78484 V with 1 decimals of accuracy
[13:28:40][V][sensor:013]: 'Voltage1': Received new state 240.256195
[13:28:40][D][sensor:092]: 'Voltage1': Sending state 240.25620 V with 1 decimals of accuracy
[13:28:40][V][sensor:013]: 'Voltage2': Received new state 241.460648
[13:28:40][D][sensor:092]: 'Voltage2': Sending state 241.46065 V with 1 decimals of accuracy
[13:28:40][V][sensor:013]: 'Voltage3': Received new state 240.637619
[13:28:40][D][sensor:092]: 'Voltage3': Sending state 240.63762 V with 1 decimals of accuracy
[13:28:40][V][sensor:013]: 'Current': Received new state 1.335342
[13:28:40][D][sensor:092]: 'Current': Sending state 1.33534 A with 1 decimals of accuracy
[13:28:40][V][sensor:013]: 'Power': Received new state 212.985321
[13:28:40][D][sensor:092]: 'Power': Sending state 212.98532 W with 1 decimals of accuracy
[13:28:40][D][SDM:047]: Imports 1302.146973 409.596008 1170.395020
[13:28:40][V][sensor:013]: 'Energy import': Received new state 2882.137939
[13:28:40][D][sensor:092]: 'Energy import': Sending state 2882.13794 kWh with 1 decimals of accuracy
[13:28:40][D][SDM:050]: End reading
[13:28:40][V][app:076]: A component took a long time in a loop() cycle (0.27 s).
[13:28:40][V][app:077]: Components should block for at most 20-30ms in loop().
[13:28:41][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=13043 (now=23043)
[13:28:44][VV][api.service:250]: on_ping_request: PingRequest {}
[13:28:44][VV][api.service:035]: send_ping_response: PingResponse {}
[13:28:49][VV][api.service:250]: on_ping_request: PingRequest {}
[13:28:49][VV][api.service:035]: send_ping_response: PingResponse {}
[13:28:50][VV][scheduler:131]: Running interval 'update' with interval=10000 last_execution=21778 (now=31778)
[13:28:50][D][SDM:025]: Start reading
[13:28:50][D][SDM:029]: Voltages 240.337692 241.527283 241.309067
[13:28:50][V][sensor:013]: 'korytarz_sdm630_Voltage': Received new state 241.058029
[13:28:50][D][sensor:092]: 'korytarz_sdm630_Voltage': Sending state 241.05803 V with 1 decimals of accuracy
[13:28:50][V][sensor:013]: 'Voltage1': Received new state 240.337692
[13:28:50][D][sensor:092]: 'Voltage1': Sending state 240.33769 V with 1 decimals of accuracy
[13:28:50][V][sensor:013]: 'Voltage2': Received new state 241.527283
[13:28:50][D][sensor:092]: 'Voltage2': Sending state 241.52728 V with 1 decimals of accuracy
[13:28:50][V][sensor:013]: 'Voltage3': Received new state 241.309067
[13:28:50][D][sensor:092]: 'Voltage3': Sending state 241.30907 V with 1 decimals of accuracy
[13:28:50][V][sensor:013]: 'Current': Received new state 1.304431
[13:28:50][D][sensor:092]: 'Current': Sending state 1.30443 A with 1 decimals of accuracy
[13:28:50][V][sensor:013]: 'Power': Received new state 207.276215
[13:28:50][D][sensor:092]: 'Power': Sending state 207.27621 W with 1 decimals of accuracy
[13:28:50][D][SDM:047]: Imports 1302.146973 409.596008 1170.395020
[13:28:50][V][sensor:013]: 'Energy import': Received new state 2882.137939
[13:28:50][D][sensor:092]: 'Energy import': Sending state 2882.13794 kWh with 1 decimals of accuracy
[13:28:50][D][SDM:050]: End reading
[13:28:50][V][app:076]: A component took a long time in a loop() cycle (0.27 s).
[13:28:50][V][app:077]: Components should block for at most 20-30ms in loop().
[13:28:51][VV][scheduler:131]: Running interval '' with interval=10000 last_execution=23043 (now=33050)
[13:28:54][VV][api.service:250]: on_ping_request: PingRequest {}
[13:28:54][VV][api.service:035]: send_ping_response: PingResponse {}
```

**Additional information and things you've tried:**

glmnet, 2020-02-20:
duplicate of #806 

I don't know much of this, but can you go to F12 tools go to the network tab and check what are your receiving in "Events"? Add as many screenshots as you like.

Thanks

closed 2020-02-20 by glmnet

glmnet, 2020-02-20:
Kind of replicated this issue, seems like events are not sent from ESPHome to the web, may be a ESPAsyncWebServer issue.
Some of them are sent then it stops updating, then browser reconnects, get the same few at the beginning and stops receiving again.

Testing with ESP8266 of course... anybody having this issue with ESP32 or is it ESP8266 only?

jarcikw, 2020-02-20:
Issue only with favoicon
[image]