esphome/issues #1054: 502 Bad Gateway for ESPHome web UI on docker HASS.io

rjalexa, 2020-02-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**RPi4 Raspbian Buster Lite with Docker - HASS.io docker on it - ESPHome 1.14.3 from https://github.com/esphome/hassio repository**
**Fresh HASS.io install with only ESPHome, Mosquitto and Config File add-ons**

**ESP (ESP32/ESP8266, Board/Sonoff):**

Will want to use ESPHome to flash several ESP32-WROOM ESP-32S NodeMCU boards but problem prevents me to get to the device flash stage

**Affected component:**

**ESPHome WEBUI inside HASS.io**

**Description of problem:**

**The Launch Web UI command from either the Add-on page or the left sidebar produces an empty page with a blue spinning circle for some 10 seconds then a simple text error -> 502: Bad Gateway**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
config file for ESPHome reverted to default (epmty); also tried with ssl: false but problem did not change
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
Hass.io Add-on: ESPHome
 ESPHome Hass.io add-on for intelligently managing all your ESP8266/ESP32 devices.
-----------------------------------------------------------
 Add-on version: 1.14.3
 You are running the latest version of this add-on.
 System: Raspbian GNU/Linux 10 (buster)  (armv7 / raspberrypi4)
 Home Assistant version: 0.105.5
 Supervisor version: 201

and each time I try to launch the WebUI I see the following in the Supervisor->System log:

20-02-18 20:45:50 ERROR (MainThread) [hassio.api.ingress] Ingress error: Cannot connect to host 172.30.32.1:62939 ssl:None [Operation timed out]

```

**Additional information and things you've tried:**

KustHardball, 2020-02-26:
Hello! Same issue. Did you find a solution?

rjalexa, 2020-02-26:
No Sir. This is making me CRAZY !!! Now the crazy thing is that the board does connect without any problems to another AP (a WiFi extender) I have at home. At the same time the main modem/router AP does not seem to have any blacklisting capabilities ... no clue so far.

KustHardball, 2020-02-26:
I think problem with a docker ports settings. I have same situation when trying to connect remotely and on local mashine. Maybe we need to change this settings?
[image]

KustHardball, 2020-02-26:
You can define port on Host field and ESPhome dashboard will be avaliable, but only on localhost.

KustHardball, 2020-02-26:
If you define port in Host field, you can get access to ESPhome dashbord without Home Assistant.

rjalexa, 2020-02-27:
Yes Sir. I tried with 6055 and then http://serverIP:6055 from a browser does work but still would love to get it working within HA.

glmnet, 2020-02-27:
Seems a ingress home assistant issue. I assume you already tried a full system restart?

rjalexa, 2020-02-27:
Hi Gimnet yes of course many system restarts, full HA reinstalls (wiping away all dockers and config files). PRoblem stays the same. I am armed with Portainer to debug but do not have the skills :( Wonder if anyone might start explaining what exactly happens when I click on the "Start UI" button.

Sergksvs, 2020-03-14:
Hi rjalexa.  I have also error. You are resolve it?

rjalexa, 2020-03-14:
Yes I have solved it :)
Do the following
Try to start the Web UI and wait for the 502 error and when you get it
Go to the supervisor and scroll down to the last lines of the system log
There you will find a red line of an error with a port number (eg 65432)
use the proper command to allow your operating system firewall to grant access to that port

This should fix the problem.
In my case I had "hardened" IPTables but the 172.x.y.z network wanted to access that port.
Hope this helps you. Let me know please.

closed 2020-03-14 by rjalexa