esphome/issues #1057: Wifi disconnects with reason 'Auth Expired'

rjalexa, 2020-02-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**esphome run in command line mode on a Mac (Sierra) against a NodeMCU ESP32S-WROOM v1.1 board with a minimal yaml file just to flash the board and get it to connect to home wifi AP (SSID rjanet)**

**ESP (ESP32/ESP8266, Board/Sonoff):**

 **NodeMCU ESP32S-WROOM v1.1 board** 
**Affected component:**

**Description of problem:**
Board used to connect just fine with same yaml file, now it refuses to connect with following messages:
[16:03:10][W][wifi_esp32:336]: Event: Disconnected ssid='rjanet' bssid=94:91:7F:6A:09:63 reason='Auth Expired'
[16:03:10][W][wifi:451]: Error while connecting to network.
[16:03:10][W][wifi:488]: Restarting WiFi adapter...

Tried rebooting the modem/AP, tried flashing the board with a different name.

All other computers and a Tasmota 8266 board connect just fine with same credentials.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: bagno
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "rjanet"
  password: "myverysecretpass"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bagno Fallback Hotspot"
    password: "5ew7OVLtpMiZ"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
```

**Logs (if applicable):**

```
(bagno) (base) iMac:iot bob$ esphome bagnosolowifi.yaml run
INFO Reading configuration bagnosolowifi.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting bagno/.pioenvs
INFO Deleting bagno/.piolibdeps
INFO Compiling app...
INFO Running:  platformio run -d bagno
Please wait while upgrading PlatformIO...
Updating tool-unity                      @ 1.20403.0      [1.20500.0]
Uninstalling tool-unity @ 1.20403.0: 	[OK]
CorePackageManager: Installing tool-unity @ 1.20500.0
tool-unity @ 1.20500.0 has been successfully installed!
Updating tool-scons                      @ 3.30101.0      [3.30102.0]
Uninstalling tool-scons @ 3.30101.0: 	[OK]
CorePackageManager: Installing tool-scons @ 3.30102.0
tool-scons @ 3.30102.0 has been successfully installed!
PlatformIO has been successfully upgraded to 4.2.1!

********************************************************************************
[... 269 lines cut ...]
[16:03:08][D][wifi:369]: - 'TIM-53331562' (78:81:02:99:F8:FB) ▂▄▆█
[16:03:08][D][wifi:369]: - 'TIM-53331562_2GEXT' (B0:39:56:B7:A0:F3) ▂▄▆█
[16:03:08][D][wifi:369]: - 'Zamunda Fibra' (18:A6:F7:F4:09:80) ▂▄▆█
[16:03:08][I][wifi:193]: WiFi Connecting to 'rjanet'...
[16:03:10][W][wifi_esp32:336]: Event: Disconnected ssid='rjanet' bssid=94:91:7F:6A:09:63 reason='Auth Expired'
[16:03:10][W][wifi:451]: Error while connecting to network.
[16:03:10][W][wifi:488]: Restarting WiFi adapter...
[16:03:10][I][wifi:193]: WiFi Connecting to 'rjanet'...
[16:03:12][W][wifi_esp32:336]: Event: Disconnected ssid='rjanet' bssid=94:91:7F:6A:09:63 reason='Auth Expired'
[16:03:12][W][wifi:451]: Error while connecting to network.
[16:03:12][W][wifi:488]: Restarting WiFi adapter...
[16:03:17][D][wifi:304]: Starting scan...
[16:03:18][D][wifi:319]: Found networks:
[16:03:18][I][wifi:365]: - 'rjanet' (94:91:7F:6A:09:63) ▂▄▆█
[16:03:18][D][wifi:366]:     Channel: 6
[16:03:18][D][wifi:367]:     RSSI: -64 dB
[16:03:18][D][wifi:369]: - 'DIRECT-2B-HP OfficeJet 3830' (C8:D9:D2:44:A6:2C) ▂▄▆█
[16:03:18][D][wifi:369]: - 'WOW FI - FASTWEB' (62:91:7F:6A:09:60) ▂▄▆█
[16:03:18][D][wifi:369]: - 'Zamunda Fibra' (A4:91:B1:39:6E:53) ▂▄▆█
[16:03:18][D][wifi:369]: - 'TIM-53331562' (78:81:02:99:F8:FB) ▂▄▆█
[16:03:18][D][wifi:369]: - 'TIM-53331562_2GEXT' (B0:39:56:B7:A0:F3) ▂▄▆█
[16:03:18][D][wifi:369]: - 'Zamunda Fibra' (18:A6:F7:F4:09:80) ▂▄▆█
```

**Additional information and things you've tried:**

jonn26, 2020-03-30:
I am also having this same issue with my ESP32s (M5stick & M5Stick-c). It will connect once to my AP if it is an ESP32 that hasn't been connected in a long time, but, then if I update the firmware, it will fail to connect with reason "Auth Expired" no matter what I do. Even rebooting the AP doesn't seem to make a difference.

@rjalexa Do you happen to have a UniFi AP? I have a UniFi AP-AC-LR AP and am wondering if that might have something to do with it..

jonn26, 2020-03-30:
I am on a mac and was able to fix this with `python -m platformio platform update`. After doing that no more "Auth Expired".

```
$ python -m platformio platform update
**************************************************************************************************************************************************************************************
Obsolete PIO Core v4.0.3 is used (previous was 4.2.1)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
**************************************************************************************************************************************************************************************
Platform Espressif 32
--------
Updating espressif32                     @ 1.11.1         [1.12.0]
Uninstalling espressif32 @ 1.11.1:      [OK]
PlatformManager: Installing espressif32 @ 1.12.0
espressif32 @ 1.12.0 has been successfully installed!
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Updating tool-mkspiffs                   @ 2.230.0        [Up-to-date]
PackageManager: Installing framework-arduinoespressif32 @ ~3.10004.191002
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
framework-arduinoespressif32 @ 3.10004.200129 has been successfully installed!

Platform Espressif 32
--------
Updating espressif32                     @ 1.11.0         [Detached]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating framework-arduinoespressif32    @ 2.10004.191002 [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Updating tool-mkspiffs                   @ 2.230.0        [Up-to-date]

Platform Espressif 32
--------
Updating espressif32                     @ 1.6.0          [Detached]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating framework-arduinoespressif32    @ 2.10001.190111 [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Updating tool-espotapy                   @ 1.1.3          [Up-to-date]
Updating tool-mkspiffs                   @ 2.230.0        [Up-to-date]

Platform Espressif 8266
--------
Updating espressif8266                   @ 2.3.1          [2.4.0]
Uninstalling espressif8266 @ 2.3.1:     [OK]
PlatformManager: Installing espressif8266 @ 2.4.0
espressif8266 @ 2.4.0 has been successfully installed!
Updating toolchain-xtensa                @ 2.40802.191122 [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-esptoolpy                  @ 1.20800.0      [Up-to-date]
Updating tool-mkspiffs                   @ 1.200.0        [Up-to-date]
Uninstalling framework-arduinoespressif8266 @ 2.20602.191128:   [OK]
PackageManager: Installing framework-arduinoespressif8266 @ ~3.20603.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
framework-arduinoespressif8266 @ 3.20603.200130 has been successfully installed!

Platform Espressif 8266
--------
Updating espressif8266                   @ 1.8.0          [Detached]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]
Updating framework-arduinoespressif8266  @ 2.20402.4      [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-espotapy                   @ 1.1.3          [Up-to-date]
Updating tool-mkspiffs                   @ 1.200.0        [Up-to-date]

Platform Espressif 8266
--------
Updating espressif8266                   @ 2.2.3          [Detached]
Updating toolchain-xtensa                @ 2.40802.191122 [Up-to-date]
Updating framework-arduinoespressif8266  @ 2.20502.0      [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Updating tool-mkspiffs                   @ 1.200.0        [Up-to-date]
```

rjalexa, 2020-03-30:
Thank you very much. Will try. No, not a Ubiquiti Unify but an ISP provided modem (Fastgate by ASKEY). Now have bought a Fritzbox 7590 and have to try both the new WiFi and also the update you suggest. Will do it in the next weeks as time during the epidemic is really scarce. Apologize.

jonn26, 2020-03-30:
It seems that updating  platformio was only a temporary fix. Today it is back to "Auth Expired" :frowning:

randybb, 2020-03-31:
I have Unify APs and around 20 ESPHome devices running without any issue.

jonn26, 2020-04-01:
@randybb Are any of your ESPHome devices ESP32s? I have many esp8266 devices connected to the same Unifi AP and they don't have this issue.

Here is the ESPHome log output I am seeing on one of my ESP32s. I hard reset it and it connected to wifi, then ~15m later it rebooted and it starts failing for reason='Auth Expired'
```
[logger:156]: Log initialized
[21:30:55][C][ota:364]: There have been 4 suspected unsuccessful boot attempts.
[21:30:55][I][app:028]: Running through setup()...
[21:30:55][D][binary_sensor:034]: 'Master Bath BME280 Status': Sending initial state OFF
[21:30:55][C][bme280.sensor:081]: Setting up BME280...
[21:30:55][C][wifi:033]: Setting up WiFi...
[21:30:56][D][wifi:304]: Starting scan...
[21:30:56][D][bme280.sensor:197]: Got temperature=25.9°C pressure=951.8hPa humidity=40.5%
[21:30:56][D][sensor:092]: 'Master Bath BME280 Temperature': Sending state 25.88000 °C with 1 decimals of accuracy
[21:30:56][D][sensor:092]: 'Master Bath BME280 Pressure': Sending state 951.82623 hPa with 1 decimals of accuracy
[21:30:56][D][sensor:092]: 'Master Bath BME280 Humidity': Sending state 40.49121 % with 1 decimals of accuracy
[21:30:57][D][wifi:319]: Found networks:
[21:30:57][I][wifi:365]: - 'NeuroNet5' (0E:EC:DA:B1:74:EC) ▂▄▆█
[21:30:57][D][wifi:366]:     Channel: 1
[21:30:57][D][wifi:367]:     RSSI: -63 dB
[21:30:57][D][wifi:369]: - 'NTGR_VMB_1542429930' (10:DA:43:BC:76:08) ▂▄▆█
[21:30:57][D][wifi:369]: - '' (FE:A6:67:A2:20:DF) ▂▄▆█
[21:30:57][D][wifi:369]: - 'DwightYouIgnorantSlut' (FC:EC:DA:B1:74:EC) ▂▄▆█
[21:30:57][D][wifi:369]: - 'WildWest' (FE:EC:DA:B1:74:EC) ▂▄▆█
[21:30:57][D][wifi:369]: - '' (1E:EC:DA:B1:74:EC) ▂▄▆█
[21:30:57][D][wifi:369]: - '' (FA:8F:CA:50:95:31) ▂▄▆█
[21:30:57][D][wifi:369]: - '16D302' (B0:93:5B:16:D3:02) ▂▄▆█
[21:30:58][D][wifi:369]: - 'lisaankle' (84:17:EF:EC:4E:A3) ▂▄▆█
[21:30:58][D][wifi:369]: - '' (84:17:EF:EC:4E:A6) ▂▄▆█
[21:30:58][D][wifi:369]: - 'NETGEAR17' (A0:63:91:9D:FF:12) ▂▄▆█
[21:30:58][D][wifi:369]: - 'NeoroNet112' (10:DA:43:C1:F5:C4) ▂▄▆█
[21:30:58][D][wifi:369]: - '' (B2:FC:0D:AE:83:E7) ▂▄▆█
[21:30:58][D][wifi:369]: - '' (B2:93:5B:16:D3:02) ▂▄▆█
[21:30:58][D][wifi:369]: - '' (1E:12:B0:5A:23:C6) ▂▄▆█
[21:30:58][D][wifi:369]: - '' (F2:93:5B:16:D3:02) ▂▄▆█
[21:30:58][D][wifi:369]: - '' (2E:D9:74:B8:2B:51) ▂▄▆█
[21:30:58][D][wifi:369]: - '' (D2:93:5B:16:D3:02) ▂▄▆█
[21:30:58][D][wifi:369]: - '' (3E:51:A4:D1:27:D9) ▂▄▆█
[21:30:58][D][wifi:369]: - 'Humbugspa' (FC:51:A4:D1:27:D9) ▂▄▆█
[21:30:58][D][wifi:369]: - '' (FE:51:A4:D1:27:D9) ▂▄▆█
[21:30:58][D][wifi:369]: - '' (D4:B9:2F:5B:08:87) ▂▄▆█
[21:30:58][D][wifi:369]: - '' (D4:B9:2F:5B:08:8B) ▂▄▆█
[21:30:58][D][wifi:369]: - '' (1E:51:A4:D1:27:D9) ▂▄▆█
[21:30:58][D][wifi:369]: - '' (54:A6:5C:AA:FF:BE) ▂▄▆█
[21:30:58][D][wifi:369]: - '' (54:A6:5C:AA:FF:C0) ▂▄▆█
[21:30:58][D][wifi:369]: - 'Anakin' (54:A6:5C:AA:FF:BD) ▂▄▆█
[21:30:58][D][wifi:369]: - '' (54:A6:5C:AA:FF:C2) ▂▄▆█
[21:30:58][I][wifi:193]: WiFi Connecting to 'NeuroNet5'...
[21:30:59][I][wifi:423]: WiFi Connected!
[21:30:59][C][wifi:283]:   SSID: 'NeuroNet5'
[21:30:59][C][wifi:284]:   IP Address: 192.168.5.12
[21:30:59][C][wifi:286]:   BSSID: 0E:EC:DA:B1:74:EC
[21:30:59][C][wifi:287]:   Hostname: 'mb_bme280'
[21:30:59][C][wifi:291]:   Signal strength: -67 dB ▂▄▆█
[21:30:59][C][wifi:295]:   Channel: 1
[21:30:59][C][wifi:296]:   Subnet: 255.255.255.0
[21:30:59][C][wifi:297]:   Gateway: 192.168.5.1
[21:30:59][C][wifi:298]:   DNS1: 192.168.5.1
[21:30:59][C][wifi:299]:   DNS2: 0.0.0.0
[21:30:59][C][ota:029]: Over-The-Air Updates:
[21:30:59][C][ota:030]:   Address: mb_bme280.jonn26.lan:3232
[21:30:59][C][ota:032]:   Using Password.
[21:30:59][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 6 restarts
[21:30:59][C][api:022]: Setting up Home Assistant API server...
[21:30:59][I][app:058]: setup() finished successfully!
[21:30:59][I][app:100]: ESPHome version 1.14.3 compiled on Mar 29 2020, 23:27:41
[21:30:59][C][wifi:415]: WiFi:
[21:30:59][C][wifi:283]:   SSID: 'NeuroNet5'
[21:30:59][C][wifi:284]:   IP Address: 192.168.5.12
[21:30:59][C][wifi:286]:   BSSID: 0E:EC:DA:B1:74:EC
[21:30:59][C][wifi:287]:   Hostname: 'mb_bme280'
[... 767 lines cut ...]

ota:
  password: !secret esphome_ota_pw

i2c:
  sda: 25
  scl: 13
  scan: True

binary_sensor:
  - platform: status
    name: $upper_devicename Status

sensor:
  - platform: bme280
    temperature:
      name: $upper_devicename Temperature
      oversampling: 16x
    pressure:
      name: $upper_devicename Pressure
    humidity:
      name: "$upper_devicename Humidity"
    address: 0x76
    update_interval: 60s
```

randybb, 2020-04-01:
ESP-32s is a clone of ESP-WROOM-32 that I use over the place, so basically we have the same. I believe the problem is in weak wifi signal you have (RSSI: -67 dB). I previously had some connection issues, but then I added one more AP and don't have any issues anymore. Try to move the device closer to AP or try to switch to ESP32 with external antenna (e.g. ESP-WROOM-32U).

rjalexa, 2020-04-01:
This is a very interesting hypothesis. Will do some experimenting and come back with results. Thank you very much.

rjalexa, 2020-04-01:
Well first experiment bodes well to @randybb hypothesis. I have a new AP now (a mesh network with Fritz 7590 and 1750E repeater) and the connection has a good signal now:
[13:31:23][I][app:100]: ESPHome version 1.14.3 compiled on Apr  1 2020, 13:30:30
[13:31:23][C][wifi:415]: WiFi:
[13:31:23][C][wifi:283]:   SSID: 'rjanet'
[13:31:23][C][wifi:284]:   IP Address: 192.168.1.54
[13:31:23][C][wifi:286]:   BSSID: 98:9B:CB:E2:3E:9B
[13:31:23][C][wifi:287]:   Hostname: 'bagno'
[13:31:23][C][wifi:291]:   Signal strength: -44 dB ▂▄▆█
and so far the NodeMCU ESP32S-WROOM v1.1 board seems to be working correctly.
Maybe it would be good instead of issuing those errors just to suggest a message on signal strength?
Thank you.

rjalexa, 2020-04-01:
Well no quite :)
Things are a bit more complex. First of all let me describe my home WiFi.
The same SSID (rjanet) is served by both the Modem (MAC DC:39:6F:EF:E9:DE) and the repeater (MAC 98:9B:CB:E2:3E:9B) and while doing this troubleshooting I am closer to the latter (hence the stronger signal). 
If you look at the following trace it will show that the dreaded auth error prevented the connection (even with the same good signal) but unlike my previous modem after a couple of retries it succeeded. Does this shed any additional light?
Hard resetting via RTS pin...
========================= [SUCCESS] Took 23.94 seconds =========================
INFO Successfully uploaded program.
INFO Starting log output from /dev/cu.SLAB_USBtoUART with baud rate 115200
[13:48:16][I][logger:156]: Log initialized
[13:48:16][C][ota:364]: There have been 3 suspected unsuccessful boot attempts.
[13:48:16][I][app:028]: Running through setup()...
[13:48:16][C][dht:011]: Setting up DHT...
[13:48:16][C][wifi:033]: Setting up WiFi...
[13:48:16][D][wifi:304]: Starting scan...
[13:48:16][D][dht:048]: Got Temperature=19.3°C Humidity=1.0%
[13:48:16][D][sensor:092]: 'Bagno T': Sending state 19.30000 °C with 1 decimals of accuracy
[13:48:16][D][sensor:092]: 'Bagno H': Sending state 1.00000 % with 0 decimals of accuracy
[13:48:18][D][wifi:319]: Found networks:
[13:48:18][I][wifi:365]: - 'rjanet' (98:9B:CB:E2:3E:9B) ▂▄▆█
[13:48:18][D][wifi:366]:     Channel: 1
[13:48:18][D][wifi:367]:     RSSI: -47 dB
[13:48:18][I][wifi:365]: - 'rjanet' (DC:39:6F:EF:E9:DE) ▂▄▆█
[13:48:18][D][wifi:366]:     Channel: 1
[13:48:18][D][wifi:367]:     RSSI: -57 dB
[13:48:18][D][wifi:369]: - 'DIRECT-2B-HP OfficeJet 3830' (C8:D9:D2:44:A6:2C) ▂▄▆█
[13:48:18][D][wifi:369]: - 'Zamunda Fibra' (A4:91:B1:39:6E:53) ▂▄▆█
[13:48:18][D][wifi:369]: - 'TIM-53331562' (78:81:02:99:F8:FB) ▂▄▆█
[13:48:18][D][wifi:369]: - 'TIM-53331562_2GEXT' (B0:39:56:B7:A0:F3) ▂▄▆█
[13:48:18][I][wifi:193]: WiFi Connecting to 'rjanet'...
[13:48:19][W][wifi_esp32:336]: Event: Disconnected ssid='rjanet' bssid=98:9B:CB:E2:3E:9B reason='Authentication Failed'
[13:48:19][W][wifi:470]: Connecting to WiFi network failed. Are the credentials wrong?
[13:48:19][I][wifi:193]: WiFi Connecting to 'rjanet'...
[13:48:19][W][wifi:470]: Connecting to WiFi network failed. Are the credentials wrong?
[13:48:24][D][wifi:304]: Starting scan...
[13:48:26][D][wifi:319]: Found networks:
[13:48:26][I][wifi:365]: - 'rjanet' (DC:39:6F:EF:E9:DE) ▂▄▆█
[13:48:26][D][wifi:366]:     Channel: 1
[13:48:26][D][wifi:367]:     RSSI: -58 dB
[13:48:26][I][wifi:365]: - 'rjanet' (98:9B:CB:E2:3E:9B) ▂▄▆█
[13:48:26][D][wifi:366]:     Channel: 1
[13:48:26][D][wifi:367]:     RSSI: -45 dB
[13:48:26][D][wifi:369]: - 'Zamunda Fibra' (A4:91:B1:39:6E:53) ▂▄▆█
[13:48:26][D][wifi:369]: - 'TIM-53331562' (78:81:02:99:F8:FB) ▂▄▆█
[13:48:26][D][wifi:369]: - 'Fritz-Router2,4GHz' (E8:DF:70:69:20:7C) ▂▄▆█
[13:48:26][D][wifi:369]: - 'TIM-53331562_2GEXT' (B0:39:56:B7:A0:F3) ▂▄▆█
[13:48:26][D][wifi:369]: - 'DIRECT-2B-HP OfficeJet 3830' (C8:D9:D2:44:A6:2C) ▂▄▆█
[13:48:26][I][wifi:193]: WiFi Connecting to 'rjanet'...
[13:48:26][I][wifi:423]: WiFi Connected!
[13:48:26][C][wifi:283]:   SS: Event: Disconnec[W][wifi_esp32:336]: Event: Disconnected ssid='rjanet' bssid=98:9B:CB:E2:3E:9B reason='Association Leave'
[13:48:26]
[13:48:26][C][wifi:284]:   IP Address: 0.0.0.0
[13:48:26][C][wifi:286]:   BSSID: 00:00:00:00:00:00
[13:48:26][C][wifi:287]:   Hostname: 'bagno'
[13:48:26][C][wifi:291]:   Signal strength: 0 dB ▂▄▆█
[13:48:26][C][wifi:295]:   Channel: 1
[13:48:26][C][wifi:296]:   Subnet: 0.0.0.0
[13:48:26][C][wifi:297]:   Gateway: 0.0.0.0
[13:48:26][C][wifi:298]:   DNS1: 192.168.1.254
[13:48:26][C][wifi:299]:   DNS2: 0.0.0.0
[13:48:26][D][wifi:432]: Disabling AP...
[13:48:26][W][wifi:100]: WiFi Connection lost... Reconnecting...
[13:48:28][W][wifi_esp32:336]: Event: Disconnected ssid='rjanet' bssid=DC:39:6F:EF:E9:DE reason='Auth Expired'
[13:48:31][D][wifi:304]: Starting scan...
[13:48:32][D][wifi:319]: Found networks:
[13:48:32][I][wifi:365]: - 'rjanet' (DC:39:6F:EF:E9:DE) ▂▄▆█
[13:48:32][D][wifi:366]:     Channel: 1
[13:48:32][D][wifi:367]:     RSSI: -58 dB
[13:48:32][I][wifi:365]: - 'rjanet' (98:9B:CB:E2:3E:9B) ▂▄▆█
[13:48:32][D][wifi:366]:     Channel: 1
[13:48:32][D][wifi:367]:     RSSI: -48 dB
[... 55 lines cut ...]
[13:48:41][C][logger:176]:   Level: DEBUG
[13:48:41][C][logger:177]:   Log Baud Rate: 115200
[13:48:41][C][logger:178]:   Hardware UART: UART0
[13:48:41][C][dht:017]: DHT:
[13:48:41][C][dht:018]:   Pin: GPIO4 (Mode: INPUT)
[13:48:41][C][dht:024]:   Model: DHT22 (or equivalent)
[13:48:41][C][dht:027]:   Update Interval: 300.0s
[13:48:41][C][dht:029]:   Temperature 'Bagno T'
[13:48:41][C][dht:029]:     Unit of Measurement: '°C'
[13:48:41][C][dht:029]:     Accuracy Decimals: 1
[13:48:41][C][dht:029]:     Icon: 'mdi:thermometer'
[13:48:41][C][dht:030]:   Humidity 'Bagno H'
[13:48:41][C][dht:030]:     Unit of Measurement: '%'
[13:48:41][C][dht:030]:     Accuracy Decimals: 0
[13:48:41][C][dht:030]:     Icon: 'mdi:water-percent'
[13:48:41][C][captive_portal:169]: Captive Portal:
[13:48:41][C][ota:029]: Over-The-Air Updates:
[13:48:41][C][ota:030]:   Address: bagno.local:3232
[13:48:41][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 7 restarts
[13:48:41][C][api:095]: API Server:
[13:48:41][C][api:096]:   Address: bagno.local:6053
[13:50:16][I][ota:046]: Boot seems successful, resetting boot loop counter.
[13:51:40][D][dht:048]: Got Temperature=19.3°C Humidity=1.0%
[13:51:40][D][sensor:092]: 'Bagno T': Sending state 19.30000 °C with 1 decimals of accuracy
[13:51:40][D][sensor:092]: 'Bagno H': Sending state 1.00000 % with 0 decimals of accuracy

rjalexa, 2020-04-01:
Anyways with this better signal strength (Fritzbox 7590 + 1750E repeater meshed) the units do connect even if sometimes the hiccup so I'd say the problem is solved for me (but the code could handle the low signal situation more gracefully perhaps).
Thank you all.

closed 2020-04-01 by rjalexa