esphome/issues #1059: Using remote transmitter with fastled crashes ESP32

Mr-Quin, 2020-02-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, esp32dev

**Affected component:**

remote_transmitter, remote_receiver

**Description of problem:**
Calling _remote_transmitter.transmit_*_, while having the fastled component, crashes the ESP 32. 
The leds does not need to be on. Logs below. 

Neopixelbus works. So far it's just fastled that's having problems

It works without the fastled component.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: light_controller
  platform: ESP32
  board: esp32dev
  
wifi:
  ssid: ${wifi_ssid}
  password: ${wifi_pass}
  fast_connect: true
  power_save_mode: none

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Fallback Hotspot"
    password: "TxKInYmmLNQq"

remote_receiver:
  pin:
    number: GPIO25
    inverted: true
  dump: all
  on_raw:
    then:
        - switch.turn_on: ir_led
        - delay: 20ms
        - switch.turn_off: ir_led
        
remote_transmitter:
  pin: GPIO33
  # Infrared remotes use a 50% carrier signal
  carrier_duty_percent: 50%
      
switch:
  - platform: template
    name: Raw Code Power Button
    id: LG_on
    turn_on_action:
      - remote_transmitter.transmit_lg:
          data: 0x20DF10EF
          nbits: 32
          repeat:
            times: 2
            wait_time: 100ms
  - platform: gpio
    pin: GPIO5
    name: LED
    id: ir_led
    #internal: true
          
esp32_touch:
  setup_mode: false

binary_sensor:
  - platform: esp32_touch
    name: "PIN GPIO2"
    pin: GPIO32
    threshold: 1000
    on_press:
      then:
        - switch.turn_on: ir_led
    on_release:
      then:
        - switch.turn_off: ir_led

light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: GPIO17
    num_leds: 60
    rgb_order: GRB
    effects:
      - addressable_rainbow:
          name: Rainbow
          speed: 10
          width: 100
      - addressable_flicker:
          name: Flicker
          update_interval: 50ms
          intensity: 60%
      - strobe:
      - addressable_scan:
      - addressable_twinkle:
          name: Twinkle
          twinkle_probability: 40%
          progress_interval: 10ms
      - addressable_random_twinkle:
          name: Random Twinkle
          twinkle_probability: 40%
          progress_interval: 20ms
    name: "FastLED"
    
time:
  - platform: homeassistant
    id: homeassistant_time
  
captive_portal:

debug:

logger:
  level: debug

# Enable Home Assistant API
api:

ota:

<<: !include ./config/secrets.yaml
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[20:34:57][D][switch:021]: 'Raw Code Power Button' Turning ON.
[20:34:57][D][esp-idf:000]: E (575) rmt: rmt_driver_install(725): RMT driver already installed for channel
[20:34:57][E][component:092]: Component was marked as failed.
[20:34:57]E (590) rmt: RMT[0] ERR
[20:34:57]E (591) rmt: status: 0x08000001
[20:34:57]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[20:34:57]Core 1 register dump:
[20:34:57]PC      : 0x400e6899  PS      : 0x00060031  A0      : 0x80085429  A1      : 0x3ffbe770
[32mINFO Need to fetch platformio IDE-data, please stand by[0m
[32mINFO Running:  platformio run -d '.\light_controller' -t idedata[0m
Traceback (most recent call last):
  File "C:\Python38\Scripts\esphome-script.py", line 11, in <module>
    load_entry_point('esphome==1.15.0.dev0', 'console_scripts', 'esphome')()
  File "c:\python38\lib\site-packages\esphome\__main__.py", line 557, in main
    return run_esphome(sys.argv)
  File "c:\python38\lib\site-packages\esphome\__main__.py", line 544, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "c:\python38\lib\site-packages\esphome\__main__.py", line 308, in command_logs
    return show_logs(config, args, port)
  File "c:\python38\lib\site-packages\esphome\__main__.py", line 207, in show_logs
    run_miniterm(config, port)
  File "c:\python38\lib\site-packages\esphome\__main__.py", line 113, in run_miniterm
    backtrace_state = platformio_api.process_stacktrace(
  File "c:\python38\lib\site-packages\esphome\platformio_api.py", line 213, in process_stacktrace
    _parse_register(config, STACKTRACE_ESP32_PC_RE, line)
  File "c:\python38\lib\site-packages\esphome\platformio_api.py", line 187, in _parse_register
    _decode_pc(config, match.group(1))
  File "c:\python38\lib\site-packages\esphome\platformio_api.py", line 166, in _decode_pc
    idedata = get_idedata(config)
  File "c:\python38\lib\site-packages\esphome\platformio_api.py", line 121, in get_idedata
    IDE_DATA = run_idedata(config)
  File "c:\python38\lib\site-packages\esphome\platformio_api.py", line 101, in run_idedata
    stdout = run_platformio_cli_run(config, False, *args, capture_stdout=True).decode()
AttributeError: 'str' object has no attribute 'decode'
```

**Additional information and things you've tried:**
I have a couple of the same board, all of them are like this. Tried different pins, same result.  
This might be related: while the fastled component is present, remote receiver throws an rmt error upon receiving any signal, then stops working.
I used this same exact YAML config (with different board and pin setups) on a ESP8266 and it works flawlessly.

Remote receiver outputs this:
```
[20:52:07]E (2351) rmt: RMT[0] ERR
[20:52:07]E (2351) rmt: status: 0x0c000000
```

poelzi, 2020-03-09:
This is basically the same problem as #858

poelzi, 2020-07-24:
still exists

OttoWinter, 2020-07-24:
Fixed in https://github.com/esphome/esphome/pull/923

closed 2020-07-24 by OttoWinter