esphome/issues #1060: Home Assistant official ESPHome Add-on doesnt't upload binary to ESP devices (OTA).

kovaccs, 2020-02-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Raspberry Pi 3B
Hassos 3.11
Home Assitant Core: 0.105.5  (a lower version does the same issue)

**Affected component:**

**Description of problem:**
After create, or edit ESPHome device configuration, the UPLOAD function starting, and stop soon.
(attached pic.)
The configurations are compiled correctly. There can be upload the downloaded binary through serial interface, and the device is working fine.
The current devices - it's my experience - Wemos D1, ESP32S. 
It's interesting that Sonoff equipments do not cause this problems. 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
#
esphome:
  name: wemos_d1_06
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pw
# ...
  manual_ip:
    static_ip: !secret wemos_d1_06_ip
    gateway: !secret lan_gw
    subnet: !secret lan_netmask
    dns1: !secret lan_dns1
    dns2: !secret lan_dns2

# Enable logging
logger:
  level: DEBUG
#  level: ERROR
# Enable Home Assistant API
api:
  password: !secret esphome_api_pw

ota:
  password: !secret esphome_ota_pw
#
#    
binary_sensor:
  - platform: gpio
    pin:
      number: 6
      mode: INPUT_PULLUP
      inverted: True
    name: "wemos_06_switch"
  - platform: gpio
    pin:
      number: 2
      mode: INPUT_PULLUP
      inverted: True
    name: "wemos_06_IR_kapu"
#
output:
  # Register the blue LED as a dimmable output ....
  - platform: esp8266_pwm
    id: blue_led_w06
    pin: 4
    frequency: 200 Hz
    inverted: True
#
light:
  # ... and then make a light out of it.
  - platform: monochromatic
    name: "WemosD1_06 Blue LED"
    output: blue_led_w06
#
sensor:
  - platform: wifi_signal
    name: "Iroda WiFi jelerősség"
    update_interval: 60s
#    

```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/wemos_d1_06.yaml...
WARNING ESP8266: Pin 6 (6-11) might already be used by the flash interface. Be warned.
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/wemos_d1_06
Processing wemos_d1_06 (board: esp01_1m; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
Compiling /data/wemos_d1_06/.pioenvs/wemos_d1_06/src/main.cpp.o
Linking /data/wemos_d1_06/.pioenvs/wemos_d1_06/firmware.elf
Retrieving maximum program size /data/wemos_d1_06/.pioenvs/wemos_d1_06/firmware.elf
Checking size /data/wemos_d1_06/.pioenvs/wemos_d1_06/firmware.elf
Building /data/wemos_d1_06/.pioenvs/wemos_d1_06/firmware.bin
DATA:    [====      ]  43.7% (used 35816 bytes from 81920 bytes)
PROGRAM: [====      ]  38.2% (used 391392 bytes from 1023984 bytes)
Creating BIN file "/data/wemos_d1_06/.pioenvs/wemos_d1_06/firmware.bin" using "/data/wemos_d1_06/.pioenvs/wemos_d1_06/firmware.elf"
========================= [SUCCESS] Took 24.46 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 172.30.1.16
INFO Uploading /data/wemos_d1_06/.pioenvs/wemos_d1_06/firmware.bin (395552 bytes)
Uploading: [==                                                          ] 3% 
ERROR Error sending data: [Errno 104] Connection reset by peer
```

**Additional information and things you've tried:**
I tried to change board: from d1_mini to esp01_1m without success.

kovaccs, 2020-02-25:
Nobody has a comment on this topic?

jaisgit, 2020-03-03:
All I can offer is that I've been experiencing the same issue with the d1_mini_pro OTA updates and can only update via the serial.

Has anyone else experienced this and managed to get it going?

pszafer, 2020-03-04:
I have similar problem.
https://github.com/esphome/issues/issues/857

Workaround is to update via `web_server`. Almost always working.

kovaccs, 2020-03-05:
pszafer@ Sorry, what does it mean exacty:  "Workaround is to update via web_server"?
The problem is persist!!!

pszafer, 2020-03-05:
It means that while there is no fix for this problem, you can use workaround so you don't have to connect your device via serial.
PR is already there: https://github.com/esphome/esphome/pull/959 but it is waiting for Otto to check it I think.

kovaccs, 2020-03-05:
Thank's for this info!

Jamiro75, 2020-03-15:
i've same problem, 
with the SONOFF basic  with "esp01_1m" works
with a simple 1 relay generic board with a esp01, with the same configuration "esp01_1m" works
but
if i use a wemos d1mini with:
  esp01_1m, d1_mini, nodemcu,  nodemcuv2
not working 
i'm using hassio 0.106.6
ESPHome1.14.3  
many thanks

kovaccs, 2020-03-24:
Every day the first upload is always successful, the followers never.

nikosiaf, 2020-04-22:
I can confirm that OTA uptades are not working for me, trying on several wemos d1mini.
I have tried both the webserver and the esphome addon (1.14.3) for hassio (0.108.6) and always get the following error:

[14:37:16][W][web_server_base:021]: OTA Update failed! Error: ERROR[7]: MD5 Failed: expected:322bd85d1caabd3fc5320d9f32d76fff, calculated:1f21000f3b9b94b69ecf7df6121ba8e4

ERROR Error Update end: Error: Finishing update failed. See the MQTT/USB logs for more information.

pszafer, 2020-04-22:
use dev version of esphome. it's already been fox8

glmnet, 2020-04-22:
Please report back if it's working for you in dev, otherwise you'll have to provide the serial (usb) logs while you attempt the OTA.

There is no much we can do for release version: https://github.com/esphome/feature-requests/issues/658

kovaccs, 2020-04-22:
great solution! It works!
When is the production upgrade expected?

> Please report back if it's working for you in dev, otherwise you'll have to provide the serial (usb) logs while you attempt the OTA.
> 
> There is no much we can do for release version: [esphome/feature-requests#658](https://github.com/esphome/feature-requests/issues/658)

Great solution! It works!
When is the production upgrade expected?

glmnet, 2020-04-22:
Please see esphome/feature-requests#658

closed 2020-04-22 by glmnet