esphome/issues #1061: Sending same position to time based cover cause it to move just slightly over and over again

Skaronator, 2020-02-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**
Shelly 2.5 which is a ESP8266 with 2 Relays and 2 Inputs

**Affected component:**

https://esphome.io/components/cover/time_based.html

**Description of problem:**

When sending the same position the a time based cover (which is already at set position) then the cover moves up and down for a few ms. I think this is due to a rounding issue but not really sure.

Here is a video with sound of the issue: https://imgur.com/a/uMD320g

Possible fix would be to check if the moving action is <1 step and then just ignore it.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
[22:31:31][D][cover:072]: 'Niklas Rolladen' - Setting
[22:31:31][D][cover:080]:   Position: 25%
[22:31:31][D][switch:021]: 'close_switch' Turning ON.
[22:31:31][D][switch:045]: 'close_switch': Sending state ON
[22:31:31][D][cover:152]: 'Niklas Rolladen' - Publishing:
[22:31:31][D][cover:155]:   Position: 25%
[22:31:31][D][cover:168]:   Current Operation: CLOSING
[22:31:31][D][switch:025]: 'open_switch' Turning OFF.
[22:31:31][D][switch:025]: 'close_switch' Turning OFF.
[22:31:31][D][switch:045]: 'close_switch': Sending state OFF
[22:31:31][D][cover:152]: 'Niklas Rolladen' - Publishing:
[22:31:31][D][cover:155]:   Position: 25%
[22:31:31][D][cover:168]:   Current Operation: IDLE
[22:31:31][D][cover:072]: 'Niklas Rolladen' - Setting
[22:31:31][D][cover:080]:   Position: 25%
[22:31:31][D][switch:021]: 'open_switch' Turning ON.
[22:31:31][D][switch:045]: 'open_switch': Sending state ON
[22:31:31][D][switch:025]: 'open_switch' Turning OFF.
[22:31:31][D][switch:045]: 'open_switch': Sending state OFF
[22:31:31][D][switch:025]: 'close_switch' Turning OFF.
[22:31:31][D][cover:152]: 'Niklas Rolladen' - Publishing:
[22:31:31][D][cover:155]:   Position: 25%
[22:31:31][D][cover:168]:   Current Operation: IDLE
[22:31:32][D][cover:072]: 'Niklas Rolladen' - Setting
[22:31:32][D][cover:080]:   Position: 25%
[22:31:32][D][switch:021]: 'close_switch' Turning ON.
[22:31:32][D][switch:045]: 'close_switch': Sending state ON
[22:31:32][D][switch:025]: 'open_switch' Turning OFF.
[22:31:32][D][switch:025]: 'close_switch' Turning OFF.
[22:31:32][D][switch:045]: 'close_switch': Sending state OFF
[22:31:32][D][cover:152]: 'Niklas Rolladen' - Publishing:
[22:31:32][D][cover:155]:   Position: 25%
[22:31:32][D][cover:168]:   Current Operation: IDLE
[22:31:32][D][cover:072]: 'Niklas Rolladen' - Setting
[22:31:32][D][cover:080]:   Position: 25%
[22:31:32][D][switch:021]: 'open_switch' Turning ON.
[22:31:32][D][switch:045]: 'open_switch': Sending state ON
[22:31:32][D][switch:025]: 'open_switch' Turning OFF.
[22:31:32][D][switch:045]: 'open_switch': Sending state OFF
[22:31:32][D][switch:025]: 'close_switch' Turning OFF.
[22:31:32][D][cover:152]: 'Niklas Rolladen' - Publishing:
[22:31:32][D][cover:155]:   Position: 25%
[22:31:32][D][cover:168]:   Current Operation: IDLE
[22:31:32][D][cover:152]: 'Niklas Rolladen' - Publishing:
[22:31:32][D][cover:155]:   Position: 25%
[22:31:32][D][cover:168]:   Current Operation: IDLE
[22:31:32][D][cover:072]: 'Niklas Rolladen' - Setting
[22:31:32][D][cover:080]:   Position: 25%
[22:31:32][D][switch:021]: 'close_switch' Turning ON.
[22:31:32][D][switch:045]: 'close_switch': Sending state ON
[22:31:32][D][switch:025]: 'open_switch' Turning OFF.
[22:31:32][D][switch:025]: 'close_switch' Turning OFF.
[22:31:32][D][switch:045]: 'close_switch': Sending state OFF
[22:31:32][D][cover:152]: 'Niklas Rolladen' - Publishing:
[22:31:32][D][cover:155]:   Position: 25%
[22:31:32][D][cover:168]:   Current Operation: IDLE

```

**Additional information and things you've tried:**

Skaronator, 2020-07-24:
Still an issue :/

Skaronator, 2020-11-22:
Not stale

jmacko, 2023-04-23:
Same issue here.  Anyone know how to handle this with a lambda as a workaround?

zd3sf, 2024-12-07:
Same issue here. For me its even worse because my cover is based on an RF command. The RF bridge cant issue a stop command fast enough, so the blinds go all the way up or all the way down. Very frustrating. 

This only happen when "set position action" is issued to the same current position. ex. automation set cover at 40% and its already at 40%

zd3sf, 2025-12-06:
I honestly don't expect a fix at this point. I will try to tackle the problem myself. Merging a pull request once took them 3 months to review and approve.