esphome/issues #1062: OTA update fails on Sparkfun ESP8266 Thing Dev Board

mcclown, 2020-02-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

- Docker version 17.09.1-ce, build e398b97
- Linux QNAP01 4.14.24-qnap #1 SMP Sat Oct 5 14:53:33 CST 2019 x86_64 GNU/Linux
- ESPHome 1.14.3
- ESPHome-Flasher-1.2-windows-x64.exe

**ESP (ESP32/ESP8266, Board/Sonoff):**

SparkFun ESP8266 Thing - Dev Board (with Headers) | WRL-13804

**Affected component:**

OTA Update - https://esphome.io/components/ota.html

**Description of problem:**

Steps to repo:

Step 1 - Compile "thingdev" - log 1
Step 2 - ESPHome-Flasher-1.2-windows-x64.exe - log 2
Step 3 - Reset (RST pin to GRD) -  log 2
Step 4 - Unplug device and put on 5v power. 1st boot - log 3
Step 5 - OTA Update fails - log 4

Result: 

ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.

Other Details:

I have tried all the same steps with "esp01_1m" specified as the board in the yaml. I still get exactly the same results.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

esphome:
  name: node01
  platform: ESP8266
  board: thingdev

wifi:
  ssid: "[redacted]"
  password: "[redacted]"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Node01 Fallback Hotspot"
    password: "[redacted]"

captive_portal:

debug:

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  password: "[redacted]"

ota:
  password: "[redacted]"

# Example configuration entry
dallas:
  - pin: 12

[... 375 lines cut ...]
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/node01/src/main.cpp.o
Linking .pioenvs/node01/firmware.elf
Building .pioenvs/node01/firmware.bin
Retrieving maximum program size .pioenvs/node01/firmware.elf
Checking size .pioenvs/node01/firmware.elf
DATA:    [====      ]  44.6% (used 36544 bytes from 81920 bytes)
PROGRAM: [========  ]  81.4% (used 406668 bytes from 499696 bytes)
Creating BIN file ".pioenvs/node01/firmware.bin" using ".pioenvs/node01/firmware.elf"
========================= [SUCCESS] Took 8.06 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of node01.local
INFO  -> 192.168.1.22
INFO Uploading /config/node01/.pioenvs/node01/firmware.bin (410816 bytes)
ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.

```

**Additional information and things you've tried:**

I have tried all the same steps with "esp01_1m" specified as the board in the yaml, as per the error message but I get the same error/results.

teno45, 2020-02-28:
Came across this issue due to having the same issue with the same board. Can get USB uploading to work no problems with: 
`esphome:
  name: sparkfun8266esp
  platform: ESP8266
  board: thingdev`

Only way I can get OTA to work, is with an initial usb flash, and subsequent OTA flashes with:
`esphome:
  name: sparkfun8266esp
  platform: ESP8266
  board: esp01_1m`

No idea if this has any lasting negative long term effects, but it works for the basic testing so far.

mcclown, 2020-02-29:
@teno45 I tried that, unfortunately I got the same result.

**Updated Config**

```YAML
esphome:
  name: node01
  platform: ESP8266
  board: esp01_1m
```

**Steps tried**

1. Flashed over USB using ESPHomeFlasher
2. Stay connected over USB and receiving serial log via ESPHome Flasher
3. Run _Show Logs_ from ESPHome Dashboard
4. Run _OTA Update_ from

Outcome: Same error as when _thingdev_ board is specified. The error says that _esp01_1m always works_ but it doesn't appear to work in this case.

**Logs from ESPFlasher**

```Using 'COM5' as serial port.
Connecting....
Detecting chip type... ESP8266
Connecting...94 %)Writing at 0x00044000... (100 %)Wrote 410816 bytes (281529 compressed) at 0x00000000 in 6.8 seconds (effective 483.1 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[11:57:23][I][logger:156]: Log initialized
[11:57:23][C][ota:364]: There have been 0 suspected unsuccessful boot attempts.
[11:57:23][I][app:028]: Running through setup()...
[11:57:23][C][dallas.sensor:032]: Setting up DallasComponent...
[11:57:23][E][dallas.sensor:193]: Reading scratchpad failed: reset
[11:57:23][C][wifi:033]: Setting up WiFi...
[11:57:23][D][wifi:304]: Starting scan...
[11:57:23][E][dallas.sensor:123]: Requesting conversion failed
[11:57:29][D][wifi:319]: Found networks:
[skipping wifi scanning]
[11:57:29][I][wifi:193]: WiFi Connecting to '[redacted]'...
[11:57:31][I][wifi:423]: WiFi Connected!
[11:57:31][C][wifi:283]:   SSID: '[redacted]'
[11:57:31][C][wifi:284]:   IP Address: 192.168.1.22
[11:57:31][C][wifi:286]:   BSSID: [redacted]
[11:57:31][C][wifi:287]:   Hostname: 'node01'
[11:57:31][C][wifi:291]:   Signal strength: -51 dB ▂▄▆█
[11:57:31][C][wifi:295]:   Channel: 8
[11:57:31][C][wifi:296]:   Subnet: 255.255.255.0
[11:57:31][C][wifi:297]:   Gateway: 192.168.1.1
[11:57:31][C][wifi:298]:   DNS1: 8.8.8.8
[11:57:31][C][wifi:299]:   DNS2: 8.8.4.4
[11:57:31][D][wifi:432]: Disabling AP...
[11:57:31][C][ota:029]: Over-The-Air Updates:
[11:57:31][C][ota:030]:   Address: node01.local:8266
[11:57:31][C][ota:032]:   Using Password.
[11:57:31][C][api:022]: Setting up Home Assistant API server...
[11:57:31][I][app:058]: setup() finished successfully!
[11:57:31][I][app:100]: ESPHome version 1.14.3 compiled on Feb 29 2020, 00:50:48
[11:57:31][C][wifi:415]: WiFi:
[11:57:31][C][wifi:283]:   SSID: '[redacted]'
[11:57:31][C][wifi:284]:   IP Address: 192.168.1.22
[11:57:31][C][wifi:286]:   BSSID: [redacted]
[11:57:31][C][wifi:287]:   Hostname: 'node01'
[11:57:31][C][wifi:291]:   Signal strength: -50 dB ▂▄▆█
[11:57:31][C][wifi:295]:   Channel: 8
[11:57:31][C][wifi:296]:   Subnet: 255.255.255.0
[... 98 lines cut ...]
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <ESPAsyncTCP-esphome> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/node01/src/main.cpp.o
Linking .pioenvs/node01/firmware.elf
Building .pioenvs/node01/firmware.bin
Retrieving maximum program size .pioenvs/node01/firmware.elf
Checking size .pioenvs/node01/firmware.elf
DATA:    [====      ]  44.6% (used 36544 bytes from 81920 bytes)
PROGRAM: [====      ]  39.7% (used 406668 bytes from 1023984 bytes)
Creating BIN file ".pioenvs/node01/firmware.bin" using ".pioenvs/node01/firmware.elf"
========================= [SUCCESS] Took 7.52 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of node01.local
INFO  -> 192.168.1.22
INFO Uploading /config/node01/.pioenvs/node01/firmware.bin (410816 bytes)
ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.```

rbender, 2020-05-24:
I'm getting the same error with my ThingDev board. Running ESPHome 1.14.3 and PlatformIO 4.3.4 and espressif8266@2.2.3.

mcclown, 2020-06-15:
I ended up giving up and getting a NodeMCU 32s instead. I still use my thingdev as a dev board but don't use it for anything I have to maintain.

TobbeJ, 2020-07-17:
Since the board have only 512kb of flash OTA is unlikely to ever work.
If i remember correctly bare minimum for any kind of OTA update 1m but preferably a bit more.
A significant part, probably half, of the flash gets reserved for OTA and your program is close to 512k so no room for OTA and therefor wont work.

I got a couple of these boards too and got very disappointed with the small flash, this makes them almost useless.

closed 2020-11-28 by stale[bot]