esphome/issues #1067: DHT11 sensor has 0 decimal precision

Scrumplex, 2020-02-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Docker with following command: 
```
docker run --rm -v (pwd):/config -it --device /dev/ttyUSB0:/dev/ttyUSB0 esphome/esphome
```

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266, NodeMCU v1.1  (PlatformIO board: nodemcuv2)

**Affected component:**
sensor -> plaform: dht -> model: DHT11

**Description of problem:**
When dumping my MQTT output I only see integer precision float values for the temperature. While I was building my own code for reading the sensor I was able to extract information of up to one decimal precision (like 23.4°C instead of just 23.0°).

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: sefa_air
  platform: ESP8266
  board: nodemcuv2

mqtt:
  broker: 10.10.10.4
  username: hass
  password: <redacted>

sensor:
  - platform: dht
    model: DHT11
    pin: D2
    temperature:
      name: "temperature"
    humidity:
      name: "humidity"
    update_interval: 10s
```

I removed unnecessary config sections like wifi or logger

**Logs (if applicable):**

```
[16:56:42][I][app:028]: Running through setup()...
[16:56:42][C][dht:011]: Setting up DHT...
[16:56:42][C][wifi:033]: Setting up WiFi...
[16:56:42][D][wifi:304]: Starting scan...
[16:56:42][D][dht:048]: Got Temperature=23.0°C Humidity=42.0%
[16:56:42][D][sensor:092]: 'temperature': Sending state 23.00000 °C with 1 decimals of accuracy
[16:56:42][D][sensor:092]: 'humidity': Sending state 42.00000 % with 0 decimals of accuracy
[16:56:48][D][wifi:319]: Found networks:

[...]

[16:56:50][D][dht:048]: Got Temperature=23.0°C Humidity=42.0%
[16:56:50][D][sensor:092]: 'temperature': Sending state 23.00000 °C with 1 decimals of accuracy
[16:56:50][D][sensor:092]: 'humidity': Sending state 42.00000 % with 0 decimals of accuracy
[16:56:50][I][wifi:423]: WiFi Connected!

[...]

[16:56:50][C][mqtt:025]: Setting up MQTT...
[16:56:50][I][mqtt:162]: Connecting to MQTT...
[16:56:50][I][mqtt:202]: MQTT Connected!
[16:56:50][I][app:058]: setup() finished successfully!
[16:56:50][I][app:100]: ESPHome version 1.14.3 compiled on Feb 27 2020, 16:56:26
[16:56:50][C][wifi:415]: WiFi:

[...]

[16:56:50][C][logger:175]: Logger:
[16:56:50][C][logger:176]:   Level: DEBUG
[16:56:50][C][logger:177]:   Log Baud Rate: 115200
[16:56:50][C][logger:178]:   Hardware UART: UART0
[16:56:50][C][dht:017]: DHT:
[16:56:50][C][dht:018]:   Pin: GPIO4 (Mode: INPUT)
[16:56:50][C][dht:022]:   Model: DHT11
[16:56:50][C][dht:027]:   Update Interval: 10.0s
[16:56:50][C][dht:029]:   Temperature 'temperature'
[16:56:50][C][dht:029]:     Unit of Measurement: '°C'
[16:56:50][C][dht:029]:     Accuracy Decimals: 1
[16:56:50][C][dht:029]:     Icon: 'mdi:thermometer'
[16:56:50][C][dht:030]:   Humidity 'humidity'
[16:56:50][C][dht:030]:     Unit of Measurement: '%'
[16:56:50][C][dht:030]:     Accuracy Decimals: 0
[16:56:50][C][dht:030]:     Icon: 'mdi:water-percent'
[16:56:50][C][mqtt:051]: MQTT:

[...]

[16:56:59][D][dht:048]: Got Temperature=23.0°C Humidity=43.0%
[16:56:59][D][sensor:092]: 'temperature': Sending state 23.00000 °C with 1 decimals of accuracy
[16:56:59][D][sensor:092]: 'humidity': Sending state 43.00000 % with 0 decimals of accuracy

```

**Additional information and things you've tried:**
I wrote my own code once to read the data from the sensor and push it directly to an InfluxDB server. It is able to read 1 decimal precision values with Adafruit's DHT sensor library.

https://gitlab.com/Scrumplex/esp8266-dht11-influxdb-collector

daned33, 2020-03-01:
Considering the accuracy of the DHT11 is ±2℃.

Having decimal precision for this sensor is pointless.

Scrumplex, 2020-03-01:
Considering it would be the spec for the chip I would still recommend implmenting this correctly, especially as ESPHome is reporting the value with 1 decimal accuracy.
```
[16:56:59][D][sensor:092]: 'temperature': Sending state 23.00000 °C with 1 decimals of accuracy
```

mannte, 2020-03-10:
This is already implemented in the dev version.

https://community.home-assistant.io/t/esphome-and-dht11-sensor-only-displaying-full-degree-changes/134254/14
https://github.com/esphome/esphome/pull/861

I think you can run both versions alongside each other.

Scrumplex, 2020-03-10:
I just compiled my config file with the latest :dev docker container and it is working. Closing this as solved with esphome/esphome#861

closed 2020-03-10 by Scrumplex