esphome/issues #1068: ERROR Error binary size: Error: ESP does not have enough space to store OTA file

jperquin, 2020-02-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**
 OTA Update

**Description of problem:**
After several succesful succesive OTA updates of these Gosund plugs (I have 7 in total), a recent OTA upload must have been so large that insufficient RAM is left over for a fresh firmware copy, resulting in this error:
`Creating BIN file "/data/gosund_plug_1/.pioenvs/gosund_plug_1/firmware.bin" using "/data/gosund_plug_1/.pioenvs/gosund_plug_1/firmware.elf"
========================= [SUCCESS] Took 16.91 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of gosund_plug_1.local
INFO  -> 192.168.30.144
INFO Uploading /data/gosund_plug_1/.pioenvs/gosund_plug_1/firmware.bin (333152 bytes)
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)`

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: $devicename
  platform: esp8266
  board: esp8285
  arduino_version: recommended
#  arduino_version: latest
#  arduino_version: dev
  board_flash_mode: dout

# Enable over-the-air updates
ota:

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_pw
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
Is there any way to create an even more minimalistic "recovery" firmware.bin so I do not have to go the USB route, which would be a real PITA given the fact that these plugs need soldering..

borisfeldman, 2020-03-03:
I ran into the same issue with a wall-mounted switch a few weeks ago. No matter what I did it wouldn’t accept an ota flash with a “minimal” firmware after a too-big flash attempt. I was getting ready to uninstall the switch from the wall and try to disassemble it to flash. Argh!

Then I thought that I should try rebooting it... After a full power cycle, I was back where I was before my failed flash attempt. Woo hoo!

When all else fails try a reboot. In your case, don’t just toggle the switch’s relay, fully disconnect it from power and then see what you get.

jperquin, 2020-04-04:
Did a full power-cycle, but still getting the same error when updating over the device's webserver..

Update Failed: ERROR[4]: Not Enough Space

Crewski, 2020-05-13:
@jperquin Did you ever figure it out?  I'm running into the same issue.  My full file size is less that 50%, but still can't even update to a completely minimal version.

borisfeldman, 2020-05-13:
@Crewski did you try fully powering down and restarting the device?  For me, things were stuck and wouldn’t flash anything, even the smallest thing. After the power down, it got out of that bad state and I was able to flash to a minimal build over ota. This let me flash ota to a file that was less than 50%.

Crewski, 2020-05-13:
@borisfeldman I tried to power cycle one of the devices, I think a sonoff th10, but still didn’t work. I’ll try it again and see what happens. Thanks.

Crewski, 2020-05-13:
@borisfeldman I tried it again and got it to work.  I flashed as minimal as possible, then added webserver, then tasmota-minimal, then tasmota.  Thanks again for the help.

borisfeldman, 2020-05-13:
Happy to help! Glad that it worked for you.

closed 2020-09-24 by stale[bot]