esphome/issues #1070: 500 Server Error: Internal Server Error

Madman833, 2020-02-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

installed 3 days ago, remote access, sync with phone, etc. the system linked with printers, computers, phones, ipads, smart tvs.
**ESP (ESP32/ESP8266, Board/Sonoff):**
 n/a

**Affected component:**

**Description of problem:**
Cannot install the esphome add-on. reveive the following error.
‘ERROR (SyncWorker_16) [hassio.docker] Can't create container from addon_15ef4d2f_esphome: 500 Server Error: Internal Server Error ("error creating overlay mount to /mnt/data/docker/overlay2/d0b9b373f1e1f6125511c1c1247b4901efa72b1c2decf26ac9bda3488a8e2990-init/merged: too many levels of symbolic links")’

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

‘ERROR (SyncWorker_16) [hassio.docker] Can't create container from addon_15ef4d2f_esphome: 500 Server Error: Internal Server Error ("error creating overlay mount to /mnt/data/docker/overlay2/d0b9b373f1e1f6125511c1c1247b4901efa72b1c2decf26ac9bda3488a8e2990-init/merged: too many levels of symbolic links")’

```

**Additional information and things you've tried:**

I tried to restart HassOs, remove the resp and attach again, restart raspberry, test connection (installed other add-on without issue)

Madman833, 2020-03-01:
I found the issue. The problem was that an image already existed on my Hassio. After I removed it It was possible to install and start to work with it.

OttoWinter, 2020-07-24:
Ok that sounds like a ~~Hass.io~~ Home Assistant bug.

closed 2020-07-24 by OttoWinter