esphome/issues #1072: Remote Receiver showing raw numbers while inactive

vazquezjm, 2020-03-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant

**ESP (ESP32/ESP8266, Board/Sonoff):**

- Wemos D1 mini
- VS1838B 38Khz IR Receiver

**Affected component:**

Remote Receiver https://esphome.io/components/remote_receiver.html

**Description of problem:**
There is nothing pointing at the receiver, but continuously getting values from the IR receiver through the logs, as you can see below. Also, not sure what values do I need to take into account, since I see lots of numbers when the remote key is not pressed.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: wemosd1
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: --------
  password: --------

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: --------
    password: --------

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

remote_receiver:
  pin: D4
  dump: all
  buffer_size: 4kb
```

**Logs (if applicable):**

```
[08:54:13][C][logger:176]:   Level: DEBUG
[08:54:13][C][logger:177]:   Log Baud Rate: 115200
[08:54:13][C][logger:178]:   Hardware UART: UART0
[08:54:13][D][remote.raw:041]: Received Raw: -144
[08:54:13][C][remote_receiver.esp8266:059]: Remote Receiver:
[08:54:13][C][remote_receiver.esp8266:060]:   Pin: GPIO2 (Mode: INPUT)
[08:54:13][W][remote_receiver.esp8266:063]: Remote Receiver Signal starts with a HIGH value. Usually this means you have to invert the signal using 'inverted: True' in the pin schema!
[08:54:13][C][remote_receiver.esp8266:065]:   Buffer Size: 4000
[08:54:13][C][remote_receiver.esp8266:066]:   Tolerance: 25%
[08:54:13][C][remote_receiver.esp8266:067]:   Filter out pulses shorter than: 50 us
[08:54:13][C][remote_receiver.esp8266:068]:   Signal is done after 10000 us of no changes
[08:54:13][D][remote.raw:041]: Received Raw: -87, 1235, -276
[08:54:13][C][captive_portal:169]: Captive Portal:
[08:54:13][C][ota:029]: Over-The-Air Updates:
[08:54:13][C][ota:030]:   Address: wemosd1.local:8266
[08:54:13][C][api:095]: API Server:
[08:54:13][C][api:096]:   Address: wemosd1.local:6053
[08:54:13][D][remote.raw:041]: Received Raw: -118, 2892, -251, 6685, -142
[08:54:13][D][remote.raw:041]: Received Raw: -91
[08:54:16][D][remote.raw:041]: Received Raw: -82
[08:54:16][D][remote.raw:041]: Received Raw: -217
[08:54:16][D][remote.raw:041]: Received Raw: -220
[08:54:17][D][remote.raw:041]: Received Raw: -139
[08:54:17][D][remote.raw:041]: Received Raw: -122
[08:54:17][D][remote.raw:041]: Received Raw: -89
[08:54:17][D][remote.raw:041]: Received Raw: -86
[08:54:17][D][remote.raw:041]: Received Raw: -98
[08:54:17][D][remote.raw:041]: Received Raw: -246
[08:54:17][D][remote.raw:041]: Received Raw: -220
[08:54:17][D][remote.raw:041]: Received Raw: -142
[08:54:17][D][remote.raw:041]: Received Raw: -64
[08:54:17][D][remote.raw:041]: Received Raw: -92, 538, -79, 1379, -184
[08:54:17][D][remote.raw:041]: Received Raw: -86, 1945, -251
[08:54:17][D][remote.raw:041]: Received Raw: -102
[08:54:17][D][remote.raw:041]: Received Raw: -76
[08:54:17][D][remote.raw:041]: Received Raw: -88
[08:54:17][D][remote.raw:041]: Received Raw: -87
[08:54:17][D][remote.raw:041]: Received Raw: -89
[08:54:17][D][remote.raw:041]: Received Raw: -89, 1752, -83
[08:54:17][D][remote.raw:041]: Received Raw: -168
[08:54:18][D][remote.raw:041]: Received Raw: -96, 3625, -220
[08:54:19][D][remote.raw:041]: Received Raw: -87
[08:54:23][D][remote.raw:041]: Received Raw: -143
[08:54:23][D][remote.raw:041]: Received Raw: -218
[08:54:23][D][remote.raw:041]: Received Raw: -83, 841, -252
[08:54:23][D][remote.raw:041]: Received Raw: -142
[08:54:25][D][remote.raw:028]: Received Raw: -9034, 4497, -547, 580, -547, 580, -547, 1706, -546, 613, -514, 582, -546, 580, -547, 619, -506, 583, -546, 1708, -544, 1707, -545, 604, -523, 1707, -546, 1706, -546, 1719, -534, 1706, -547, 1706, -547, 612, -515, 613, -512, 585, -542, 
[08:54:25][D][remote.raw:041]:   1708, -546, 582, -545, 581, -547, 581, -546, 580, -546, 1739, -514, 1707, -547, 1706, -546, 581, -546, 1739, -514, 1707, -546, 1709, -542, 1709, -545
[08:54:25][D][remote.raw:041]: Received Raw: -9047, 2273, -546
[08:54:25][D][remote.raw:041]: Received Raw: -161
[08:54:25][D][remote.raw:041]: Received Raw: -208
[08:54:25][D][remote.raw:041]: Received Raw: -121
[08:54:25][D][remote.raw:041]: Received Raw: -217
[08:54:25][D][remote.raw:041]: Received Raw: -136
[08:54:25][D][remote.raw:041]: Received Raw: -219
[08:54:25][D][remote.raw:041]: Received Raw: -163
```

**Additional information and things you've tried:**

I did try with different values for `remote_receiver`, like so:
```
tolerance: 50%
filter: 250us
idle: 4ms
```
I also added a decoupling capacitor on the IR receiver. I moved it away from the WiFi antenna on the Wemos module. Basically, the same outcome. 

Something interesting to notice (at least for me) is this:

[image]

As you can see, when the remote key (power button) is pressed, log shows `[remote.raw:028]` instead of `[remote.raw:041]`

glmnet, 2020-03-03:
all this is kind of normal, your sensor is picking noise, probably sun light
It happens more often with RF receiver though.
It does not happens most of the time, I believe it is there just to show you there is noise
try using D5 and adding a pull up resistor in the pin, or configuring it with INPUT_PULLUP

````yaml
remote_receiver:
  id: rcvr
  pin:
    number: D5
    inverted: True
    mode: INPUT_PULLUP
````

vazquezjm, 2020-03-03:
Thanks @glmnet 

I wired the IR sensor as described in Part 1, here: http://www.electronoobs.com/eng_arduino_tut34

Connected to pin D5 in Wemos as you suggested and everything is much more stable now. The thing is that I still cannot find a clear patter... These are the codes I'm getting (AC):

Turn on:
```
[remote.raw:028]: Received Raw: -4405, 4399, -548, 1635, -545, 551, -546, 1638, -547, 1637, -546, 548, -547, 549, -546, 1636, -546, 553, -544, 526, -572, 1639, -547, 523, -572, 523, -572, 1636, -547, 1635, -546, 550, -547, 1616, -571, 553, -545, 523, -571, 524, -571, 
[remote.raw:028]:   1611, -572, 1635, -547, 1636, -545, 1636, -547, 1612, -574, 1615, -573, 1609, -572, 1636, -546, 548, -547, 523, -573, 523, -572, 549, -545, 526, -572, 1613, -571, 1635, -548, 548, -547, 1639, -546, 524, -572, 1640, -546, 549, -546, 525, -572, 526, 
[D][remote.raw:028]:   -571, 548, -547, 1614, -572, 523, -572, 1615, -572, 547, -547, 1615, -572, 1611, -541, 5269, -4406, 4385, -547, 1610, -572, 551, -544, 1640, -546, 1635, -548, 523, -572, 548, -548, 1634, -547, 525, -572, 551, -546, 1613, -574, 549, -544, 550, -546, 
(Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[D][remote.raw:041]:   -546, 548, -548, 1613, -574, 522, -572, 1614, -572, 523, -572, 525, -572, 551, -546, 549, -548, 1612, -573, 552, -543, 1614, -572, 550, -544, 1641, -546, 1638, -491
```

Turn off:
```
[D][remote.raw:028]: Received Raw: -4404, 4383, -545, 1636, -547, 549, -546, 1639, -547, 1637, -546, 522, -574, 522, -572, 1635, -547, 526, -571, 526, -571, 1640, -547, 549, -545, 550, -546, 1609, -573, 1637, -544, 549, -547, 1616, -572, 555, -515, 1637, -572, 1610, -572, 
[D][remote.raw:028]:   1637, -545, 1638, -545, 548, -546, 1614, -572, 1638, -547, 1641, -545, 550, -546, 551, -544, 524, -572, 549, -546, 1636, -545, 524, -572, 551, -546, 1641, -546, 1636, -547, 1610, -572, 522, -573, 549, -546, 523, -572, 524, -571, 551, -546, 552, -545, 
[D][remote.raw:028]:   548, -547, 522, -573, 1640, -546, 1636, -545, 1638, -545, 1636, -547, 1610, -542, 5251, -4432, 4365, -568, 1636, -545, 524, -571, 1640, -547, 1636, -546, 548, -546, 550, -546, 1637, -544, 526, -571, 551, -547, 1639, -546, 553, -542, 550, -545, 1635, 
(Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[D][remote.raw:041]:   1611, -572, 551, -544, 522, -574, 549, -545, 549, -546, 525, -572, 525, -572, 550, -546, 548, -545, 1640, -548, 1634, -546, 1638, -546, 1609, -572, 1640, -513
```

Temp down:
```
[D][remote.raw:028]: Received Raw: -4406, 4391, -520, 1660, -547, 548, -547, 1640, -521, 1661, -520, 574, -547, 548, -547, 1635, -521, 551, -572, 551, -544, 1642, -546, 523, -571, 531, -541, 1634, -572, 1609, -573, 549, -520, 1667, -547, 550, -521, 573, -523, 575, -519, 
[D][remote.raw:028]:   1661, -521, 1641, -540, 1636, -572, 1610, -572, 1611, -573, 1641, -521, 1660, -547, 1609, -573, 549, -520, 576, -519, 575, -521, 575, -519, 577, -520, 550, -572, 548, -547, 1636, -521, 1635, -571, 550, -519, 1665, -521, 575, -521, 549, -572, 1616, 
[D][remote.raw:028]:   -547, 1660, -522, 548, -573, 522, -572, 1615, -541, 575, -520, 1666, -520, 1661, -515, 5277, -4380, 4390, -541, 1661, -546, 552, -519, 1664, -546, 1636, -547, 537, -558, 548, -543, 1638, -546, 554, -519, 575, -521, 1666, -544, 545, -543, 557, -522, 
(Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[D][remote.raw:041]:   1636, -571, 1616, -565, 550, -520, 1639, -571, 523, -572, 526, -572, 1641, -521, 1661, -546, 548, -547, 548, -546, 1610, -571, 549, -548, 1638, -546, 1620, -509
```

Temp up:
```
[D][remote.raw:028]: Received Raw: -4430, 4370, -573, 1609, -572, 523, -572, 1614, -572, 1635, -546, 549, -546, 548, -546, 1636, -546, 553, -545, 524, -572, 1614, -572, 549, -546, 523, -572, 1616, -565, 1635, -548, 547, -547, 1641, -547, 553, -543, 549, -547, 523, -572, 
[D][remote.raw:028]:   1635, -546, 1637, -545, 1635, -546, 1637, -544, 1612, -572, 1645, -544, 1609, -572, 1635, -547, 549, -545, 560, -536, 553, -541, 524, -571, 552, -547, 523, -572, 548, -547, 1610, -571, 523, -573, 523, -570, 1637, -545, 524, -572, 550, -547, 1642, -545, 
[D][remote.raw:028]:   1637, -546, 522, -573, 1639, -547, 1634, -547, 549, -545, 1614, -573, 1636, -491, 5274, -4430, 4360, -572, 1611, -571, 549, -546, 1614, -572, 1610, -572, 550, -544, 551, -545, 1635, -546, 552, -545, 526, -571, 1614, -572, 549, -546, 549, -546, 1634, 
(Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[D][remote.raw:041]:   -546, 551, -544, 524, -571, 1637, -544, 524, -572, 552, -544, 1642, -547, 1609, -573, 522, -572, 1641, -544, 1637, -545, 524, -571, 1642, -545, 1612, -541
```

glmnet, 2020-03-03:
That's the best you will get with ESPHome, it is not suitable to do much signal analysis.
The AC's long code is split in several lines because it is long, the lines 028 are chunks, and line 041 is the last chunk
the "message skipped" has a similar explanation, are ESPHome inner workings related to logging.

I however feel like your signal is inverted, usually the first number should be positive, try inverting your value of `inverted`

which AC is this? ESPHome supports a few AC already but not many.

vazquezjm, 2020-03-03:
> I however feel like your signal is inverted, usually the first number should be positive, try inverting your value of `inverted`

Weird, it looks like it does not support it?
[image]

> which AC is this? ESPHome supports a few AC already but not many.
BGH. I think it is not supported.

glmnet, 2020-03-03:
Check the indentation 😀 (look carefully at my first answer)

---

those Argentinian AC

Try the whirlpool one, it is available in the https://github.com/esphome/esphome/tree/climate-whirlpool branch

glmnet, 2020-03-03:
And here with receiver https://github.com/mmanza/esphome/tree/climate-whirlpool

vazquezjm, 2020-03-03:
> Check the indentation 😀 (look carefully at my first answer)

That did the trick 💯 

Now I have this definition:
```
remote_receiver:
  id: rcvr
  dump: all
  pin:
    number: D5
    inverted: True
```

Look how the response changed when I add `inverted: true`:

Turn on: 
`[15:29:18][D][remote.samsung:055]: Received Samsung: data=0xB24D9F60`

Turn off:
`[15:29:37][D][remote.samsung:055]: Received Samsung: data=0xB24D7B84`

Temp up:
`[15:29:51][D][remote.samsung:055]: Received Samsung: data=0xB24D9F60`

Temp down:
`[15:30:33][D][remote.samsung:055]: Received Samsung: data=0xB24D9F60`

Same data for On, Temp down, Temp up commands?

glmnet, 2020-03-03:
Usually AC remotes don't work like common TVs or other appliances remotes.

They send a full stream of the state, including power state, temperature, mode, fan speed etc.
ESPHome is not going to tell you that doing this.

you first need to know which protocol your ac uses, I recommend you try the https://github.com/crankyoldgit/IRremoteESP8266  project for that, then you can use a [climate integration](https://esphome.io/components/climate/coolix.html) to control your AC

vazquezjm, 2020-03-03:
Will do then. Thank you so much for your help @glmnet !!

closed 2020-03-03 by vazquezjm