esphome/issues #1073: on_boot i cannot get homeassistant.event to work

donnib, 2020-03-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: smoke_kontor
  platform: ESP8266
  board: esp01_1m
  on_boot:
    then:
      - logger.log: "sending event"
      - homeassistant.event:
          event: esphome.event_test
          data:
            title: My event

```
but i also tried 
```yaml
esphome:
  name: smoke_kontor
  platform: ESP8266
  board: esp01_1m
  on_boot:
    priority: -100
    then:
      - logger.log: "sending event"
      - homeassistant.event:
          event: esphome.event_test
          data:
            title: My event
```
but no matter what i do i don't get events in home assistant Developer tools -> Events and i listen to esphome.event_test however i do see the log "sending event" in the logs.

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

randybb, 2020-03-04:
Because it takes few seconds to get it connected to HA. Maybe you can try to add some [delay](https://esphome.io/guides/automations.html#delay-action) before your actions.

donnib, 2020-03-04:
I don’t like delays, can it be done without delay, hooking up to an event when it is connected?

randybb, 2020-03-04:
https://esphome.io/components/binary_sensor/status.html and then _on_state:_ is what you are looking for.

donnib, 2020-03-04:
Thanks but I want the event on boot. How is that possible ?

randybb, 2020-03-04:
There is nothing that would trigger an event after device has been rebooted and connected, but you can define lambda condition (https://esphome.io/guides/automations.html#lambda-condition) and do any complex decisions in automation - something like if status is changed to connected and uptime counter (https://esphome.io/components/sensor/uptime.html) is < 5 minutes, then it will fire your event. And sure, you can check status and uptime from HA perspective and have the logic in one place without relaying on configuration on ESPHome site.

estebanz01, 2020-05-22:
I'm experiencing a similar behaviour with homeassistant.service. It does not get called. I added a `wait_until:` and `priority: -500` without success. Here's my `on_boot`:
```yaml
esphome:
  name: esp32cam
  platform: ESP32
  board: esp32cam
  on_boot:
    priority: -500
    then:
      - wait_until:
          api.connected:
      - logger.log: Api connected!
      - homeassistant.service:
          service: input_boolean.turn_on
          data:
            entity_id: input_boolean.cameraon
      - logger.log: Homeassistant service called!
  on_shutdown:
    - logger.log: Turning off!
    - homeassistant.service:
        service: input_boolean.turn_off
        data:
          entity_id: input_boolean.cameraon
```

Sometimes, the `on_shutdown:` works, sometimes it doesn't, but I'm not concerned about that one.

estebanz01, 2020-05-22:
Here's my relevant output log:
```
[18:21:38][C][esp32_camera:043]: ESP32 Camera:
[18:21:38][C][esp32_camera:044]:   Name: Camara Vigilancia
[18:21:38][C][esp32_camera:045]:   Board Has PSRAM: YES
[18:21:38][C][esp32_camera:047]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[18:21:38][C][esp32_camera:048]:   VSYNC Pin: 25
[18:21:38][C][esp32_camera:049]:   HREF Pin: 23
[18:21:38][C][esp32_camera:050]:   Pixel Clock Pin: 22
[18:21:38][C][esp32_camera:051]:   External Clock: Pin:0 Frequency:20000000
[18:21:38][C][esp32_camera:052]:   I2C Pins: SDA:26 SCL:27
[18:21:38][C][esp32_camera:053]:   Reset Pin: -1
[18:21:38][C][esp32_camera:077]:   Resolution: 800x600 (SVGA)
[18:21:38][C][esp32_camera:097]:   JPEG Quality: 10
[18:21:38][C][esp32_camera:099]:   Contrast: 0
[18:21:38][C][esp32_camera:100]:   Brightness: 0
[18:21:38][C][esp32_camera:101]:   Saturation: 0
[18:21:38][C][esp32_camera:102]:   Vertical Flip: ON
[18:21:38][C][esp32_camera:103]:   Horizontal Mirror: ON
[18:21:39][C][esp32_camera:120]:   Test Pattern: NO
[18:21:39][C][captive_portal:169]: Captive Portal:
[18:21:39][C][ota:029]: Over-The-Air Updates:
[18:21:39][C][ota:030]:   Address: 192.168.1.84:3232
[18:21:39][C][ota:032]:   Using Password.
[18:21:39][C][api:095]: API Server:
[18:21:39][C][api:096]:   Address: 192.168.1.84:6053
[18:21:39][C][deep_sleep:020]: Setting up Deep Sleep...
[18:21:39][C][deep_sleep:026]:   Run Duration: 120000 ms
[18:21:39][C][deep_sleep:030]:   Wakeup Pin: GPIO13 (Mode: INPUT)
[18:21:39][D][esp32_camera:157]: Got Image: len=47840
[18:21:39][D][esp32_camera:157]: Got Image: len=48265
[18:21:39][D][esp32_camera:157]: Got Image: len=48756
[18:21:39][D][esp32_camera:157]: Got Image: len=48869
[18:21:39][D][esp32_camera:157]: Got Image: len=48809
[18:21:39][D][esp32_camera:157]: Got Image: len=48855
[18:21:40][D][esp32_camera:157]: Got Image: len=48716
[18:21:40][D][esp32_camera:157]: Got Image: len=48251
[18:21:40][D][esp32_camera:157]: Got Image: len=47887
[18:21:40][D][esp32_camera:157]: Got Image: len=47713
[18:21:40][D][esp32_camera:157]: Got Image: len=46558
[18:21:40][D][esp32_camera:157]: Got Image: len=45143
[18:21:41][D][esp32_camera:157]: Got Image: len=43467
[18:21:41][D][esp32_camera:157]: Got Image: len=45729
[18:21:41][D][esp32_camera:157]: Got Image: len=49601
[18:21:41][D][esp32_camera:157]: Got Image: len=46605
[18:21:41][D][esp32_camera:157]: Got Image: len=45303
[18:21:45][D][main:106]: Api connected!
[18:21:45][D][main:194]: Homeassistant service called!
[18:21:45][D][api.connection:583]: Client 'Home Assistant 0.110.1 (192.168.1.131)' connected successfully!
```

estebanz01, 2020-05-22:
I added a delay of two seconds before the homeassistant call and it worked.

closed 2020-10-04 by stale[bot]