esphome/issues #1076: ESP32: Postential to increase app partitions by 192kb.

MichalFoksa, 2020-03-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
- Docker 

**ESP (ESP32/ESP8266, Board/Sonoff):**
- ESP32

**Affected component:**
- N/A

**Description of problem:**
Hey, ESPHome is awesome product - good luck, keep going!

I have got my favorite `The program size (1639694 bytes) is greater than maximum allowed (1638400 bytes)` error message.

I looked up into ESP32 partition table and found out there is potentially un-allocated 384KB space between end of `app0` partition and start of `app1` partition.

When subtracted app1 offset from app0 offset and app0 size, I got 384kb un-allocated space.
`0x200000` - `0x010000` - `0x190000` = `0x60000` -> `384kb`.

Current partition table from in `MY-ESP-xxx/partitions.csv`:

```
Name      Type  SubType  Offset    Size
-------------------------------------------
nvs,      data, nvs,     0x009000, 0x005000
otadata,  data, ota,     0x00e000, 0x002000
app0,     app,  ota_0,   0x010000, 0x190000
app1,     app,  ota_1,   0x200000, 0x190000
eeprom,   data, 0x99,    0x390000, 0x001000
spiffs,   data, spiffs,  0x391000, 0x00F000 
```

Am I right? Can it be exploited to increase application partitions by 192kb each?

If yes, following is partition table where app0 and app1 partitions are increased by 192kb each.

```
Name,     Type, SubType, Offset,   Size
-------------------------------------------
nvs,      data, nvs,     0x009000, 0x005000,
otadata,  data, ota,     0x00e000, 0x002000,
app0,     app,  ota_0,   0x010000, 0x1C0000,
app1,     app,  ota_1,   0x1D0000, 0x1C0000,
eeprom,   data, 0x99,    0x390000, 0x001000,
```

Michal

**Logs (if applicable):**

```
Processing homeyduino1 (board: esp32dev; framework: arduino; platform: espressif32@1.11.0)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
....
Checking size .pioenvs/homeyduino1/firmware.elf
DATA:    [==        ]  18.7% (used 61364 bytes from 327680 bytes)
Error: The program size (1639694 bytes) is greater than maximum allowed (1638400 bytes)
PROGRAM: [==========]  100.1% (used 1639694 bytes from 1638400 bytes)
*** [checkprogsize] Explicit exit, status 1

```

**Additional information and things you've tried:**

OttoWinter, 2020-07-25:
Closed in https://github.com/esphome/esphome/pull/1197

OttoWinter, 2020-07-25:
Thanks for spotting this @MichalFoksa - Indeed there's more space available there. I apparently had a brainfart and though in decimal when initially writing that code.

closed 2020-07-25 by OttoWinter

MichalFoksa, 2020-07-25:
You are welcome! Happy to help. 

Does it really work, cause I could not make it boot successfully?